dotnet/iot #1402: Board: RPI4B, System: Ubuntu2004,  Question: Not Found (libc.so、libbcm_host.so、libgpiod.so) Library!

527395632, 2021-01-07:
English Language (After translation):
Hello!
       My Name is Yan Kai, The board I use (rpi4b, ubuntu20.04, arm64) reported an error when referring to this library. The specific error content is not found“ libc.so 、libbcm_ host.so , libgpiod.so I checked a lot of information about these library files and found them in the directory of / usr / lib / arm linux gnueabihf libc.so , but still can't find "libbcm"_ host.so , libgpiod.so "Two library files. I also looked in GitHub and found "libgpiod" related open source projects“ https://github.com/brgl/libgpiod ", not sure what is referenced in the project" libgpiod.so "Is it the assembly generated by this project? In addition," libbcm said_ host.so "The library file has not been found. Can I ask for the support of the great God?

Chinese Language (Untranslated):
您好! 
       我的名字是鄢凯, 我使用的板子(RPI4B, Ubuntu20.04, Arm64), 在引用此库的时候报错了, 具体的错误内容为找不到"libc.so、libbcm_host.so、libgpiod.so"这几个库文件, 我查了很多资料, 在"/usr/lib/arm-linux-gnueabihf"目录下找到了libc.so, 但是仍然找不到"libbcm_host.so、libgpiod.so"两个库文件。我在Github中也查找, 找到"libgpiod"相关开源项目 "https://github.com/brgl/libgpiod", 不确定项目中引用的"libgpiod.so"是否就是此项目生成的程序集, 另外, "libbcm_host.so"库文件始终未找到, 是否可以请求大神支援?

krwq, 2021-01-11:
@527395632 can you try running `sudo apt-get install -y libgpiod-dev`

527395632, 2021-01-19:
The problem has been solved. Thank you very much

krwq, 2021-01-19:
Closing the issue, please re-open and comment should you have any more questions

closed 2021-01-19 by krwq