dotnet/iot #1403: We should replace dependency on System.Drawing with ImageSharp

krwq, 2021-01-07:
System.Drawing is officially deprecated, we should use ImageSharp instead. See: https://docs.microsoft.com/en-us/dotnet/api/system.drawing?view=dotnet-plat-ext-5.0#remarks

Ellerbach, 2021-02-18:
[Triage] We need to list all the bindings that uses System.Drawing and identify groups which need to be adjusted together. We also need to make sure that the person who'll do some of the work has the proper hardware to test it.

Ellerbach, 2021-02-19:
Here is the list of references to System.Drawing in the repo:
- samples\led-matrix-weather
- src\devices\Apa102 -> seems to only use Color
- src\devices\Bh1745 -> seems to only use Color
- src\devices\Common\Iot\Device\Graphics -> seems to only use Color
- src\devices\GoPiGo3 -> seems to only use Color
- src\devices\Media\VideoDevice -> uses Color and PixelFormat, Size and maybe few others. This will be more complicated to migrate
- src\devices\PiJuice -> seems to only use Color
- src\devices\RGBLedMatrix -> Color, Rectangle and most likely few others. This may require quite some efforts
- src\devices\SenseHat -> seems to only use Color
- src\devices\Ssd1351 -> uses Color, Bitmap, Point. This will require quite some efforts
- src\devices\Tcs3472x -> seems to only use Color
- src\devices\Ws28xx -> seems to only use Color

Those only using Color are quite straight forward to migrate. They require always the same pattern to transform the color space to the right one. A limited hardware can be needed to test but the pattern will always be the same. So that can limit the errors when the hardware is not available.

The problematic ones will be VideoDevice, RGBLedMAtrix and Ssd1351. They will for sure require the hardware to test and make sure all is working perfectly.

MaxPrimeAERY, 2021-02-25:
VideoDevice doesn't work for me properly. I get an error in this part of code
 video = device.CaptureContinuous();
 Bitmap myBitmap = new Bitmap(video);

System.ArgumentException: Parameter is not valid. 
   at System.Drawing.Image.InitializeFromStream(Stream stream)
   at System.Drawing.Bitmap..ctor(Stream stream)

Ellerbach, 2021-02-25:
@MaxPrimeAERY can you please create a separate issue for this problem? This is not related to this specific issue.

JimBobSquarePants, 2021-07-20:
We're gonna need help implementing Font Hinting to be able to render test properly on low resolution devices. Any assistance there would be most welcome.

https://github.com/SixLabors/Fonts/issues/30

krwq, 2021-07-20:
@JimBobSquarePants notice we currently use BDF fonts for very low resolution (couple pixels by couple pixels) but I agree it would be good to have that support

A-J-Bauer, 2022-01-18:
I understand the reason for moving away from System.Drawing.
[System.Drawing.Common only supported on Windows](https://docs.microsoft.com/en-us/dotnet/core/compatibility/core-libraries/6.0/system-drawing-common-windows-only)

Is there a particular reason for picking ImageSharp over e.g. SkiaSharp?

krwq, 2022-01-18:
@A-J-Bauer I guess at this point we've already started using it here in some bindings, I can't remember exactly now but I think it might have been also the only alternative in the [article](https://docs.microsoft.com/en-us/dotnet/api/system.drawing?view=dotnet-plat-ext-5.0#remarks) at the time this issue got created (now I can see it mentions other alternatives, either overlook or it wasn't there before). I think this is still not set in stone but at this point folks got most familiar with it so might be the least work. I'm happy to reiterate on this decision, I personally don't have strong feelings to any library but we will need to test it on all devices if we plan to change this again. I'll be happy to hear any good comparison between couple of alternatives. In most of the devices I doubt library choice will be a bottleneck - I suspect SPI/I2C transmission will be much slower than any library choice we have.

raffaeler, 2022-01-18:
I am not very familiar with SkiaSharp, but the two APIs are quite different and porting code could take a while.

Another reason to stay with System.Drawing is the amount of examples/code available in the wild.

I agree with @krwq that any library would be faster than I2C/SPI devices. But since this is an abstraction, it could make sense to use a library that can **also** be adopted when drawing on a regular display over HDMI. Anyway, I am not aware of any perf gain with SkiaSharp on the RPi for HDMI. Any comment on this is also welcome.

my 2 cents

A-J-Bauer, 2022-01-19:
@krwq The bottleneck analogy suggests that drawing functions are done in parallel and the program flow is constantly waiting for the bus to deliver bytes to the display's chip. In practice the drawing functions add significantly to the time needed to show a frame on the display since it is usually done sequentially (game loop). Even if using back buffers, doing things like 'Graphics DrawImage' from GDI+ or BitBlt from GDI in pure managed code without an underlying unmanaged custom memcpy (not Buffer.Blockcopy) somwhere is just very slow.

@raffaelerI think trying to draw directly over HDMI is not applicable. For larger screens people use a browser (possibly in kiosk mode if on the same device) as frontend and a web server on the device for dealing the HTML and Script (interestingly most browsers use Skia for the rendering). The question should probably be what is the best option for a graphics engine/library for small I2C/SPI displays with a dedicated chip when using C# for multiple plattforms.

SkiaSharp (backed by Microsoft Xamarin), Skia (backed by Google Chromium), ImageSharp is not nearly finished it seems..

SkiaSharp is more like System.Drawing than ImageSharp actually and provides Alias fonts without fiddeling around on the target plattform font (via configs) which is especially helpful for readability on black/white displays and small displays in general.

<details>
  <summary>alias antialias text sample code</summary>

<br/>ran on a Raspberry Pi W2:

[image] [image]

```C#

// declare an array of colors
SKColor[] colors = new SKColor[]
{
    SKColors.White,
    SKColors.Green,
    SKColors.Blue,
    SKColors.Red,
    // SKColors.Cyan,
    // SKColors.Yellow
};

// get the default font manager
SKFontManager fontManager = SKFontManager.Default;

// output the font names returned by the font manager to console
foreach (string fontFamilyName in fontManager.FontFamilies)
{
    Console.WriteLine(fontFamilyName);
}

// create a paint object
SKPaint paint = new SKPaint();

// create two back buffer bitmaps high enough to print all font family names on
SKBitmap bitmapFontsAlias = new SKBitmap(128, fontManager.FontFamilyCount * 10, SKColorType.Rgb565, SKAlphaType.Premul);
SKBitmap bitmapFontsAntiAlias = new SKBitmap(128, fontManager.FontFamilyCount * 10, SKColorType.Rgb565, SKAlphaType.Premul);

// create two canvases to be able to draw on the bitmaps
SKCanvas canvasFontsAlias = new SKCanvas(bitmapFontsAlias);
SKCanvas canvasFontsAntiAlias = new SKCanvas(bitmapFontsAntiAlias);

// write the font names returned by the font manager to the canvases (and automatically to the underlying bitmaps)
int y = 0;
foreach (string familyName in fontManager.FontFamilies)
{
    SKFont font = new SKFont(SKTypeface.FromFamilyName(familyName), 14);

    // pick a 'random' color from the color array
    paint.Color = colors[Random.Shared.Next(colors.Length)];

    font.Edging = SKFontEdging.Alias;
    canvasFontsAlias.DrawText(familyName, 0, y, font, paint);

    font.Edging = SKFontEdging.Antialias;
    canvasFontsAntiAlias.DrawText(familyName, 0, y, font, paint);

    y += 14;
}

// save the alias bitmap
using (var sKFileWStream = new SKFileWStream("fontAlias.png"))
{
    if (sKFileWStream != null)
[... 378 lines cut ...]
        {
            await Task.Run(() => Update());
        }

        /// <summary>
        /// Power off and clean up
        /// </summary>
        public void Dispose()
        {
            PowerOff();

            _bitmap.Dispose();
            _canvas.Dispose();
            _spiDevice.Dispose();
            _gpioController.Dispose();
        }
    }
}
```
</details>

[image]

https://dev.to/mindplay/comment/aka9
https://swharden.com/CsharpDataVis/alt/drawing-with-ImageSharp.md.html

A-J-Bauer, 2022-01-19:
@raffaeler no offense, not saying you can't, just thinking people that are using (usually bigger) displays (> 1080x720 pixels) probably want to have widgets, textboxes, input a.s.o. and doing all this from scratch is just not justified, but for custom visualization or a game there is [spezial builds of SKIA](https://skia.org/docs/user/special/) for OpenGL,  not entirely sure but for iot devices this might or might not be usefull.

raffaeler, 2022-01-19:
@A-J-Bauer It is very much use-case dependent. You do not always need a controls library. 
For example, if you are using the display to just show some data being acquired (graphs are the typical example), you may want to avoid the browser and html at all (which requires the whole desktop environment and running the browser in kiosk mode, sacrificing a lot of memory). Instead it would be far easier using a standard drawing library. 
I am stressing that it would be useful to provide the same abstraction for both i2c/spi devices and the normal GPU/HDMI displays.
For example I am using 3.5" and 7" displays that are both attached to the RPi via HDMI. 

Ideally, the best library from this perspective would be the one providing an accelerated "driver" (OpenGL?) and also a simple way to create "custom drivers" for i2c/spi devices.
I am not proficient with Skia to say if it fits in this scenario.

A-J-Bauer, 2022-01-19:
@raffaeler 
_"It is very much use-case dependent"_ 
Totally agree,
_"For example I am using 3.5" and 7" displays that are both attached to the RPi via HDMI"_ 
Do you know which library you are using for this or what type of display (brand) you are using?
SKIA might have the potential for future utilization of the GPU if the host supports this,

I have the feeling that SkiaSharp would be the better choice for making the necessary move away from System.Drawing, that's all.

A-J-Bauer, 2022-01-19:
@raffaeler
ok you convinced me, so I found some promising links to dig thru regarding directly drawing over HDMI (which I now know on Linux is into a frame buffer device), the last two look especially interesting.

https://medium.com/@avik.das/writing-gui-applications-on-the-raspberry-pi-without-a-desktop-environment-8f8f840d9867
https://github.com/mono/SkiaSharp/issues/492
https://stackoverflow.com/questions/53360854/skiasharp-drawing-image-with-gpu-acceleration

krwq, 2022-01-19:
I don't want to get into a very lengthy thread on this issue, can you guys create a new thread and try to compile some table where we can easily compare benefits of any proposed library? I'd suggest at minimum include:
- license
- support level / responsiveness on github (i.e. in case of blocking issue or security issue how long will we wait for the update; how long if we contribute fix ourselves)
- small fonts support (i.e. will it look good with 32x32 LED matrix or some other tiny display)
- transition difficulty
- anything relevant to dotnet/iot scenarios you can think of

Alternatively if compiling table is hard, once there is enough data we can perhaps schedule some short meeting with everyone on the thread + everyone attending triage meeting where we can discuss this subject.

krwq, 2022-01-19:
or you can hijack #displays channel on discord

Ellerbach, 2022-01-19:
@A-J-Bauer we can use this Discord channel to discuss and then do a summary here (link here: https://discord.gg/GyEfttux)

raffaeler, 2022-01-19:
@A-J-Bauer  ok, just to wrap-up the discussion for future readers:
- I don't remember the library name, sorry. I don't have the repo available at this time
- If Skiasharp fits the requirement, my thumb is up

I will open a different thread for the requirements so that it gets indexed. For any other discussion, let's do it on discord (my handle is raf).

joperezr, 2022-04-14:
Hey @michaelgsharp  some time ago in an offline conversation I believe you were considering creating a drawing abstraction layer for ML.NET, and at that time I told you that if you were to do this we (dotnet/iot) would be interested in potentially use that abstraction as well. Are we still planning on doing this?

michaelgsharp, 2022-04-15:
We are still planning on doing it, though we haven't figured out exactly what we will do or when we will do it. Worst case scenario is that it needs to be done before the release of .NET 7 in November (since .NET 7 no longer supports System.Drawing on Unix). My guess is that it will be much sooner than that, but I can't guarantee that.

Do you have any thoughts on what you would like it to look like?

joperezr, 2022-04-15:
That's great to hear! We haven't really thought about it too much, but we are currently using ImageSharp so the closer it is to that the easier it would be for us to adapt our code. That said, I don't really think it would not be too bad to adapt to the design that you guys decide on. I'd be interested to be on the loop for API reviews where this gets proposed just to make sure that the design would accommodate our needs.

joperezr, 2022-06-09:
Closing this in favor of https://github.com/dotnet/iot/issues/1767

closed 2022-06-09 by joperezr