dotnet/iot #1407: Explanation about SPI speed

JohnyCage2020, 2021-01-13:
Hello , I started to using yours library. I am impressed with how many things I have achieved on my Raspberry PI with this library. I created a device which comunicate with sensor throught SPI. But there is one isue which I can't solve. It's about SPI speed, while I am using class **SoftwareSpi** I am achiving sample rate around 40k samples/s. When I am trying to use your class **SpiDevice** which from my understanding is destined to use hardware SPI my sample rate is around 20k samples/s. Can you explain me what its caused by?  From the definition hardware SPI should be faster than software.  It's there any solution to achive faster SPI speed while using
hardware SPI ?  Can you give more detail's about SPI speed in the documentation's. Thanks for your's answer.

krwq, 2021-01-13:
Hey @JohnyCage2020, when you say you are using SpiDevice do you mean you create it using SpiDevice.Create? SoftwareSpi is also an SpiDevice.
When you say samples/s do you mean bytes, bits or some other unit per second? (what I wrote below is in bits/s so you will need to divide accordingly if you meant other unit)

For faster hardware SPI speeds I believe you will need something similar we do here:
https://github.com/dotnet/iot/tree/master/src/devices/Ws28xx#raspberry-pi-setup-bootconfigtxt

and set it with SpiConnectionSettings, i.e.:
https://github.com/dotnet/iot/blob/master/src/devices/Ws28xx/samples/Program.cs#L17

I've never measured what the actual speed is but Ws2812b is pretty demanding so I'd assume it's faster than 40kHz (my expectation would be at least 1MHz).

For SoftwareSpi to achieve best results you need to make sure you are using GpioController with RaspberryPi3Driver or LibGpiodDriver, I'm not certain what the actual speed is (raw writing for RaspberryPi3Driver if I remember correctly should be around 500kHz but it's been more than a while since we last measured).

JohnyCage2020, 2021-01-13:
Hey, @krwq thanks for yours answer. I set **core_freq=250** and **core_freq_min=250** but it didn't solved my problem. By saying samples/s I meant how many reads from 12 bit ADC converter I achived. It's propably my missunderstanding of using library. Bellow is my code of hardware spi and software. 

**HardwareSPI** when I am trying to make it like this I achive 25k reads from ADC/s .
```
            var hardwareSpiSettings = new SpiConnectionSettings(0, 0);
            hardwareSpiSettings.ClockFrequency = 2000000;
            hardwareSpiSettings.Mode = SpiMode.Mode0;

            using SpiDevice spi1 = SpiDevice.Create(hardwareSpiSettings);
             using var mcp = new Mcp3208(softwareSPi);
while(true)
{
var msg = mcp.Read(0);

}

```
**SoftwareSPI** it's allows me to get 40k reads from ADC/s

```
            var softwareSpiSetting = new SpiConnectionSettings(0, 0);
            softwareSpiSetting.ClockFrequency = 2000000;
            SoftwareSpi softwareSPi = new SoftwareSpi(18, 23, 24, 25, softwareSpiSetting);

            using var mcp = new Mcp3208(softwareSPi);

while(true)
{
var msg = mcp.Read(0);

}
```

krwq, 2021-01-13:
@JohnyCage2020 can you share your code for reading? Possibly the hardware is faster if you send a lot of data but doing a lot of many small reads my end up in large overhead from the OS. With SoftwareSpi overhead of writing bulk vs many small operations is more or less the same. Possibly this is something which can be fixed at the ADC implementation (likely would need a new API though)

JohnyCage2020, 2021-01-13:
@krwq yes, there Is my code for hardwareSPI read:
  using var mcp = new Mcp3208(spi1);

```
            Task taskWrite = Task.Run(() =>
                {
                    while (true)
                    {
                        try
                        {
                            var msg = mcp.Read(0);
                           
                            data.Add(msg+"/n");
                        }
                        catch (Exception ex)
                        {
                            Console.WriteLine(ex);
                        }
                    }
                });
```

krwq, 2021-01-13:
I'm not sure we'll be able to do anything about the hardware implementation. The only thing we can perhaps try is to:

Option 1:
- do Initialize once in the constructor and not call it again (that should remove a lock on every operation)
not sure if that will change anything since we have a fast path for `_deviceFileDescriptor >= 0` in there which should be fast enough.

Option 2:
- change hardware implementation to do read/write instead of ioctl for one direction operations
- split mcp code so it does read/write separately (that might theoretically cause issues for some devices)
presumably read/write will be faster than ioctl.

I definitely did not expect the SoftwareSpi to be faster. We should keep this issue open until we figure out if we can make hardware SPI faster somehow

JohnyCage2020, 2021-01-13:
@krwq  I did not expect this too. What do you mean by?

> change hardware implementation to do read/write instead of ioctl for one direction operations
Do I need to Write some bytes after read like bellow ?
`  using var mcp = new Mcp3208(spi1);
 using SpiDevice spi1 = SpiDevice.Create(hardwareSpiSettings);
  var msg=0;
            Task taskWrite = Task.Run(() =>
                {
                    while (true)
                    {
                        try
                        {
                          msg = mcp.Read(0);
                              spi.WriteByte(0x42);
                            data.Add(msg+"/n");
                        }
                        catch (Exception ex)
                        {
                            Console.WriteLine(ex);
                        }
                    }
                });`

pgrawehr, 2021-01-13:
@JohnyCage2020 That was more a hint to maybe change the implementation of the SPI driver. 

It would be interesting to test (maybe using a scope) whether the bad performance comes from operating system overhead - meaning it takes a long time to set everything up to send the data - or because for some reason the clock speed is not properly set.

JohnyCage2020, 2021-01-13:
@pgrawehr @krwq  Sorry for misunderstanding.  I have made some scope measurments of clock speed there are my results:
Asking for 2Mhz
[https://imgur.com/a/ZyGC9IO](url)
I have noticed that software PWM is slower when core_freq is set to 250.

krwq, 2021-01-14:
Thanks @JohnyCage2020! So seems HardwareSpi is as it should be so likely this is delay caused by the OS, the SoftwareSpi is around 2x slower but because of no delay on WriteRead it ends up working faster for normal usage.

We should investigate if we can get rid of that delay somehow

JohnyCage2020, 2021-01-16:
Hello, @krwq I have made some more tests of clock signal and I found some another issue.  It may be some another my misunderstanding but there are my results:

I try to set clock signal to 1Mhz or less  by using software SPI class like this:

```
 var softwareSpiSetting = new SpiConnectionSettings(0, 0);
                        softwareSpiSetting.ClockFrequency = 900000;

                        SoftwareSpi softwareSPi = new SoftwareSpi(18, 23, 24, 25, softwareSpiSetting);
                        using var mcp = new Mcp3208(softwareSPi);
```

I measured it with scope. And as a output I alway's get's clock signal frequency 1.3 Mhz. What is it caused by?

krwq, 2021-01-16:
@JohnyCage2020 Software SPI is currently ignoring speed setting because we expected it to work slower than any lowest setting user might want to set... See the note at: https://github.com/dotnet/iot/blob/master/src/devices/SoftwareSpi/README.md#software-spi

JohnyCage2020, 2021-01-16:
@krwq  Thank you so much 👍