dotnet/iot #1411: "No GPIO controllers exist on this system"

PsBernie, 2021-01-15:
**Describe the bug**

In a C# project, I referenced the "System.Device.Gpio" package, and tried to create a GpioController object or a Windows10Driver object.

**Steps to reproduce**

// put sample code here

       var gpio = new GpioController();

or alternatively

    GpioDriver driver = new Windows10Driver();

**Expected behavior**

I expect such an object to be created.

**Actual behavior**

I get a System.NotSupportedException with the message "No GPIO controllers exist on this system"

**Versions used**

Add following information:

- `dotnet --info` on the machine being used to build (the app is NOT being run here)
.NET SDK (reflecting any global.json):
 Version:   5.0.102
 Commit:    71365b4d42

Runtime Environment:
 OS Name:     Windows
 OS Version:  6.1.7601
 OS Platform: Windows
 RID:         win7-x64
 Base Path:   C:\Program Files\dotnet\sdk\5.0.102\

Host (useful for support):
  Version: 5.0.2
  Commit:  cb5f173b96

.NET SDKs installed:
  5.0.102 [C:\Program Files\dotnet\sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.All 2.1.24 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.App 2.1.24 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.1.11 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 5.0.2 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 2.1.24 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.1.11 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 5.0.2 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.WindowsDesktop.App 3.1.11 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 5.0.2 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]

To install additional .NET runtimes or SDKs:
  https://aka.ms/dotnet-download

- `dotnet --info` on the machine where app is being run (not applicable for self-contained apps)
That executable could not be found on the deployment machine - there is NO folder "C:\Program Files\dotnet". 
Microsoft .Net Framework 4.8 SDK is installed. 
OS: Windows 10 Pro 20H2 x64
“Microsoft Visual C++ Redistributable for Visual Studio 2015, 2017 and 2019” (“VC_redist.x64.exe”)  is installed.
The GPIO device is shown in the device manager ("Intel(R) Serial IO GPIO Host Controller - INT34BB")

- Version of `System.Device.Gpio` package
1.3.0

- Version of `Iot.Device.Bindings` package (not needed if bug is in `System.Device.Gpio`)

Ellerbach, 2021-01-15:
@PsBernie are you running this code on a Raspberry PI with Windows 10 installed? Or on a regular Windows 10 desktop (or server)?
In the second case, the behavior you describe is normal as there is no GPIO controller on such systems.

PsBernie, 2021-01-15:
It is a ULT5 embedded computer (see https://www.ieiworld.com/en/product/model.php?II=690 ), not a Raspberry. It does have a GPIO connector.
The vendor specific drivers are installed. The device is shown in the device manager:
[image]

Ellerbach, 2021-01-15:
I may have missed something from the specifications but it seems that the controller is a serial controller:
[image]
So you could most likely access thru serial.
I may have missed something.

Ellerbach, 2021-01-15:
OK, I see there are 4 output and 4 input IO. But there is nothing in the documentation giving any information regarding the driver.
To access it with .NET IoT, you would have to have a specific wrapper. For that, you'll need the documentation of the driver.

PsBernie, 2021-01-15:
OK, thanks.

Ellerbach, 2021-01-15:
@PsBernie do you have a pointer on the SDK or anything that describes how the diver works? It can be quite simple to create a .NET driver to provide the GPIO functionalities.

PsBernie, 2021-01-15:
Unfortunately, I have not yet found any documentation. Windows Device Manager shows that it uses "iaLPSS2_GPIO2.sys", and that's all I could find out meanwhile...

Ellerbach, 2021-01-17:
@PsBernie I made some research and it seems that the driver is following the GPIO driver requirements: https://docs.microsoft.com/en-us/windows-hardware/drivers/gpio/
Currently, we do support Windows 10 drivers for GPIO/I2C/SPI/PWM when there is an already existing Windows.Devices namespace. so far the support is for Windows 10 IoT systems. 
We should see if it makes sense to support Windows 10 exposing standard drivers as well.

krwq, 2021-01-18:
I'll reopen the issue for tracking this potential work

PsBernie, 2021-01-18:
Thanks a lot for your efforts in this case.
Mean while, I received some samples from the vendor, containing a dll for communicating with the GPIO PINs and a few lines a "C#" code (well, rather C without # code, but it works).

ASalomePM, 2021-03-30:
Please, can you give me that dll?. I have the same problem with the same "Intel Serial IO GPIO Host Controller" and I am lost.

Ellerbach, 2021-03-31:
> Please, can you give me that dll?. I have the same problem with the same "Intel Serial IO GPIO Host Controller" and I am lost.

@PsBernie I guess this message is for you

PsBernie, 2021-04-07:
@ASalomePM If you are referring to the "samples from the vendor" I mentioned above, please contact IEI via the "Service Form" on their web site - I do not think that I can share them here.

pgrawehr, 2021-04-07:
@PsBernie Would be nice if you could clarify with the vendor whether that library can be freely distributed. That would allow us to write a wrapper implementation for it, so that it can be used with this library.

jtdent, 2021-11-09:
So after many headaches. I got this working for the Dell Embedded 5000 pc

> https://www.intel.com/content/dam/support/us/en/documents/boardsandkits/custom-solutions-header-whitepaper.pdf

I found a usefull open library class called Ring0. Which allows hardware access to the I/O Ports. 

I followed the Linux guide and the commands they sent to the Super-IO and from that got the GPIO pins to work.
>https://github.com/GermanAizek/WinRing0
Is an example of the library. It can be called in c++ and c#. 

I coded mine to work in c#

and used MyOls.WriteIoPortByte and MyOls.ReadIoPortByte instead of INb and OUTb (linux code)

pgrawehr, 2021-11-10:
@jtdent Great, that sounds like a doable way. If you want, you can create an implementation of GpioDriver that wraps this library.

jtdent, 2021-11-10:
any previous examples of this. and I will have a look.

Nickellukas2001, 2021-11-24:
Hey,
so i have the same Problem and i read the comments above and found out that it seems like i cant run the system.device.gpio 
on a raspberry pi with raspbian and mono. Am i right with this?
If i´m right can anyone tell me how to use the gpios with raspbian and mono? I really would like to run an win forms app on the pi.
Would be really great to get help here.
Greetings Lukas

Ellerbach, 2021-11-25:
> so i have the same Problem and i read the comments above and found out that it seems like i cant run the system.device.gpio
> on a raspberry pi with raspbian and mono. Am i right with this?

Hi, can you be more precise? Which hardware are you running on?
If it's a Raspberry PI 2+ running Raspbian, you don't need mono or anything else. .NET 5.0/.NET 6.0 works perfectly, you'll get the drivers recognize, all good.
This thread is about non standard hardware running on Windows with some GPIO capacities.

Nickellukas2001, 2021-11-25:
Hi, so im using the raspberry pi 4b. Are you talking about running the raspberry PI with Windows 10? Because i read that Win10 on the pi is not really performing very good and needs some time to do tasks...

Ellerbach, 2021-11-25:
> Hi, so im using the raspberry pi 4b. Are you talking about running the raspberry PI with Windows 10? Because i read that Win10 on the pi is not really performing very good and needs some time to do tasks...

Raspbian on the RPI4 is the way to go to get proper support for all those GPIO libraries. And note there is no official support for Windows 10 on the RPI4.

Nickellukas2001, 2021-11-25:
Yeah I thought so too, but how do I get access to the gpios or how do I use them in c# ? That's my problem rn

Ellerbach, 2021-11-25:
> Yeah I thought so too, but how do I get access to the gpios or how do I use them in c# ? That's my problem rn

This repository is al about that :-) Check the blinky example: https://github.com/dotnet/iot/tree/main/samples/led-blink

Nickellukas2001, 2021-11-25:
Thanks, I already followed the blinky example. I'm using mono on my PI because I want to run a windows forms application. Can I run the win forms app without mono and still have the User Interface of my app?

Ellerbach, 2021-11-25:
> Thanks, I already followed the blinky example. I'm using mono on my PI because I want to run a windows forms application. Can I run the win forms app without mono and still have the User Interface of my app?

non, you cannot run a Windows form using .NET 5.0 or .NET 6.0 on Linux. So correct, you have to go for mono in this case.

Nickellukas2001, 2021-11-25:
Okay,but the repository is incompatible with Mono am I right? If it is, do you have an idea how to get access to the gpios with c# in mono on the raspberry PI? 
Thanks

Ellerbach, 2021-11-25:
> Okay,but the repository is incompatible with Mono am I right? If it is, do you have an idea how to get access to the gpios with c# in mono on the raspberry PI? Thanks

it's compatible. There is a .NET standard implementation. Both for the System.Device.Gpio and the Iot.Device bindings. Use the latest preview releases.

Nickellukas2001, 2021-11-25:
Hmmm okay, I tried the blinky tutorial and run it on my PI but nothing happened and my program just closed... So you say I just have to use the latest preview release of system.device.gpio and run it with mono on my PI and it should work? Because I couldn't manage to get voltage out of an gpio... I am really sorry for my questions but I'm a beginner...

Ellerbach, 2021-11-25:
> Hmmm okay, I tried the blinky tutorial and run it on my PI but nothing happened and my program just closed... So you say I just have to use the latest preview release of system.device.gpio and run [...]

It works great. Check out this old example I did: https://github.com/Ellerbach/Nabaztag.Net/tree/master/Nabaztag.Server

See the project, it's a .NET 4.8, the System.Device.Gpio library is there, and Gpio are used, some bindings as well. At that moment, they were not .NET Standard 2.0 compatible but it's now the case.

You can build your project on your pc, deploy by copying the build to the pi and using a mono nameofyourprogram.exe on the py

Nickellukas2001, 2021-11-25:
Alright, thank you very much. I'll try this out later.

joperezr, 2021-11-25:
Also, make sure when running you execute the program as sudo. GPIO is very restrictive for non root users given it is so close to hardware, so most of the gpio interaction usually requires sudo

Nickellukas2001, 2021-11-25:
Okay i zried it now but i didnt achieve anything... When I run my application with mono it immediately closes it again. If I change the code to initializing the controller and letting the led blink, it crashes when I push the button. Has anybody an idea of what could be the problem? I'm really lost here rn...

Nickellukas2001, 2021-11-25:
I just tried starting my program with "mono program.exe" and it threw the exception "System.IO.FileNotFoundException: Could not load file or assembly 'System.Device.Gpio'...
Do you know how to get rid of that @Ellerbach ?

pgrawehr, 2021-11-25:
Is the file there? Did you correctly copy all relevant files to the directory containing program.exe?

Nickellukas2001, 2021-11-25:
@pgrawehr sorry I don't understand what exactly you mean... I just took the .exe out of my obj folder and put it on the pi... In Visual Studio the folder should contain the files.

Ellerbach, 2021-11-26:
> @pgrawehr sorry I don't understand what exactly you mean... I just took the .exe out of my obj folder and put it on the pi... In Visual Studio the folder should contain the files.

Make sure you copy everything. Don't miss any file. If the dependencies are not present with the exe, it won't work. I really had absolutely no issue with mono. Used it on a Raspberry Pi Zero (armv6). Standard .NET 4.8 project, added the System.Device.Gpio, build normally, took all the files and copy them into the PI. I'm using WinSCP for the transfert.

Nickellukas2001, 2021-11-26:
Alright, I'll have a look on my files again. Just to be sure: what exactly is meant with building it and taking all the files to the pi? Is there a certain folder I have to move to the pi or where do I know which files I need to transfer?

pgrawehr, 2021-11-26:
You have to copy all the files that are in the "bin" folder (or "bin\debug"). You normally don't ever need to look at the "obj" folder. This one contains intermediate files only and are used by the compiler. You can also use the "publish" option to create a folder that contains all the required dependencies.

Nickellukas2001, 2021-11-26:
Hey, so i copied the whole bin\debug folder to my pi and tried running my application. I just got this Error:
[image]

pgrawehr, 2021-11-26:
For mono to work correctly, you have to install libgpiod. Run the following command: `sudo apt install -y libgpiod-dev`.

Nickellukas2001, 2021-11-26:
Hey, big thank you to all of you, now I can control my gpios with buttons in a win forms app! Thank you very much for the help!

Ellerbach, 2021-11-26:
> Hey, big thank you to all of you, now I can control my gpios with buttons in a win forms app! Thank you very much for the help!

Great! Happy you made it :-) I guess we can keep this thread somewhere hot for anyone who wants to run it on mono!
Let us know about the final result!

Nickellukas2001, 2021-11-27:
Hey, so everything worked good so far and i managed to conttrol the gpios in a test program. But now i tried to include this into my project and changed my code and now i get the error:
"1> C: \ Program Files \ Microsoft Visual Studio \ 2022 \ Community \ MSBuild \ Current \ Bin \ amd64 \ Microsoft.Common.CurrentVersion.targets (4322,5): error MSB4096: The element "C: \ Users \ lukas \ source \ repos \ Ampelsteuerung BSV Schefflenz \ packages \ System.Runtime.WindowsRuntime.4.7.0 \ build \ net461 \ .. \ .. \ ref \ netstandard2.0 \ System.Runtime.WindowsRuntime.dll "in the element list" ReferencePath " does not define a value for the "CopyLocal" metadata. To use this metadata, qualify it by specifying% (ReferencePath.CopyLocal) or make sure that all items in the list define a value for this metadata."
Has anyone ever had that problem? I searched in the inetrnet and didnt find anything...

raffaeler, 2021-11-27:
Any WindowsRuntime dll (or any other dll depending on it) can't work on linux with or without mono.
You should limit your dependencies to just what mono is able to consume and carefully avoiding any Windows related dependency.
Windows Forms are an exception because mono ported the library, but the Windows Runtime is a sandboxing environment with specific Windows APIs that does not exist outside Windows.

Nickellukas2001, 2021-11-27:
But im trying to build on windows 10 in visual studio and not in Linux, so there is no mono. And I didnt find a working solution for this error...

raffaeler, 2021-11-27:
I am quite confused. You opened this thread about mono and GPIO. If you are asking an unrelated question, please open another thread, but not here given this repository is just about the dotnet/iot libraries.

If your question is about building a .NET project, try posting in this repo: https://github.com/dotnet/msbuild but be more specific, and give more details on the refrences you are adding.

Nickellukas2001, 2021-11-27:
Okay I'm sorry for that... Didn't think very much about it.

raffaeler, 2021-11-27:
No problem, you will certainly find more useful info there.
BTW, if your original question is fully answered, please close the comment.
Thank you

raffaeler, 2021-11-30:
OMG, I didn't realize that, sorry :(

krwq, 2022-06-30:
[Triage] Duplicate of https://github.com/dotnet/iot/issues/1878

closed 2022-06-30 by krwq