dotnet/iot #1415: Raspberry Pi Pico as a potential target

robertsundstrom, 2021-01-21:
Would it be possible to target [Raspberry Pi Pico](https://www.raspberrypi.org/products/raspberry-pi-pico/)?

See my discussion: https://github.com/dotnet/runtime/discussions/47270

krwq, 2021-01-21:
Hey @robertsundstrom, did you check if you can get it to work with Mono or nano-framework? If any of those two work we will likely be able to give you some pointers

pgrawehr, 2021-01-21:
What likely will work is put a Firmata Firmware on it and access it using the Arduino binding. But of course that's not quite the same as running .NET code on the MCU itself.

krwq, 2021-01-21:
nano-framework should work too I think but it's not straightforward yet how to get .NET IoT run on it

Ellerbach, 2021-01-22:
+1 for .NET nanoFramework which is a more reasonable target for such a processor. Check it out https://github.com/nanoframework/Home. There are already some contacts to see how to make that happening!

Ellerbach, 2021-02-18:
[Triage] The .NET nanoFramework community is working on a port to have .NET nanoFramework supported. This will take a little time.
The other element is that as soon as there will be a support for Arduino IDE, you'll be able to upload a Firmata firmware that will allow you to be used with .NET IoT from your Windows machine for example. This support is coming soon on our side, see #1039.

nh43de, 2021-02-26:
The only issue is that nanoFramework doesn't support .NET standard libs (yet?) as far as I'm aware. Makes sharing code between platforms a little more tricky, and all the IoT libs here unusable. NanoFramework has it's own specialized packages.

pgrawehr, 2021-02-26:
@nh43de Yes, that's absolutelly a limitation. We'll certainly be trying to support the Raspberry Pico trough our new Firmata binding. Unfortunatelly, there's currently no support to build for the Pico using the Arduino IDE (yet), which will cause an unknown extra effort in building that library. I also don't have the hardware so far. But please don't expect that you'll be able to run a full framework on the Pico, it's hardware (especially the memory) is to limited for that.

Ellerbach, 2021-02-26:
@nh43de it has the possibility to use .NET standard as project reference, then everything needs to be compiled for nanoFramework. mscorlib is a small subset of the full framework. With minimum numbers of methods.
You can check an example project using .NET Standard project reference between a reach .NET code base and a nano based one in the nanoFramework Unit Test Framework: https://github.com/nanoframework/nanoFramework.TestFramework
And yes, nanoFramewok has specific packages.

Mallon94, 2021-04-21:
> @nh43de Yes, that's absolutelly a limitation. We'll certainly be trying to support the Raspberry Pico trough our new Firmata binding. Unfortunatelly, there's currently no support to build for the [...]

Now that the Pico does have Arduino IDE support, can we expect some firmata capabilities?

pgrawehr, 2021-04-21:
@Mallon94 As far as I know, there are only projects in Beta state right now (or are you looking at a specific toolkit)? I have not tested any of those yet, for lack of time. 

In theory, it should be easy: Check out https://github.com/pgrawehr/ConfigurableFirmata (branch develop!), and update the file "Boards.h" to include the hardware configuration for the pico.

Mallon94, 2021-04-21:
@pgrawehr Thanks for the advice. I had a quick look at using configurablefirmata and updating the Boards.h file, however it still would not recognise the board everytime I tried to compile. It appeared like the name was wrong in the Boards.h file but i was using the correct build identifer from the boards.txt pico file. 

I found there is something in works http://abyz.me.uk/picod/index.html that looks like a nice firmata alternative.

pgrawehr, 2021-04-21:
@Mallon94 Can you tell me which pico-for-arduino library you tried? I might give it a go if you want (I do not have the hardware, but maybe at least I could get it to compile)

Mallon94, 2021-04-21:
@pgrawehr that would be great! Im using this https://github.com/earlephilhower/arduino-pico and using the standard board, not a variant.

pgrawehr, 2021-04-21:
@Mallon94 I have created a branch "Pico" that builds. Since the Firmata protocol doesn't currently support pin selections, I have hardcoded that the I2C bus is on GPIO pins 4 and 5 (if enabled) and SPI is on GPIO pins 6-9. As said, I do not have the hardware, so it would be nice if you could do some testing.

Mallon94, 2021-04-22:
Thanks for your time @pgrawehr I appreciate it. I tried that branch,  but it was having compile errors. Did you hardcode the SPI using SetTX and SetRX? 
`Arduino/libraries/ConfigurableFirmata/src/SpiFirmata.h:296:7: error: 'class arduino::MbedSPI' has no member named 'setTX'
   SPI.setTX(7);
       ^~~~~`
Seems to happen with MBed12C aswell

pgrawehr, 2021-04-22:
It seems you have a different board setting or something like that. If I purposedly missspell that call, the error message is `C:\Users\....\Arduino\libraries\ConfigurableFirmata\src/SpiFirmata.h:296:7: error: 'class SPIClassRP2040' has no member named 'set2TX'; did you mean 'setTX'?` 

Not sure why the type is `MbedSPI` for you, but `SPIClassRP2040` for me. I am using Version 1.1.0 of the Pico library and the board is set to "Raspberry Pi Pico". The code [https://github.com/earlephilhower/arduino-pico/blob/master/libraries/SPI/SPI.h](here) clearly suggests that `SPIClassRP2040` should be used. It seems that for whatever reason, the system finds another SPI library on your system. 

Look for lines such as the following in the output:
```
Alternatives for Wire.h: [Wire@1.0]
ResolveLibrary(Wire.h)
  -> candidates: [Wire@1.0]
```
and 
```
Alternatives for SPI.h: [SPI@1.0]
ResolveLibrary(SPI.h)
  -> candidates: [SPI@1.0]
```

Does it list more than one candidate?

Mallon94, 2021-04-22:
I fixed it, was defaulting to a different board i had previously installed which was a Mbed OS RP2040 board. Got rid of it and it used 1.1.0 of the Pico lib. DHT was causing a timeout and preventing build so i disabled it and it built! Will get testing this now.

mattjlewis, 2021-06-30:
FYI I updated the Arduino Firmata Boards.h file to include this:
```
// Raspberry Pi Pico
#elif defined(TARGET_RP2040) || defined(TARGET_RASPBERRY_PI_PICO)
#define TOTAL_ANALOG_PINS       3
#define TOTAL_PINS              29
#define VERSION_BLINK_PIN       LED_BUILTIN
#define IS_PIN_DIGITAL(p)       ((p) >= 0 && (p) < 26)
#define IS_PIN_ANALOG(p)        ((p) >= 26 && (p) < 26 + TOTAL_ANALOG_PINS)
#define IS_PIN_PWM(p)           digitalPinHasPWM(p)
#define IS_PIN_SERVO(p)         IS_PIN_DIGITAL(p)
#define IS_PIN_I2C(p)           0
#define IS_PIN_SPI(p)           0
#define PIN_TO_DIGITAL(p)       (p)
#define PIN_TO_ANALOG(p)        ((p) - 26)
#define PIN_TO_PWM(p)           PIN_TO_DIGITAL(p)
#define PIN_TO_SERVO(p)         (p)
```
and so far, so good - I can retrieve the board capabilities and blink an LED. I've not done this before so not sure if the above is entirely correct - I've ignored I2C and SPI for now.

mattjlewis, 2021-07-24:
FYI I confirm that I2C works, raised this issue in ConfigurableFirmata: https://github.com/firmata/ConfigurableFirmata/issues/115. I'll retest against the latest codebase (I had used the 2.10.1 release from 2018 and had to modify a couple of files) and submit a pull request instead.

krwq, 2025-07-31:
[Triage] Closing because it had no activity in the last 4 years and we have firmata implementation here as well

closed 2025-07-31 by krwq