dotnet/iot #1419: System.IO.IOException: Error 13 initializing the Gpio driver. Ubuntu 20.04 Arm64 on RPi4

scegg, 2021-01-26:
**Describe the bug**

Unhandled exception. System.IO.IOException: Error 13 initializing the Gpio driver.
   at System.Device.Gpio.Drivers.RaspberryPi3LinuxDriver.Initialize()
   at System.Device.Gpio.Drivers.RaspberryPi3LinuxDriver.OpenPin(Int32 pinNumber)
   at System.Device.Gpio.Drivers.RaspberryPi3Driver.OpenPin(Int32 pinNumber)
   at System.Device.Gpio.GpioController.OpenPinCore(Int32 pinNumber)
   at System.Device.Gpio.GpioController.OpenPin(Int32 pinNumber)
   at System.Device.Gpio.GpioController.OpenPin(Int32 pinNumber, PinMode mode)

**Steps to reproduce**

```csharp
using GpioController gpioController = new GpioController();
gpioController.OpenPin(18, PinMode.Output);
```

**Expected behavior**

Pin 18 is opened.

**Actual behavior**

Exception thrown.

**Versions used**

Dotnet version: 3.1.11 (f5eceb8105), 5.0.2 (cb5f173b96)
Gpio: 1.3.0

pgrawehr, 2021-01-27:
Possible that this is by design. Raspbian has a set of udev rules that by default enable access to GPIO / I2C / etc for non-root users. It is possible that these are missing in ubuntu. The file in question is `99-com.rules`, you'll find it [here](https://github.com/RPi-Distro/raspberrypi-sys-mods/tree/master/etc.armhf/udev/rules.d). I don't know whether that can be installed directly on ubuntu, though.

scegg, 2021-01-27:
Thanks for your clarification.

closed 2021-01-27 by scegg