dotnet/iot #1423: WriteGPIO and ReadGPIO on Pn532 are not implemented as Pn532 specs define

rodrigobengoechea, 2021-01-28:
**Describe the bug**

I've been trying to use the WriteGPIO and ReadGPIO methods defined on Pn532 class and wouldn't be able to modify the values of Port7. Doing some research on how WriteGPIO and ReadGPIO are implemented I found 2 issues on the write method and 1 issue on the read method.

WriteGPIO method:
The write gpio message should be sent as the following = [Command_Identifier, Port3_Value, Port7_Value] where:
Command_Identifier = 0x0E (Write Gpio Command)

If we are going to modify values on Port3 then bit 7 on this value should be 1, otherwise bit 7 should be 0. Same logic applies to Port7.

If you take a look at how WriteGPIO method is implemented, this bit modification is never done so bit 7 for any port is set to 0 everytime and no modification on Ports values is done by the board.

In addition the message is being built with the following structure: [Command_Identifier, Port7_Value, Port3_Value] which is not how is defined by the specifications of Pn532.

ReadGPIO method:
Response for Read GPIO Command on Pn532 should be interpreted as the following: [Port3_Value, Port7_Value, Operating_Mode] as it's stated on Pn532 specifications.

Implementation on this method interprets response as the following [Port7_Value, Port3_Value, Operating_Mode]

Pn532 specifications: https://www.nxp.com/docs/en/user-guide/141520.pdf. 
Write Gpio => page 81
Read Gpio => page 79

**Steps to reproduce**

```
var pn = new Pn532("/dev/ttyO4");
pn.SetSerialBaudRate(BaudRate.B0115200);
pn.WakeUp();
            
var readGpio = pn.ReadGpio(out var p7, out var p3, out var operatingMode);
var writeGpio = pn.WriteGpio((Port7)0, (Port3)0);
readGpio = pn.ReadGpio(out p7, out p3, out operatingMode);
```
**Expected behavior**

p7 and p3 values should be 0

**Actual behavior**

p3 and p7 values are not 0

**Versions used**

Add following information:

- `dotnet --info` on the machine being used to build
.NET SDK (reflecting any global.json):
 Version:   5.0.101
 Commit:    d05174dc5a

Runtime Environment:
 OS Name:     Windows
 OS Version:  10.0.18363
 OS Platform: Windows
 RID:         win10-x64
 Base Path:   C:\Program Files\dotnet\sdk\5.0.101\

Host (useful for support):
  Version: 5.0.1
  Commit:  b02e13abab

.NET SDKs installed:
  2.1.509 [C:\Program Files\dotnet\sdk]
  2.1.603 [C:\Program Files\dotnet\sdk]
  2.2.203 [C:\Program Files\dotnet\sdk]
  3.0.103 [C:\Program Files\dotnet\sdk]
  5.0.101 [C:\Program Files\dotnet\sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.All 2.1.10 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.13 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.23 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.2.4 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.App 2.1.10 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.13 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.23 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.2.4 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.0.3 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.1.10 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 5.0.1 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 2.1.10 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.13 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.23 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.2.4 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.0.3 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.1.10 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 5.0.1 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.WindowsDesktop.App 3.0.3 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 3.1.10 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 5.0.1 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]

To install additional .NET runtimes or SDKs:
  https://aka.ms/dotnet-download

- Version of `System.Device.Gpio` package 1.3.0
- Version of `Iot.Device.Bindings` package 1.3.0

Ellerbach, 2021-01-29:
Thanks for the detailed submission on this bug. If you found the correct way of fixing this issue, would you mind submitting a PR?
As you can test this that would be perfect.

rodrigobengoechea, 2021-01-29:
Sure! i'll try to do it during today. =D

rodrigobengoechea, 2021-02-01:
Hey @Ellerbach I've got the fix ready to deliver but everytime I try to push changes to remote something related with authentication fails.

> Failed to push to the remote repository. See the Output window for more details.
> Switched to branch 'fixWriteGPIOOnPn532'
> Checked out fixWriteGPIOOnPn532
> Pushing fixWriteGPIOOnPn532
> Error encountered while pushing branch to the remote repository: Git failed with a fatal error.
> unable to access 'https://github.com/dotnet/iot/': The requested URL returned error: 403
> Pushing to https://github.com/dotnet/iot
> 
> Failed to push the branch to the remote repository. See the Output window for more details.

Ellerbach, 2021-02-02:
@robinmanuelthiel that's normal :-) It's because you can't push anything directly to this repository. Here are the steps for you to be able to submit a PR:

- step 1: clone the repository in your own GitHub. Click the Fork button on the top right of the page and select your own repository
[image]
- step 2: clone your own repository on your hard drive
  - alternative: you can add this new origin to the already cloned dotnet/iot:
    `git remote add myiot https://guthub.com/robrigobengoechea/iot.git`
    In this case, you'll have the possibility to both work on your own repository and pull the latest versions of the main dotnet/iot repo
- step 3: (more than recommended not to work on the main branch) create a new branch on your own repository
- step 4: make the changes on this newly created branch on your own repository
- step 5: push those changes on your own repository
  - tip: if you're not comfortable with command lines, you can do steps 3 to 5 with VS Code for example. Navigate to the git ico on the left, the menu has different options, like `push to...` and `branch` where you can do all those manipulation very easilly.
  - tip: once you've done the changes, you'll need to push to your own repository called `myiot` in our case
- step 6: PR time! You can now go to https://github.com/robrigobengoechea/iot and you'll see that you're offered to make a PR. Just follow the steps and voilà!

Please let me know if you need any other assistance. Don't hesitate to ask for help as well in the Discord channel.

PR #1434 (2021-02-04): Fixes #1423
Fixes #1423 :
Fixing methods WriteGPIO and ReadGPIO on Pn532 device. Now methods are implemented according Pn532 specifications.

###### Microsoft Reviewers: [Open in CodeFlow](https://portal.fabricbot.ms/api/codeflow?pullrequest=https://github.com/dotnet/iot/pull/1434)
commit eeecc887 (2021-02-04): Fixes #1423 (#1434)

* fixing issue #1423
closed 2021-02-04 by Ellerbach