dotnet/iot #1430: Error message functionality for Iot.Device.SocketCan.CanRaw

GKooijmans, 2021-01-31:
The Linux SocketCan module supports Error Message Frames. These frames offer a mechanism for userspace applications to be notified when the state of the CAN Bus changes. By default these are filtered out. They can be enabled by a setsockopt call.
The current implementation of the Raw CAN socket in Iot.Device.SocketCan.CanRaw contains a method for setting a regular Message Filter of type CAN_RAW_FILTER, but not one for setting an Error Message filter (CAN_RAW_ERR_FILTER).
If it were to be added, it would give an application a lot more control over the underlying interface.

krwq, 2021-02-01:
@GKooijmans what APIs would you suggest?

GKooijmans, 2021-02-01:
@krwq Thanks for replying. Perhaps something like:

void FilterErrorMsgs(CanErrors errorfilter);

 public enum CanErrors : uint
    {
        CAN_ERR_TX_TIMEOUT = 0x00000001U,
        CAN_ERR_LOSTARB = 0x00000002U,
        (complete defs at https://github.com/torvalds/linux/blob/master/include/uapi/linux/can/error.h)
     }

GKooijmans, 2021-02-01:
Also the CanRaw.TryReadFrame method may need some changes. Because an error frame will have CanFrame.IsValid set to false 
And that case the method does not return any data. However an Error Frame does hold data with information regarding the Error that occurred. You could also check for CanId.Error which will be set to true in these frames.

krwq, 2021-02-03:
@GKooijmans let's wait until https://github.com/dotnet/iot/pull/1277 is merged before we do any work here.

What about removing https://github.com/dotnet/iot/blob/main/src/devices/SocketCan/CanRaw.cs#L92-L98 and returning `frame.IsValid`. We would still need to somehow make sure the [assert below](https://github.com/dotnet/iot/blob/main/src/devices/SocketCan/CanRaw.cs#L101) is not hit.

Are the error frames currently completely filtered out? (don't have a setup closeby and can't remember from the top of my haead) I'd personally simplify this API a bit:

```csharp
public bool SkipErrorFrames { get; set; } // getter and setter would do a bit more, this is just an API shape
```

do we actually care about type of error? If yes then perhaps:

```csharp
public SocketCanErrorFiltering ErrorFiltering { get; set; }

public enum SocketCanErrorFiltering
{
  NoErrorFrames,
  AllErrorFrames,
  // the other two options with user friendly name
}
```

GKooijmans, 2021-02-04:
Yes, @krwq it makes sense to wait for #1277 indeed.
Returning frame.isValid makes sense too because it's nothing to act on anymore in this method.
The assert you mention will not be hit  because an error frame would still have the framelength properly match the length of the data array. In that sense an error frame is just like any other frame, although their data length is always 8 I believe.
The error property can be used by an application in combination with the isvalid flag to differentiate between an error message that needs special handling, an application message, or simply an invalid message that should be discarded.

Right now error frames are completely filtered out. This is the default mode in SocketCan. You have to 'unfilter' to get them. There are 9 types of error frames you can filter for. Personally I'm not interrested in all of them becuase it'd be too much overhead for my application to handle. Some of these can flood easily under certain conditions.
So your seond approach would work for me, but then with 9 enum (flagged) options instead of 2. Or a changed or overloaded Filter method would work equally well. A method would be more in line with the rest of the class than a property because you already have the regular filter method. Also it wouldn't hide the fact that the interop call downstream is still a method.

krwq, 2021-02-05:
@GKooijmans if there are 9 types we should not pick 2 arbitrarily and support either all of them or have a binary answer.

The buffer length is always 8 bytes but the actual data length is determined by the field inside of the CAN frame (technically you could add some extra info past the data if data is less than 8 - currently we ignore that info).

We also should determine if TryReadFrame should return true or false for erronous frames after you set filtering to allow for errors. Perhaps we should have another method just "ReadFrame" which allows for any frame to be read regardless of error (perhaps with the bool flag "allowErrorFrames"). Maybe it might make sense to only allow error frames per single operation rather than having separate property, so the API would look like:

```csharp
public void ReadFrame(Span<byte> data, out int frameLength, out CanId id, bool throwOnErrorFrames = true)
```

but we should figure out what the behavior would be when we set `throwOnErrorFrames=false` and we don't read anything.

Out of curiosity can you tell what do you need error frames for? Diagnostics?

GKooijmans, 2021-02-05:
@krwq  I agree, we should support all of them.
To know whether TryReadFrame should return True or False in case of an error frame depends on what we define as a valid frame.
An error message frame is a valid frame in the strict sense of the word as it adheres to all the limitations and specs of the CAN protocoll. Its data length is equal to the dlc value. Its ID is a valid 29 bit identifier, etc.
However it's not meant to be processed by an application as part of its own higher level protocoll as the application could confuse the ID of the error message with an ID it's using itself. Hence the Error flag is really important.
So whether or not a Try to read a frame is succesfull and should be answered with a true or false result depends on what you expect to read. 
If you set an Error Message filter, you want to receive them. I'd say this should still result in a True. If you don't set a filter, you won't get any such frames, and the method will never return False. 
So these arguments are actually all in favor of your suggestion for
`public void ReadFrame(Span<byte> data, out int frameLength, out CanId id, bool throwOnErrorFrames = true)`

It makes sense to me. You have to catch Exceptions for this method anyway because you may get an IOException from the Interop for example.

So why do I need the error frames? Well a couple of reasons for this:
- I want my application to know when there's no one receiving the messages I send. The system I'm working on has a Bus system that is connected with cables that can be plugged and unplugged. I know when all cables are unplugged, by monitoring whenever the CAN Bus goes into the state TX Error Passive. Unfortunately with CAN there's no carrier detection mechanism like with ethernet so you don't really know what the link state is, untill you start to transmit.
- I want to be able to cancel the messages I send when the cable is not connected. Because the message contains an instruction that is to be executed right now, not 24 hours later when someone happens to plug the cable back in. I do this by performing a reset on the interface with the IP utility as soon as the CAN Bus goes into state TX Error Passive.
- I flash the link indicator LED's from userspace, because I haven't had the time yet to figure out how to hook up the MCP251x driver to the kernel's LED subsystem :)