dotnet/iot #1431: Work with hardware manufacturers on IoT.Device.Bindings.

vsfeedback, 2021-02-01:
_This issue has been moved from [a ticket on Developer Community](https://developercommunity2.visualstudio.com/t/Work-with-hardware-manufacturers-on-IoT/1312283)._

---
This applies to a number of interfaces: 
https://docs.microsoft.com/en-gb/dotnet/iot/tutorials/temp-sensor
describes how to get e.g. a Raspberry Pi to talk to e.g. a bme280

It would be so much easier to have an I2C interface card on the dev workstation, and write / test code there, only deploying to the Pi when its is baked.

There are I2C interface cards around ($400+ and poorly supported) There are also USB-I2C dongles, but MS could put some pressure on hardware manufacturers to make our lives easier.

Yes, I know about LattePanda, but I want a PCI card for my workstation, not another PC.

I come from the embedded processor world, my first workstation was an Intel ICE

---
### Original Comments

#### Feedback Bot on 1/17/2021, 11:56 PM: 

Thank you for taking the time to provide your suggestion.  We will do some preliminary checks to make sure we can proceed further.  We'll provide an update once the issue has been triaged by the product team.

Ellerbach, 2021-02-02:
@vsfeedback and @terrajobst , there are cheap and perfectly working USB dongle based out of FT4222 chips offering I2C, SPI and GPIO. See in our [repository here](https://github.com/dotnet/iot/tree/main/src/devices/Ft4222), you can buy some in multiple places like [BitWiazrd](http://bitwizard.nl/shop/FT4222h-Breakout-Board?search=ft4222), in this case it cost 13€.

We've been working with FTDI which has created the FT4222 chip to offers the best support for our Windows, Mac and Linux desktop developers and accessing the best of development experience. You can have multiple ones as well.

I do **all** my developments, including new bindings, with an FT4222 dongle. At the very end, I do deploy on a device like a Raspberry Pi to make sure all is working and for the production deployment.

pgrawehr, 2021-02-02:
We'll also (hopefully soon) be providing an Arduino/Firmata binding, which allows to use an Arduino Nano or Uno (around $20) as an USB dongle, providing Gpio, Spi, I2c and PWM.

Ellerbach, 2021-02-18:
[Triage] As we have cheap solutions, we feel that this is being worked on. Please feel free to reopen if there is something specific.

closed 2021-02-18 by Ellerbach