dotnet/iot #1441: Large4Digit7SegmentDisplay does not drive decimal points when connected to AdaFruit 0.56" 7 Seg Display

PJTewkesbury, 2021-02-12:
**Bug Description**

I have connected a ([AdaFruit 0.56" 4 digit, 7 segment ](https://learn.adafruit.com/adafruit-led-backpack/0-dot-56-seven-segment-backpack) ) display using I2C bus to a Raspberry PI. The digits display correctly, but I cannot control the decimal points on any of the digits. The centre ':' can be controlled.

**Steps to reproduce**

```csharp
I2cConnectionSettings LedDisplaySettings = new I2cConnectionSettings(1, 0x70);
I2cDevice LedDisplayDevice = I2cDevice.Create(LedDisplaySettings);
Large4Digit7SegmentDisplay d = new Large4Digit7SegmentDisplay(LedDisplayDevice);
d.Brightness = 1;
d.BlinkRate = BlinkRate.Off;
d.DisplayOn = true;

string sTime = dt.ToString("12:15");
Segment[] s = new Segment[display.NumberOfDigits];
for (int i = 0; i < display.NumberOfDigits; i++)
{
    s[i] = GetSegmentForNumber(sTime[i] - 48);
    s[i] = s[i] | Segment.Dot;    
}
d.Dots=Dot.CenterColon;
ReadOnlySpan<Segment> span = new ReadOnlySpan<Segment>(s);
d.Write(span, 0);
```

**Expected behavior**
I expect the following output
1.2.:1.5.

**Actual behavior**
I always get 
12:15

Looking at the code, I have seen that there is the following in the class Large4Digit7SegmentDisplay
``` csharp
private const byte SegmentMask = 127; 
```
but this constant is never referenced. Also in the method 
``` csharp
public override void Write(ReadOnlySpan<byte> digits, int startAddress = 0)
        {
            if (digits.Length != 0)
            {
                if (startAddress < 0 || startAddress >= 4)
                {
                    throw new ArgumentOutOfRangeException("startAddress");
                }

                if (digits.Length + startAddress > 4)
                {
                    throw new ArgumentOutOfRangeException(string.Format("{0} only supports {1} digits starting from address {2}", "Large4Digit7SegmentDisplay", 4 - startAddress, startAddress));
                }

                ReadOnlySpan<byte> readOnlySpan = digits;
                foreach (byte b in readOnlySpan)
                {
                    _displayBuffer[(int)s_digitAddressList[startAddress++]] = (byte)(b & 0x7f); // <<<------- HERE 
                }

                AutoFlush();
            }
        }
```
In this method, when the _displayBuffer is being filled, the values are being masked by a hard coded value of 0x7f. Should this value be hard coded or reference the const already defined. Also Should the const value be a constant or should the caller be allowed to change this value to allow the code to correctly drive the AdaFruit 0.56" 4 Digi 7 Segment display?

**Versions used**
I am using Iot.Device.Bindings V1.3.0

Add following information:

- `dotnet --info` on the machine being used to build
- `dotnet --info` on the machine where app is being run (not applicable for self-contained apps)
- .NET SDK (reflecting any global.json):
 Version:   5.0.102
 Commit:    71365b4d42

- Runtime Environment:
 OS Name:     raspbian
 OS Version:  10
 OS Platform: Linux
 RID:         linux-arm
 Base Path:   /usr/share/dotnet/sdk/5.0.102/

- Host (useful for support):
  Version: 5.0.2
  Commit:  cb5f173b96

- .NET SDKs installed:
  5.0.102 [/usr/share/dotnet/sdk]

- .NET runtimes installed:
  Microsoft.AspNetCore.App 5.0.2 [/usr/share/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 5.0.2 [/usr/share/dotnet/shared/Microsoft.NETCore.App]

- To install additional .NET runtimes or SDKs:
  https://aka.ms/dotnet-download

- Version of `System.Device.Gpio` package = 1.3.0
- Version of `Iot.Device.Bindings` package = 1.3.0

Ellerbach, 2021-02-16:
Thanks @PJTewkesbury  for submitting this issue. I looked at the code and you're right, it seems that the mask is always apply.
So this should be fixed. You are right, the SegmentMask can be removed. So it will allow the scenario you describe.
If you are willing to submit a PR, that would be great. Just let us know so someone else won't have to make the change.

Ellerbach, 2021-02-19:
Will be fixed thru #1392.

PJTewkesbury, 2021-02-25:
I have created a pull request #1459 which fixes the issue.

Ellerbach, 2021-03-25:
[Triage] Fixed with the #1392

closed 2021-03-25 by Ellerbach