dotnet/iot #1445: The islastread successful of dht22 is always false. Even if the correct temperature is read, this is always false?

bbhxwl, 2021-02-19:
The islastread successful of dht22 is always false. Even if the correct temperature is read, this is always false?

Ellerbach, 2021-02-20:
You should follow the pattern by reading first and checking if the read is correct:

```csharp
        var temp = dht.Temperature;
        var hum = dht.Humidity;
        // You can only display temperature and humidity if the read is successful otherwise, this will raise an exception as
        // both temperature and humidity are NAN
        if (dht.IsLastReadSuccessful)
        {
            Console.WriteLine($"Temperature: {temp.DegreesCelsius}\u00B0C, Relative humidity: {hum.Percent}%");
        }
        else
        {
            Console.WriteLine("Error reading DHT sensor");
        }

        // You must wait some time before trying to read the next value
        Thread.Sleep(2000);
```

bbhxwl, 2021-02-20:
> You should follow the pattern by reading first and checking if the read is correct:
> 
> ```cs
>         var temp = dht.Temperature;
>         var hum = dht.Humidity;
>         // You can only display temperature and humidity if the read is successful otherwise, this will raise an exception as
>         // both temperature and humidity are NAN
>         if (dht.IsLastReadSuccessful)
>         {
>             Console.WriteLine($"Temperature: {temp.DegreesCelsius}\u00B0C, Relative humidity: {hum.Percent}%");
>         }
>         else
>         {
>             Console.WriteLine("Error reading DHT sensor");
>         }
> 
>         // You must wait some time before trying to read the next value
>         Thread.Sleep(2000);
> ```

   try
            {
               
                using (Iot.Device.DHTxx.Dht22 dht = new Iot.Device.DHTxx.Dht22(26, PinNumberingScheme.Logical))
                {
                    while (true)
                    {
                        if (dht.IsLastReadSuccessful)
                        {
                            Console.WriteLine($"Temperature: {dht.Temperature.DegreesCelsius} ℃, Humidity: { dht.Humidity.ToString("0.0")} %");

                        }
                        Thread.Sleep(1000);
                    }
                }
            }
            catch (Exception ex)
            {
                Console.WriteLine(ex.Message);
            }

Of course I waited, but islastread successful is always false, like a bug

pgrawehr, 2021-02-21:
Your code does not even attempt to read from the sensor. Use the pattern given above:

```
using (Iot.Device.DHTxx.Dht22 dht = new Iot.Device.DHTxx.Dht22(26, PinNumberingScheme.Logical)); // Create the sensor binding.
var temp = dht.Temperature; // Try reading the temperature
var hum = dht.Humidity; // Try reading the humidity
// You can only display temperature and humidity if the read is successful otherwise, this will raise an exception as both temperature and humidity are NAN
if (dht.IsLastReadSuccessful)
{
     Console.WriteLine($"Temperature: {temp.DegreesCelsius}\u00B0C, Relative humidity: {hum.Percent}%");
}
else
{
     Console.WriteLine("Error reading DHT sensor");
}

// You must wait some time before trying to read the next value
Thread.Sleep(2500);
```

So first the properties `Temperature` and `Humidity` are read, and then we check whether the reading was valid. But if you don't read, the validity will never change. Note also that you might need retries. Not every read is valid, so you cannot wait indefinitely but should try again after around 2.5 seconds.

krwq, 2021-02-22:
@pgrawehr @Ellerbach I think the APIs might not be exactly intuitive. It's not really obvious when reading happens without reading the implementation code.

pgrawehr, 2021-02-22:
@krwq I agree, I think this was discussed a few times, but so far nobody wanted to do the breaking change of converting the properties to TryGet methods.

bbhxwl, 2021-02-22:
> Your code does not even attempt to read from the sensor. Use the pattern given above:
>
> ```
> using (Iot.Device.DHTxx.Dht22 dht = new Iot.Device.DHTxx.Dht22(26, PinNumberingScheme.Logical)); // Create [...]

pi@raspberrypi:~/dht $ dotnet dht22.dll
Temperature: -273.15 ℃, Humidity: 0.0 %RH %
Temperature: -273.15 ℃, Humidity: 0.0 %RH %
Temperature: 25.3 ℃, Humidity: 29.1 %RH %
Temperature: -273.15 ℃, Humidity: 0.0 %RH %
Temperature: -273.15 ℃, Humidity: 0.0 %RH %
Temperature: 25.3 ℃, Humidity: 28.3 %RH %
Temperature: 25.3 ℃, Humidity: 28.2 %RH %
Temperature: -273.15 ℃, Humidity: 0.0 %RH %
Temperature: 25.3 ℃, Humidity: 28.3 %RH %

bbhxwl, 2021-02-22:
···
 static void Main(string[] args)
        {
            while (true)
            {
                using (Iot.Device.DHTxx.Dht22 dht = new Iot.Device.DHTxx.Dht22(26, PinNumberingScheme.Logical))
                {
                    var temp = dht.Temperature; // Try reading the temperature
                    var hum = dht.Humidity; // Try reading the humidity
                                            // You can only display temperature and humidity if the read is successful otherwise, this will raise an exception as both temperature and humidity are NAN
                    if (dht.IsLastReadSuccessful)
                    {
                        Console.WriteLine($"Temperature: {temp.DegreesCelsius}\u00B0C, Relative humidity: {hum.Percent}%");
                    }
                    else
                    {
                        Console.WriteLine("Error reading DHT sensor");
                    }
                }  
                 Thread.Sleep(2500);
            }
        }
    
···

bbhxwl, 2021-02-22:
`
 static void Main(string[] args)
        {
            while (true)
            {
                using (Iot.Device.DHTxx.Dht22 dht = new Iot.Device.DHTxx.Dht22(26, PinNumberingScheme.Logical))
                {
                    var temp = dht.Temperature; // Try reading the temperature
                    var hum = dht.Humidity; // Try reading the humidity
                                            // You can only display temperature and humidity if the read is successful otherwise, this will raise an exception as both temperature and humidity are NAN
                    if (dht.IsLastReadSuccessful)
                    {
                        Console.WriteLine($"Temperature: {temp.DegreesCelsius}\u00B0C, Relative humidity: {hum.Percent}%");
                    }
                    else
                    {
                        Console.WriteLine("Error reading DHT sensor");
                    }
                }  
                 Thread.Sleep(2500);
            }
        }
    
`

Ellerbach, 2021-02-24:
> @krwq I agree, I think this was discussed a few times, but so far nobody wanted to do the breaking change of converting the properties to TryGet methods.

We can always create a new way, put [Obsolete] on the old way?
And of course, fully agree with the fact it's not super obvious.

Ellerbach, 2021-02-25:
[Triage] It's been agreed to make a new proposal for API for DHT more intuitive and based out of the Try pattern. Properties can be removed as well.

gecclesinc, 2021-08-22:
Has anyone gotten this to work?

My DHT22 works fine with my Arduino using the Adafruit_Sensor and DHT libraries. It also works on my Raspberry Pi 4 using the Adafruit_CircuitPython_DHT library (https://github.com/adafruit/Adafruit_CircuitPython_DHT).

Would be awesome to get this working with the C# library.

> `
> static void Main(string[] args)
> {
> while (true)
> {
> using (Iot.Device.DHTxx.Dht22 dht = new Iot.Device.DHTxx.Dht22(26, PinNumberingScheme.Logical))
> {
> var temp = dht.Temperature; // Try reading the [...]

Tried this code and so many others because I really wanted this to work in C#.

I eventually wrote a script in Python that reads the DHT22 and writes it to SQL, but ALL of the rest of my project is in C# using dotnet/iot and the humidity sensor is just hanging out all on its own.

Anyways, appreciate everyone trying to help fix it, but it does seem the library is broken right now, at least on my RPi 4.

pgrawehr, 2021-08-23:
@gecclesinc : What exactly do you get as result? I have never tested a DHT22, but it works fine with DHT11 on Rpi4, and the two are very similar.

Mr-Technician, 2021-09-19:
I'm having a similar issue. The sensor works fine on python, but the read always fails from the .net library.

closed 2021-12-05 by pgrawehr

Geims83, 2022-01-02:
Hi all,
If I clone the repo and run a test, the code doesn't work.
It always return -273° C, 0% humidity and Success false.

I am running che code on a Pi2. Could this be a problem?
Thanks.

pgrawehr, 2022-01-03:
No, if the Pi2 was the problem, nothing would work and you weren't even able to start the application usually. 

That rather points to a problem with your wiring. Please double check the connections of your sensor.

Geims83, 2022-01-03:
Thanks @pgrawehr,
I had this doubt too, but as using Python with the Adafruit library the reading is ok I think the wiring is good.

pgrawehr, 2022-01-03:
Then I don't know what's wrong. Can you confirm that you are using a Raspberry Pi Zero 2 and not a Rasberry Pi 2B? Because the two are completely different and the later is not supported. Did you try long enough? It sometimes takes 5 to 10 attempts for a successful read.

Geims83, 2022-01-04:
Bingo! I'm using a Pi 2B. So it isn't supported right? :(
In think I missed that... In the docs it is stated that the libraries are not supported on ARMv6, so on devices prior to the Pi 2, and the projects are building correctly on net 6.0.... :(

pgrawehr, 2022-01-04:
I guess nobody ever tested this on a Pi 2B. As far as I know, the Pi2B _is_ ARMv6? Or can you test whether other samples work or some simple LED test?

Geims83, 2022-01-04:
The Pi 2 B 1.1 is ARMv7.
I'll test some leds and I'll let you know.

Ellerbach, 2022-01-05:
The Pi2B is a bit slower than the others but it still should work. @Geims83 what core are you using? Are you following the pattern here: https://github.com/dotnet/iot/tree/main/src/devices/Dhtxx#1-wire-protocol

PR #1701 (2021-12-05): Add TryReadHumidity/TryReadTemperature
Fixes #1445 

Adds TryReadXXX methods to Dht sensor binding. Tested with DHT22 on RPI4.

###### Microsoft Reviewers: [Open in CodeFlow](https://portal.fabricbot.ms/api/codeflow?pullrequest=https://github.com/dotnet/iot/pull/1701)