dotnet/iot #1467: Enabling optimal GPIO driver in containers

richlander, 2021-03-04:
I was playing with running our samples in a container. I've done that before, but wanted to validate which driver was being used. I was reading this [article](https://blog.alexellis.io/gpio-on-swarm/) to guide me.

I'm looking for three things:

- Documentation on the container device models we support (how to get access to the underlying device: GPIO, SPI, ...).
- Guidance on how to select the most optimal driver.
- API (or something) that enables you to know which driver was selected.

Here are the approaches that I've tried that have worked for me with the [led-blink sample](https://github.com/dotnet/iot/tree/main/samples/led-blink). I cannot tell which driver was actually selected.

```console
pi@raspberrypineapple:~/iot/samples/led-blink $ docker run --rm led-blink
Unhandled exception. System.IO.IOException: Error 2 initializing the Gpio driver.
   at System.Device.Gpio.Drivers.RaspberryPi3LinuxDriver.Initialize()
   at System.Device.Gpio.Drivers.RaspberryPi3LinuxDriver.OpenPin(Int32 pinNumber)
   at System.Device.Gpio.Drivers.RaspberryPi3Driver.OpenPin(Int32 pinNumber)
   at System.Device.Gpio.GpioController.OpenPinCore(Int32 pinNumber)
   at System.Device.Gpio.GpioController.OpenPin(Int32 pinNumber)
   at <Program>$.<Main>$(String[] args) in /app/Program.cs:line 13
pi@raspberrypineapple:~/iot/samples/led-blink $ docker run --rm --device /dev/gpiomem led-blink
GPIO pin enabled: 18
Light for 1000ms
Dim for 200ms
// snip ...
pi@raspberrypineapple:~/iot/samples/led-blink $ docker run --rm --privileged led-blink
GPIO pin enabled: 18
Light for 1000ms
// snip ...
pi@raspberrypineapple:~/iot/samples/led-blink $ docker run --rm --privileged -v /sys:/sys led-blink
GPIO pin enabled: 18
Light for 1000ms
Dim for 200ms
// snip ...
pi@raspberrypineapple:~/iot/samples/led-blink $ docker run --rm -v /sys:/sys led-blink
Unhandled exception. System.IO.IOException: Error 2 initializing the Gpio driver.
   at System.Device.Gpio.Drivers.RaspberryPi3LinuxDriver.Initialize()
   at System.Device.Gpio.Drivers.RaspberryPi3LinuxDriver.OpenPin(Int32 pinNumber)
   at System.Device.Gpio.Drivers.RaspberryPi3Driver.OpenPin(Int32 pinNumber)
   at System.Device.Gpio.GpioController.OpenPinCore(Int32 pinNumber)
   at System.Device.Gpio.GpioController.OpenPin(Int32 pinNumber)
   at <Program>$.<Main>$(String[] args) in /app/Program.cs:line 13
```

I am not sure how to test libgpiod, specifically which device to pass in. I know how to install it.

The Dockerfile in the repo for this sample is stale (I'm working on a PR for that). Here's the Dockerfile I used:

```Dockerfile
FROM mcr.microsoft.com/dotnet/sdk:5.0 AS build
WORKDIR /app

# copy csproj and restore as distinct layers
COPY *.csproj .
RUN dotnet restore

# copy and build app
COPY . .
RUN dotnet publish -c release -o out

FROM mcr.microsoft.com/dotnet/runtime:5.0-buster-slim-arm32v7
WORKDIR /app
COPY --from=build /app/out ./
ENTRYPOINT ["dotnet", "led-blink.dll"]
```

joperezr, 2021-03-05:
> Guidance on how to select the most optimal driver.

There shouldn't really be a need for this, as the whole idea is that if you don't specifically need/want one driver, you should be using the `new GpioController()` constructor which will pick the most optimal one based on your environment for you. This might be tricky in terms of docker containers, but I think that if anything we should probably adapt/change our default picking logic to support docker containers instead of try to teach users on how to do that and to not call the simple constructor instead.

>Some way to know which driver was selected.

That's fair, I'm not sure how much real use would this have in terms of functioning apps, but I definitely see benefit here for diagnosing issues and investigating problems as the implementations are very different across drivers. Best way I can think of how to do this today is either by debugging, or by using reflection to get the Type like:
```c#
var _driverField = typeof(GpioController).GetField("_driver", System.Reflection.BindingFlags.NonPublic | System.Reflection.BindingFlags.Instance);
Type selectedDriverType = _driverField!.GetValue(myController)!.GetType();
```

richlander, 2021-03-05:
Totally agree ... `GpioController` is the way to go.

I think we should just report the name somehow. We expose lots of information about the OS, machine env, and .NET itself in various APIs: https://github.com/dotnet/core/blob/main/samples/dotnet-runtimeinfo/Program.cs. We've had similar arguments about not exposing that information because apps shouldn't be making runtime decisions about that info, yet we end up doing it anyway.

At the very least, how about overriding `ToString()`?

Ellerbach, 2021-07-01:
[Triage] @richlander is there any open issue on this? Is this still valid? With Environment variable passed in the container (container good practice), you can read your environment variable and then do the mapping you want.

closed 2023-01-12 by krwq