dotnet/iot #1468: Enable selecting driver via environment variable

richlander, 2021-03-04:
I was playing with containers and wanted to try using libgpiod. I was unable to do that in containers, w/o changing my program. I'd like to be able to select the driver via environment variable.

Context: #1467

I would like to be able to enable the following two cases:

- `docker run --rm --device /dev/gpiomem led-blink`
- `docker run --rm -it --device /dev/gpiochip0 -e DOTNET_GPIO_DRIVE=libpdiod led-blink`

I agree that the current Raspberry Pi driver is the best default, but we should enable selecting the other one.

joperezr, 2021-03-04:
While I do think that some people might find this useful, I don't think that the right place to do this would be in System.Device.Gpio.dll. Instead, I would suggest doing this on the application side like:

```c#
GpioController myController;

string? GpioDriverToUse = Environment.GetEnvironmentVariable("DOTNET_GPIO_DRIVER");

if (GpioDriverToUse != null)
{
    myController = GpioDriverToUse switch
    {
        "LibGpiodDriver" => new GpioController(PinNumberingScheme.Logical, new LibGpiodDriver()),
        "RaspberryPiDriver" => new GpioController(PinNumberingScheme.Logical, new RaspberryPi3Driver()),
        // ... other built-in drivers
        _ => throw new ArgumentException($"Unkown driver: {GpioDriverToUse}")
    };
}
else
{
    // if env variable not set just use the default
    myController = new GpioController();
}

// use myController...
```

Basically two things here:
- I don't think that it is the right thing for a down-level library to read environment variables in general unless absolutely necessary.
- By letting the app being the one that decide this, then they can control other ways of providing this default selection, like using appcontext switches, config files, or pulling the default value from the network instead.

Thoughts?

richlander, 2021-03-04:
CoreCLR (which is kinda a library), for example, honors tons of ENV. We also support appcontext switches in the libraries. I think honoring ENVs is fine.

I don't think getting people to include this code in their apps is reasonable. S.D.G is the one that is exposing the black box. Its odd to me to not make it configurable. If we go with the app model, it just means we have a non-standard way to host apps in containers, and our docs and blog posts will be complicated.

This isn't a direct analogue, but check out the guidance I wrote on how to [host ASP.NET Core in containers with ENV injection](https://github.com/dotnet/dotnet-docker/blob/main/samples/host-aspnetcore-https.md). It's all standard gestures, and it works for all ASP.NET Core apps (w/o modification). I spent a fair bit of time with that team writing those docs (and they are used a lot).

I thought about creating a wrapper over `GpioController` in I.D.B that does what you are suggesting. We could do that, but it means we need another class (like `ConfigurableGpioController`) and then we force people to take another dependency.

I'd like to value prop to be:

- We support running (any) .NET IoT apps in containers.
- You can choose to use gpiomem, libgpio or sysfs as the backing GPIO "API".

The only way we can offer that model is if the underlying library exposes more configurability.

Put another way, are you OK with me putting that code into our samples? That would be sad.

krwq, 2021-03-05:
I think I'm ok with the environment variable here. There was plenty of times we got issue and switching driver was the easiest solution we could provide, it's much easier to ask people to try running the app with some magic environment variable rather than changing their code, redeploying and testing. I assume over time there will be lots of apps which already shipped and will work on some OS-es but not the other because some driver might not be available. Asking authors to change the code doesn't seem like a right approach. With environment variable we can ask users to quickly test if that works and have a quick workaround. I'd call it `DOTNET_IOT_DEFAULT_GPIO_DRIVER` though if we decide to do it. We'd also need to make sure this is very cheap: environment variable is read only once per execution and check is very simple and throws when invalid value is used.

joperezr, 2021-03-05:
Before we decide or not, let's take one step back and think what actual problem this would be addressing and whether that is common and if the solution is the right fix or not. Questions that jump to mind are:
- Why would somebody want to change the driver that gets picked by default?
- How common would a user want to do this?
- How common would it be for the user to have more than one app running on their device and potentially be affected by changing the env variable?
- If the user wants a specific driver to get picked, why are they not using our already existing logic to pass in a driver directly?

Basically when we added the overload to GpioController that took in a driver we were thinking about two major scenarios:
- People that wanted to go around the default driver due to a bug on the underlying driver (for example, a bug on the raspberryPi driver and so user might want to pass in the libgpiod driver instead)
- People that wanted to pass in a custom driver they wrote that would not be picked by our default logic. For example, the drivers that have been recently added under src/devices/gpio.

I'm not necessarily against adding an Env variable, the point I'm trying to make is that I don't think it would be at all common for folks to want to mess with the default picking logic as I don't really see much benefit on doing so, other than having their app already compiled and running and wanted to change the underlying implementation on the fly without recompiling, which to me is an edge case.

joperezr, 2021-03-05:
>There was plenty of times we got issue and switching driver was the easiest solution we could provide, it's much easier to ask people to try running the app with some magic environment variable rather than changing their code, redeploying and testing.

I don't think that the right solution for this problem is to have a magical switch to change the default driver selection. I think that the right fix for this, would be to have much better diagnostic tools/errors/messages into our library that helps us better identify what the underlying problem is. If we just do it with the variable, it means we will have a workaround for the user, but we will not be able to find out what/where the actual bug is in one of our default drivers potentially impacting many other users.

pgrawehr, 2021-03-07:
What about taking the (automatic) driver choice out of S.D.G and give the user some freedom to it? With the board PR, there's currently a second place where the device detection is required, since now we want to automatically choose the right board and not only the right GPIO driver. So far, we have also more or less ignored, that the same problem as with the GPIO drivers also exist for other drivers, like I2C or SPI. The problem isn't so big there, because there is only one for Windows and one for Linux, but this might not stay like that. 

One way of doing this would be to add a (static?) method to each driver that detects whether this driver is compatible with the given hardware. Or add something like a "DriverInfo" class that returns capabilities of a driver, so that the client can choose what he needs.

joperezr, 2021-03-07:
We can't take it out as we definitely want to keep at least one GpioController constructor that abstracts the driver concept out from the user. The GpioDriver is theoretically an implementation detail, so I'd expect most cases to not really care about the driver that gets picked. Regarding the comment of letting users choose or write their own auto-detection logic, that is already the case, as there is nothing stopping users from doing a quick runtime analysis of the board that they are running in, and depending on that having a switch statement that pick which driver they pass to their GpioController constructor. Also, regarding the static method to test if a driver can run on the machine or not, that may be interesting but won't really solve this issue, as even if we do that you would need to have some sort of order of preference in case two drivers return true, which would be equivalent to what we have today.

In summary it is not really that I'm against changing how the default logic works, but if we look for precedence in the rest of the framework, we usually have the same case as we have here where you have two ways of initializing things: the easy way that does some defaults for you depending on the environment, and the way that let's you specify which values to be used instead of defaulting. I don't really see much benefit of adding a third way that is in the middle of those two where we let you override the defaults that get picked, I think that if that is needed, this belongs as a separate abstraction on the consuming application that would set the right defaults and then call the second constructor that takes in the driver.

pgrawehr, 2021-03-07:
... except that with what we have in place right now, it is very ugly to write your own detection logic, because the Driver interface is not public. Of course you could write everything yourself, but then you're duplicating logic that is already in SDG. So if you want to test whether a particular driver works, you have to instantiate it and test whether opening a pin does not throw.

joperezr, 2021-03-08:
Mmm not sure what you mean with the Driver interface not being public, I wasn't thinking of trying to test and calling a pin to see if it worked as that is not the way we do it for most of our drivers, we instead do an ahead of time check to see if the platform supports it. I was thinking some user app to look a bit like:

```c#
class Program
{
    static void Main(string[] args)
    {
        GpioController myController;
        GpioDriver theDriverPicked = null;

        // Check if we are currently on MySpecial board
        if (MySpecialBoardDriver.IsThisDriverSupported())
        {
            theDriverPicked = new MySpecialBoardDriver();
        }
        else
        {
            // ...Check if this is Raspberry Pi
            if (WeAreRunningOnARaspberryPi())
                theDriverPicked = new RaspberryPi3Driver();

            // Check if this has Libgpiod
            else if (LibGpiodIsInstalled())
                theDriverPicked = new LibGpiodDriver();

            // else if....   Continue more logic here
        }

        myController = new GpioController(PinNumberingScheme.Logical, theDriverPicked);

        // use myController...
    }

    static bool WeAreRunningOnARaspberryPi() => throw new NotImplementedException();

    static bool LibGpiodIsInstalled() => throw new NotImplementedException();
}

public class MySpecialBoardDriver : GpioDriver
{
    public static bool IsThisDriverSupported() => throw new NotImplementedException();
    // ... the rest of GpioDriver implementation
}
```

joperezr, 2021-03-11:
>With this, each and every application would need to replicate code that's actually already present in the drivers (as the drivers test the hardware).

With that  you are assuming that **every application** would want to do this, which is exactly my point. I think that it will be super rare for people to want to do this which is kind of why I push back here. I would rather hear from the community how popular this feature request is, and specially hear why they want it (since it is likely that if a lot of folks are trying to change the default logic is because the default logic is wrong).

Just to reiterate my point, I think here we are trying to fix an issue as if it were super common that we actually don't have data of, so before we go and add a bunch of API and code in order to enable this, let's make sure that enough people would benefit from this to see if it would have enough value.

krwq, 2022-11-17:
[Triage] If we're gonna do anything it sounds like we should add API for factory and environmental variable would then become implementation detail. Closing this as by design

krwq, 2022-11-17:
ref: #1962

closed 2022-11-17 by krwq