dotnet/iot #1469: Better experience when using LibGpiodDriver

richlander, 2021-03-04:
I was using `LibGpiodDriver` as part of #1467. I ran into two problems.

I installed `gpiod` instead of `libgpiod-dev`. I saw this error:

```console
pi@raspberrypineapple:~/iot/samples/led-blink $ ./bin/Debug/net5.0/led-blink
Unhandled exception. System.PlatformNotSupportedException: Libgpiod driver not installed. More information on: https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/about/
at System.Device.Gpio.Drivers.LibGpiodDriver..ctor(Int32 gpioChip) in /home/pi/iot/src/System.Device.Gpio/System/Device/Gpio/Drivers/LibGpiodDriver.cs:line 83
at <Program>$.<Main>$(String[] args) in /home/pi/iot/samples/led-blink/Program.cs:line 13
Aborted
```

I was then told to install `libgpiod-dev`. That worked.

We should do two things:

- Enable using `gpiod`.
- Provide a better error message if the underlying binary is missing (or the system is misconfigured). I'd suggest pointing to a README in this repo. I read the [referenced doc](https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/about/). It wasn't useful for me to resolve my issue.

joperezr, 2021-03-04:
>Enable using `gpiod`.

This part is already covered by issue: https://github.com/dotnet/iot/issues/1070 so let's keep this issue really about providing a better error message if the binary is missing and providing a pointer to a doc that explains how to fix it.

richlander, 2021-03-04:
I changed the title. Do you want me to start a doc, and then we can change the code to point there, and update the doc as we go?

joperezr, 2021-03-05:
Sounds good to me. I suppose you were already intending to do this, but when you say point we should really be creating an aka.ms link that points there and have the code go to the aka.ms link 😄

honzapatCZ, 2024-03-20:
I'd just add that there's now libgpiod.so.3 in my yocto install. As for a quick fix I'll install the dev package, which will hopefully fix this. But this might want to be revisited.