dotnet/iot #1470: No way to use pin for pwm after calling SetPinMode.

007roy, 2021-03-04:
**Describe the bug**
On Raspberry Pi 4
If GpioController.SetPinMode or OpenPin are used to set PinMode.Output at anytime system must be rebooted before PwmChannel can be used.

**Steps to reproduce**
Add 'dtoverlay=pwm,pin=18,func=2' line to /boot/config.txt and reboot

exec this code:
...
var pwm=PwmChannel.Create(0,0,500,1);
pwm.start();
...

exec this code:
...
controller.OpenPin(18,PinMode.Output);
...

then repeat:
...
var pwm=PwmChannel.Create(0,0,500,1);
pwm.start();
...

**Expected behavior**

Describe what did you expect to happen

**Actual behavior**

Describe what actually happened

**Versions used**

Add following information:

.NET SDK (reflecting any global.json):
 Version:   5.0.100
 Commit:    5044b93829

Runtime Environment:
 OS Name:     raspbian
 OS Version:  10
 OS Platform: Linux
 RID:         linux-arm
 Base Path:   /home/pi/dotnet/sdk/5.0.100/

Host (useful for support):
  Version: 5.0.0
  Commit:  cf258a14b7

.NET SDKs installed:
  5.0.100 [/home/pi/dotnet/sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.App 5.0.0 [/home/pi/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 5.0.0 [/home/pi/dotnet/shared/Microsoft.NETCore.App]

pgrawehr, 2021-03-05:
Hi @007roy 

This is a known problem, see #874, #878 and others. If you've got wiringpi installed (the latest version), you can use a command like `gpio -g mode 12 ALT0` to reset pin 12 to PWM. 

PR #1128 is going to fix this. Since it's a big change, review is taking its time. But we definitelly want this in the next release.

krwq, 2021-03-05:
@007roy thank you for filing the issue - this gives us feel that this is not a potential problem happening with tests abusing the APIs and it's an actual pain point. Just to clarify: next release => next major release timeline. This is something we do want to ship but it has non-trivial design - getting this wrong would have really bad long-term user experience impact which we'd prefer to avoid and that's why we prefer to take extra time on the APIs.

closed 2021-03-06 by msftbot[bot]