dotnet/iot #1471: Error while waiting for pin interrupts

vulild, 2021-03-05:
Unhandled exception. System.IO.IOException: Error while waiting for pin interrupts.
Mar 05 01:26:30 raspberrypi1 dotnet[6912]:    at System.Device.Gpio.Drivers.SysFsDriver.WasEventDetected(Int32 pollFileDescriptor, Int32 valueFileDescriptor, Int32& pinNumber, CancellationToken cancellationToken)
Mar 05 01:26:30 raspberrypi1 dotnet[6912]:    at System.Device.Gpio.Drivers.SysFsDriver.DetectEvents()
Mar 05 01:26:30 raspberrypi1 dotnet[6912]:    at System.Threading.ThreadHelper.ThreadStart_Context(Object state)
Mar 05 01:26:30 raspberrypi1 dotnet[6912]:    at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
Mar 05 01:26:30 raspberrypi1 dotnet[6912]: --- End of stack trace from previous location where exception was thrown ---
Mar 05 01:26:30 raspberrypi1 dotnet[6912]:    at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
Mar 05 01:26:30 raspberrypi1 dotnet[6912]:    at System.Threading.ThreadHelper.ThreadStart()

pgrawehr, 2021-03-05:
Probably a duplicate of #378. Does it always happen for you, or only intermittenly?

Ellerbach, 2021-03-25:
[Triage] I will go ahead and close this one as duplicate of #378. Please @vulild feel free to add anything new to the other issue. Thanks.

closed 2021-03-25 by Ellerbach

vulild, 2021-03-29:
安装了libgpiod-dev之后，不会报错了，但是程序会挂掉。建议把这个错误做成回调，或者屏蔽掉，我测试过屏蔽掉错误不会影响程序运行，也不影响引脚的监测。