dotnet/iot #1472: Issues with I2C speed of command-execution

manuelgass, 2021-03-05:
Hello first of all i would like to give you a short overview of the system i am using.

```
.NET Core SDK Version:   3.1.112 running on a BeagleBone Black with Debian10.5-arm.
System.Device.Gpio 1.3.0
```

To build the project i'm using the command
`dotnet publish -c release -o publish -r linux-arm`.

Everything works perfectly and the project builds without any problems.
There are also no problems with the execution.
However, I had to determine with some measurements that between the write phases long times nothing happens.
Reading and writing takes place with a bus speed of 100kHZ and still works fine.
This was also checked with a logic analyzer.

What does not meet my expectations, however, is the time of 13ms for two writes.
With other controllers and programming languages I have made other experiences here and would have expected a processing time of the two commands of less than 1ms.

Is this a problem of the Beaglebone, my application or has anyone made similar observations or even completely different observations?

With kind regards Manuel

This is the code im using

```
           var i2cDevice = I2cDevice.Create(new I2cConnectionSettings(2, 0x21));
            Stopwatch watch;
            watch = Stopwatch.StartNew();
            for (int i = 0; i < 1000; i++)
            {
                i2cDevice.Write(new byte[] {0x08, 0x00, 0x00, 0x00, 0x00, 0x00});
                i2cDevice.Write(new byte[] {0x08, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF});
            }
            Console.WriteLine(watch.ElapsedMilliseconds / 1000);
```

Additional info
`.NET Core SDK (reflecting any global.json):
 Version:   3.1.112
 Commit:    cd75074bf4

Runtime Environment:
 OS Name:     debian
 OS Version:  10
 OS Platform: Linux
 RID:         debian.10-arm
 Base Path:   /home/debian/dotnet-arm32/sdk/3.1.112/

Host (useful for support):
  Version: 3.1.12
  Commit:  0267ad09c6

.NET Core SDKs installed:
  3.1.112 [/home/debian/dotnet-arm32/sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.App 3.1.12 [/home/debian/dotnet-arm32/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 3.1.12 [/home/debian/dotnet-arm32/shared/Microsoft.NETCore.App]`

joperezr, 2021-03-05:
Thanks a lot for logging the issue @manuelgass and for the detailed explanation above of the environment and the repro steps. Few notes to add here:
- Stopwatch is not being stopped after the for-loop so just to keep in mind that you are getting defenitly some overhead on the Console.WriteLine + the allocation of the bytes to send + the jit itself may not be warm enough the first few rounds of the loop. In order to get a better measurement that we can investigate, could you try adding the benchmarkdotnet package to your repro project and refactor the code to look like:
```c#
using BenchmarkDotNet.Attributes;
using BenchmarkDotNet.Running;
using System.Device.I2c;

namespace benchmark
{
    class Program
    {
        static void Main(string[] args)
        {
            var summary = BenchmarkRunner.Run<I2CBenchmark>();
        }
    }

    public class I2CBenchmark
    {
        private readonly byte[] byteArray1 = new byte[] { 0x08, 0x00, 0x00, 0x00, 0x00, 0x00 };
        private readonly byte[] byteArray2 = new byte[] { 0x08, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF };

        [Benchmark]
        public void TestWriteSpeed()
        {
            using (var i2cDevice = I2cDevice.Create(new I2cConnectionSettings(2, 0x21)))
            {
                for (int i = 0; i < 1000; i++)
                {
                    i2cDevice.Write(new byte[] { 0x08, 0x00, 0x00, 0x00, 0x00, 0x00 });
                    i2cDevice.Write(new byte[] { 0x08, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF });
                }
            }
        }
    }
}
```

After doing that, just run it like `dotnet run -c Release` on your beaglebone to make sure that it runs it in the Release configuration, and hopefully that will get better perf diagnostics so that we can compare that to other devices like RPis. If we find similar measurements than other devices, then we can profile the run to try to find out why is there a slowness and see if we can fix something here.

manuelgass, 2021-03-06:
Good morning and thank you for the quick reply,

I ran the tests with the setup you suggested on both the BeagleBone Black and a Raspberry Pi3b.
The results are exactly the same and listed below.
However, since I had no previous experience with the I2C slave I used, I wrote a short Python script this morning to make the results comparable.
Now to my findings: Both with Python and with the library used here, an average of 13.84 seconds is needed for 1000 loop passes.
So I think this issue can be closed.

Anyway, thanks for the quick answer and I hope @joperezr is not angry with me for not making the comparison with Python before I opened the issue.
I was unfortunately of the firm belief by experiences from my past with other chips that the interval between two writes is faster.

To make it complete here now the results and the used Python code.

Python-Code:
```
import smbus
import time
DEVICE_BUS = 1
DEVICE_ADDR = 0x21
bus = smbus.SMBus(DEVICE_BUS)

all00 = [0x00, 0x00, 0x00, 0x00, 0x00]
allff = [0xff, 0xff, 0xff, 0xff, 0xff]

start = time.time()
for x in range(0, 1000):
	bus.write_i2c_block_data(DEVICE_ADDR, 0x08, all00)
	bus.write_i2c_block_data(DEVICE_ADDR, 0x08, allff)

end = time.time()
print(end-start)
```

BeagleBoneBlack Benchmark
```
// Validating benchmarks:
// ***** BenchmarkRunner: Start   *****
// ***** Found 1 benchmark(s) in total *****
// ***** Building 1 exe(s) in Parallel: Start   *****
// start dotnet restore  /p:UseSharedCompilation=false /p:BuildInParallel=false /m:1 in /home/debian/HelloWorld/bin/Release/netcoreapp3.1/c0849e1a-8517-4020-a905-1da328c81723
// command took 24.17s and exited with 0
// start dotnet build -c Release  --no-restore /p:UseSharedCompilation=false /p:BuildInParallel=false /m:1 in /home/debian/HelloWorld/bin/Release/netcoreapp3.1/c0849e1a-8517-4020-a905-1da328c81723
// command took 58.3s and exited with 0
// ***** Done, took 00:01:24 (84.09 sec)   *****
// Found 1 benchmarks:
//   I2CBenchmark.TestWriteSpeed: DefaultJob

// **************************
// Benchmark: I2CBenchmark.TestWriteSpeed: DefaultJob
// *** Execute ***
// Launch: 1 / 1
// Execute: dotnet "c0849e1a-8517-4020-a905-1da328c81723.dll" --benchmarkName "HelloWorld.I2CBenchmark.TestWriteSpeed" --job "Default" --benchmarkId 0 in /home/debian/HelloWorld/bin/Release/netcoreapp3.1/c0849e1a-8517-4020-a905-1da328c81723/bin/Release/netcoreapp3.1
Failed to set up high priority. Make sure you have the right permissions. Message: Permission denied
// BeforeAnythingElse

// Benchmark Process Environment Information:
// Runtime=.NET Core 3.1.12 (CoreCLR 4.700.21.6504, CoreFX 4.700.21.6905), Arm RyuJIT
// GC=Non-concurrent Workstation
// Job: DefaultJob

OverheadJitting  1: 1 op, 5012193.00 ns, 5.0122 ms/op
WorkloadJitting  1: 1 op, 13911071780.00 ns, 13.9111 s/op

WorkloadWarmup   1: 1 op, 13850720064.00 ns, 13.8507 s/op
WorkloadWarmup   2: 1 op, 13852085444.00 ns, 13.8521 s/op
WorkloadWarmup   3: 1 op, 13851781485.00 ns, 13.8518 s/op
WorkloadWarmup   4: 1 op, 13852579171.00 ns, 13.8526 s/op
WorkloadWarmup   5: 1 op, 13864495774.00 ns, 13.8645 s/op
WorkloadWarmup   6: 1 op, 13856483223.00 ns, 13.8565 s/op

[... 194 lines cut ...]
.NET Core SDK=3.1.102
  [Host]     : .NET Core 3.1.2 (CoreCLR 4.700.20.6602, CoreFX 4.700.20.6702), Arm RyuJIT
  DefaultJob : .NET Core 3.1.2 (CoreCLR 4.700.20.6602, CoreFX 4.700.20.6702), Arm RyuJIT

|         Method |    Mean |   Error |  StdDev |
|--------------- |--------:|--------:|--------:|
| TestWriteSpeed | 13.85 s | 0.005 s | 0.005 s |

// * Hints *
Outliers
  I2CBenchmark.TestWriteSpeed: Default -> 1 outlier  was  detected (13.84 s)

// * Legends *
  Mean   : Arithmetic mean of all measurements
  Error  : Half of 99.9% confidence interval
  StdDev : Standard deviation of all measurements
  1 s    : 1 Second (1 sec)

// ***** BenchmarkRunner: End *****
// ** Remained 0 benchmark(s) to run **
Run time: 00:05:07 (307.85 sec), executed benchmarks: 1

Global total time: 00:05:35 (335.04 sec), executed benchmarks: 1
// * Artifacts cleanup *
```

joperezr, 2021-03-06:
I'm not angry at all 😄, quite the opposite actually. I'm very happy to have users like you being part of the community that a) log detailed issues for investigation when you detect something wrong, and b) help us diagnose what might be wrong with the library. People like you are exactly the ones that make our libraries better so I really appreciate you taking the time to follow through. We hope to continue seeing contributions from you and if you hit any issues or see something wrong, please do log them so that we can take a look. Issues opened don't necessarily need to be bugs all the time, having investigation or discussion issues is totally fine too.

Thanks again for the quick validation and following up, I'll go ahead and close this for now but feel free to reopen if you see any other perf issues again.

closed 2021-03-06 by joperezr