dotnet/iot #1479: Opening an output pin with libpgiod driver changes the pin value

WillooWisp, 2021-03-11:
**Describe the bug**

I want to read the value from an output pin, but by just opening an output pin it gets set to high. So reading it always returns high of course.

But using gpioget to read the pin value returns it as low, if that was what it was.

**Steps to reproduce**

Create GPIO controller
Open pin (for one that is low)
It is set to high, without an explicit write

```csharp
var led1 = 19;

using var gpio = new GpioController(PinNumberingScheme.Logical, new LibGpiodDriver(2));

gpio.OpenPin(led1, PinMode.Output);
// gpio.Write(led1, PinValue.High);
gpio.ClosePin(led1);
```

**Expected behavior**

I expected to be able to open the pin without its state being affected, so I could read the state afterwards.

**Actual behavior**

Opening the pin changes state.

**Versions used**

Debian 10 (arm64)
System.Device.Gpio package: 1.3.0
.NET Core: 3.1

Dockerfile
```
FROM mcr.microsoft.com/dotnet/sdk:3.1 as build

WORKDIR /src
COPY GpioApp.csproj ./
RUN dotnet restore GpioApp.csproj

COPY Program.cs ./
RUN dotnet build GpioApp.csproj -c Release -o /app

FROM mcr.microsoft.com/dotnet/runtime:3.1-alpine-arm64v8 as final

RUN apk update && apk add libgpiod-dev

WORKDIR /app
COPY --from=build /app .
ENTRYPOINT ["dotnet", "GpioApp.dll"]
```

docker run --privileged --rm dotnet-gpio:3.1

pgrawehr, 2021-03-11:
When you want to read the pin value, why do you open it in output mode? When the pin is set to output, the value must be set to either low or high. For the libgpiod driver the default value is apparently high.

Currently, the default value for an output pin is driver dependent, but we're working on fixing that. In either case, reading the value from an output pin is never returning the previous value.

WillooWisp, 2021-03-12:
I just read these issues, https://github.com/dotnet/iot/issues/183 and the fix https://github.com/dotnet/iot/pull/986 for allowing reading of output values and assumed that would work.

Are you saying that it should never be possible to read back the state from an output pin, if previously written? Why is that?

pgrawehr, 2021-03-12:
That fix just technically allows to read output pins (previously, there was an exception when doing so). But reading the output pin alwaysy returns the value that was previously written, because that's its value. If an output is set to High, you can read its value, but it will always return High, because that's what it's set to. 

On the hardware level, a pin set to high is techically connected to VCC, so that means it's outputing 3.3V (on the Pi). If you read that back, that's obviously a High. To read the value of that pin as a Low would mean its voltage would have to be < 1.2V, which is only possible if the pin is shorted to ground. That, however, would damage the pin, so it's not supposed to ever happen.

krwq, 2021-03-12:
@WillooWisp if I understand correctly you expect that following will happen, correct?

```csharp
var led1 = 19;
using var gpio = new GpioController(PinNumberingScheme.Logical, new LibGpiodDriver(2));

gpio.OpenPin(led1, PinMode.Output);
gpio.Write(led1, PinValue.High);
gpio.ClosePin(led1);

gpio.OpenPin(led1, PinMode.Output);
// value is high after open
gpio.Write(led1, PinValue.Low);
gpio.ClosePin(led1);
gpio.OpenPin(led1, PinMode.Output);
// value is low after open
```

I can see this point if you look at it from low-level perspective since opening pin there doesn't mean anything and pin is either input or output.

WillooWisp, 2021-03-12:
Two different things...

1)
Yes, I would not have assumed opening the pin for writing would change the state (before I can actually write what I want), since that can be rather nasty, depending on what it controls.

2)
But I thought it would also be possible to verify the state after writing, like this (assuming that opening the pin does not change the state)...

```
var led1 = 19;
using var gpio = new GpioController(PinNumberingScheme.Logical, new LibGpiodDriver(2));

gpio.OpenPin(led1, PinMode.Output);
gpio.Write(led1, PinValue.Low);
gpio.ClosePin(led1);

gpio.OpenPin(led1, PinMode.Output);
// Reads last state, Low in this case
Console.WriteLine($"LED 1: {gpio.Read(led1)}");
gpio.ClosePin(led1);

gpio.OpenPin(led1, PinMode.Output);
gpio.Write(led1, PinValue.High);
gpio.ClosePin(led1);

gpio.OpenPin(led1, PinMode.Output);
// Reads last state, High in this case
Console.WriteLine($"LED 1: {gpio.Read(led1)}");
gpio.ClosePin(led1);
```

pgrawehr, 2021-03-12:
If you want to read the state after writing, that's possible. But you should not close and reopen the pin then. Actually, closing an output pin should reset the value to Low (at least that's what we have specified).

WillooWisp, 2021-03-12:
So if I understand you correctly.

1)
In order to keep an output pin high, you must have the pin open all the time, e.g. if the application ends, pin gets closed, the pin will get set to low? 

Isn't this rather limiting the use cases, why resetting the pin to low when closing an output pin? To me it feels more correct to keep it high until device is power cycled.

2)
Application reboots, there is now way at all to read the current state of the output pin? If the above is true I guess you should always assume it to be low.

3)
Opening a pin now sets it to high to that cannot be correct, right?

4)
If two programs control the same output pin, there is no way to read the state that one or the other has written, since not possible to read state from an output tag?

5)
The issues mentioned earlier, and also merged, where reading from an output tag was fixed and allowed, what is the use case for that if not to find out what the state is upon application start?

krwq, 2021-03-12:
@WillooWisp it's being reset on open, not on close if I remember correctly. I also don't agree with this design to be honest.

WillooWisp, 2021-03-12:
Yes that is what I see as well, but in my case the reset with the LibGpiodDriver actually sets it to High rather than Low.

1. Since this clearly cannot be what you want, or can expect everyone to want, I believe it should be changed in order to put the choice in the hands of the application developer and not the API developer. How can we address this issue?

2. What does this merge fix do then? https://github.com/dotnet/iot/pull/986

krwq, 2021-03-12:
Possibly I remembered wrong. We would need to discuss this during our next triage meeting as this would be breaking if someone relied on the behavior of this being set to Low after Open and updating it now could case issues.

WillooWisp, 2021-03-25:
When can such a fix be released for the use of the libgpio driver?

pgrawehr, 2021-03-28:
I try to work on this next week, so I expect it to be in the next release. Time frame is 4 to 6 weeks for a new release, maybe faster if you build yourself.

pgrawehr, 2021-04-07:
Observations so far:

- The RaspberryPi3LinuxDriver sets the mode to INPUT on open (no other driver does that)
- Similarly, it sets the mode to INPUT and LOW on close
- The SysFs driver does neither, the corresponding test (OpenPinDefaultsModeToInput) actually fails on the SysFsDriver if running isolated. 
- The LibGpiodDriver does neither, too. Also, it does not read the initial state of the line on open, and always returns Input as "cached" value, until a mode has been set. 

My suggestion would be to take the breaking change for the RaspberryPi3LinuxDriver and remove any code in Open() or Close() that changes the state or direction. The impact shouldn't really be big because most people would anyway set the mode after opening a pin. Also fix the LibGpiodDriver.

joperezr, 2021-04-08:
[Triage]: Agreed with @pgrawehr. Next steps here are to remove any specific logic we currently have on drivers to try to emulate consistency across them, and instead leave the state unchanged when the Open method is called, unless OpenPin overload with initialValue is called.

rmc47, 2021-04-11:
Along similar lines to @pgrawehr's comment, the toggle to input on OpenPin in the driver causes a slightly odd effect: if the pin was previously set to output and low, then calling `GpioController.OpenPin(pinNumber, PinMode.Output)` currently causes a toggle of the pin's state to input, then back to output.

On the Pi, that causes a momentary pull-up on the pin while it's in the input state.

(Since `GpioController.OpenPin(int, PinMode)` turns into `OpenPin` followed by `SetPinMode` internally, it isn't possible to avoid this transition at the moment.)

WillooWisp, 2021-05-25:
Any status update to share on the progress regarding this and any potential release date?

pgrawehr, 2021-05-25:
@WillooWisp This was blocked due to some related fix. Now PR #1520 is almost ready to merge. With luck, I'll be able to finish it later this week. We expect to have a new release very soon after.

raffaeler, 2021-06-01:
The gpio utility on the RPi allows to write the state of the port regardless the pin mode.
After writing the pin to either 0 or 1, I can then set the pin mode to output and start using the pin with the safety of not toggling the state inappropriately.

When there is a relay connected to that pin and the raspberry starts, I have to preset the initial state so that the relay does not toggle when the pin is opened.

I am trying to set the pin states at boot with the gpio utility before starting the my .net service. Is any other better solution to avoid gpio? Will the upcoming PR solve the problem?

TIA

raffaeler, 2021-06-01:
The following overload looks promising: `public void OpenPin(int pinNumber, PinMode mode, PinValue initialValue)`
but it is still not available in pre-release `1.5.0-prerelease.21212.1` (which is not tagged in the repo).
Any upcoming packet?

TIA

pgrawehr, 2021-06-01:
@raffaeler PR #1520 should hopefully be merged this week (we are investigating on a possible regression that is not understood yet). We expect a full release shortly after.

raffaeler, 2021-06-08:
@pgrawehr Could you please tell me the rough amount of time needed to see at least a preview package?
Without this PR I definitely need to rewrite the code using a different library and of course I would like to avoid it.

Thanks and sorry for pressing, but at least I can decide what to do.

pgrawehr, 2021-06-10:
@raffaeler I do hope this gets merged this week, but the responsible person is very busy right now. I'll have to push again.

closed 2021-07-09 by krwq

nzain, 2021-07-13:
Good to see this fixed and merged! 👍 

@raffaeler if you can't wait for a package, you can pull the code,`dotnet pack` your own nuget package, and put it in a local nuget feed. This project is fairly complex but it builds fine when you follow the contributing document. You would usually define a very small version increment such that the next official release immediately replaces your custom build - but I was overwhelmed by build complexity and couldn't find the version number magic.

Eventually @pgrawehr or @krwq have a rough timeline for us?

krwq, 2021-07-13:
Can't promise anything but I'm thinking somewhere this month. We will likely discuss this during our next triage meeting.

raffaeler, 2021-07-13:
Thank yuu @nzain I am already familiar with manually repackaging the dotnet repositories :)
Anyway I would expect would publish sooner the prerelease package on this other nuget feed `https://pkgs.dev.azure.com/dnceng/public/_packaging/dotnet5/nuget/v3/index.json` 
Am I correct @krwq ?

krwq, 2021-07-13:
@raffaeler theoretically:
https://pkgs.dev.azure.com/dnceng/public/_packaging/dotnet6/nuget/v3/index.json (note dotnet6 not dotnet5)

has the latest but seems to me the last package is from June 4 which seems something is wrong with the publishing even though I'm seeing successful builds in the pipeline. I will ask our engineering team if they know what's wrong

raffaeler, 2021-07-13:
Oops, sure :)
Thank you @kwrq ! Look forward to it

krwq, 2021-07-16:
@raffaeler we now have a new nightly package you can try out. The nightly should work now (minus flaky tests which can be retried)

PR #1520 (2021-07-09): Opening/Closing a pin should not affect its state if possible
Fixes #1222 
Fixes #1479 

###### Microsoft Reviewers: [Open in CodeFlow](https://portal.fabricbot.ms/api/codeflow?pullrequest=https://github.com/dotnet/iot/pull/1520)