dotnet/iot #1483: Kernel SPI buffer size limitation with SpiDevice.Write()

thecaptncode, 2021-03-15:
I am a contributor on a Waveshare E-Paper device driver for the Raspberry Pi 2 or later that uses the SPI bus.  The solution uses System.Device.Gpio v1.3.0 and targets .NET Standard 2.0.  In trying to boost performance, one of the things I am doing is moving to writing to the SPI device using ```SpiDevice.Write(ReadOnlySpan<Byte>)``` since a single screen on the Waveshare 7.5inch E-Paper (B) driver is over 120KB of data and must be written in full with every update.

On my Raspberry Pi 3B+ running Raspbian 10 (buster) I see that as long as I write slices of 4KB or smaller to the SpiDevice, this works fine, but as soon as a slice is at all larger than 4KB, it throws the exception:
```
Unhandled exception. System.IO.IOException: Error 90 performing SPI data transfer.
   at System.Device.Spi.UnixSpiDevice.Transfer(Byte* writeBufferPtr, Byte* readBufferPtr, Int32 buffersLength)
   at System.Device.Spi.UnixSpiDevice.Write(ReadOnlySpan`1 buffer)
   at Waveshare.Common.EPaperDisplayHardware.Write(ReadOnlySpan`1 buffer) in C:\Users\captn\source\repos\thecaptncode\Waveshare.EPaperDisplay\Waveshare\Common\EPaperDisplayHardware.cs:line 186
   at Waveshare.Common.EPaperDisplayBase.SendData(Byte[] data) in C:\Users\captn\source\repos\thecaptncode\Waveshare.EPaperDisplay\Waveshare\Common\EPaperDisplayBase.cs:line 275
   at Waveshare.Common.EPaperDisplayBase.DisplayImage(Bitmap image) in C:\Users\captn\source\repos\thecaptncode\Waveshare.EPaperDisplay\Waveshare\Common\EPaperDisplayBase.cs:line 187
   at Waveshare.Example.Program.Main() in C:\Users\captn\source\repos\thecaptncode\Waveshare.EPaperDisplay\Waveshare.Example\Program.cs:line 54

```

By the way, this is a self-contained deployment with no SDK installed on the RPi.

From my tests and searches, it appears that the above error occurs whenever a slice is written that is larger than the system kernel spidev bufsiz.  See issue #997 in this repository as one example.  I have changed that setting in the OS to different buffer sizes, which required root access and a reboot, and then I was able to write a slice up to the new buffer size without any problem, but the above error was thrown as soon as I wrote a large slice.

Would it be possible to have the kernel spidevice buffer size as a property available from System.Device.Spi?  This would be very helpful in a performance scenario where you are pushing a notable amount of data across the SPI bus.

Also, knowing this kernel parameter value and buffer limitation, seems very important to anyone using SpiDevice.Write() for exactly the same reasons I am finding.

The optimal solution seems to me to be having SpiDevice.Write() take any size data without error and slice the ```ReadOnlySpan<Byte>``` up to match the current kernel buffer size.  However, I do not pretend to know the practicality of that from the API standpoint.

As it currently stands, I see only three options:
1) Stick to the kernel's default buffer size of 4K as my slice size and document the error that occurs if someone has a smaller than default buffer size.

2) Use a larger slice size for performance and document that the user of the driver must change their buffer size to no smaller than the size I am using.  This is not desirable.

3) Per @krwq suggestion in discord, use ```int.Parse(File.ReadAllText("/sys/module/spidev/parameters/bufsiz"))``` to get the buffer size and set my slice size accordingly.  This seems to be the least *undesirable* option.

I was seeing a notable performance improvement by using slices larger than 4K.  I would very much like to take advantage of this improvement with at least a 64K buffer but I think requiring the user to change their buffer size to use this E-Paper driver at all is less than optimal. 

Best Regards and thank you very much for your consideration.

Greg

krwq, 2021-03-16:
I assume this setting is global per all SPI devices, right? In that case it probably wouldn't be ideal to have that as an instance property on SpiDevice. Also we do not really know how to support that on Windows.

If we had SpiBus it would probably make sense to put that in there and throw PlatformNotSupportedException on Windows.

IMO best course of action for now would be to add this in Iot.Device.Bindings in a separate class. I.e.

```csharp
public static class SpiBusInfo
{
    private static int s_bufferSize = GetBufferSize();

    // we would need to decide if this should throw when size can't be found or just return -1 or other placeholder
    public static int BufferSize { get { /* ... */ } }

    private static int GetBufferSize()
    {
        // non-throwing version of int.Parse(File.ReadAllText("/sys/module/spidev/parameters/bufsiz"))
        // return -1 on error
    }
}
```

if we ever add SpiBus we would move this there directly

krwq, 2021-03-19:
@thecaptncode we've briefly talked about this during our triage meeting and we're ok with this being added in Iot.Device.Bindings. Shape is ok as I suggested above. We don't know how to support this on Linux yet. Are you interested in creating a PR?

thecaptncode, 2021-03-19:
@krwq just to make sure we are on the same page, are you talking about me forking this repo,, add the stub above as a separate class in https://github.com/dotnet/iot/tree/main/src and creating a pull request?

If so, I can do that.  Is there a desired subfolder and filename?

krwq, 2021-03-19:
Correct. I'd suggest somewhere here: https://github.com/dotnet/iot/tree/main/src/devices/Common/System/Device - perhaps create an `Spi` subfolder.

thecaptncode, 2021-03-19:
Will do.  Awesome!  Thanks a lot :)

Ellerbach, 2021-07-01:
[Triage] This seems closed now. Closing this issue. Do not hesitate to reopen if not fully addressed.

closed 2021-07-01 by Ellerbach