dotnet/iot #1490: Abnormal waveform of Raspberry Pi Linux GPIO driver

ZhangGaoxing, 2021-03-18:
Recently, I tested the speed of Raspberry Pi Linux GPIO driver, and observed the square wave output on the oscilloscope. I found that the output waveform is very strange. It only outputs high level in a very short time, and compared with WiringPi (C), the speed is very very slow (although I know it will be slow, I didn't expect the gap will be so big). I have two questions: 
1. The existing Raspberry Pi driver should have some problems in speed, although it can be used normally. Can we further optimize and simplify the code to improve the running efficiency.
2. At the same time, it can be observed that the speed of `LibGpiodDriver` is not very fast. If there is no logic problem in C# code, can we further improve the execution efficiency of P/Invoke. Of course, I think it should be very difficult to solve this problem.

## Test result

| Drivers| Language | Library Version | Average Frequency |  |
| :-: | :-: | :-: | :-: | :-: |
| RaspberryPi3Driver | C# | - | 363 KHz | [image] |
| SysFsDriver | C# | System.Device.Gpio 1.3.0  | 1.74 KHz | [image] |
| LibGpiodDriver | C# | System.Device.Gpio 1.3.0 <br/> libgpiod 1.2-3 | 198 KHz | [image] |
| WiringPi | C | e9821ab | 17.8 MHz | [image] |

## Test program

### C#

```C#
int pin = 6;
GpioController controller;

Console.WriteLine("Select GPIO driver: ");
Console.WriteLine("1. SysFsDriver; 2. LibGpiodDriver; 3. RaspberryPi3Driver ");

string key = Console.ReadLine();
switch (key)
{
    case "1":
        controller = new GpioController(PinNumberingScheme.Logical, new SysFsDriver());
        break;
    case "2":
        controller = new GpioController(PinNumberingScheme.Logical, new LibGpiodDriver());
        break;
    case "3":
        controller = new GpioController(PinNumberingScheme.Logical, new RaspberryPi3Driver ());
        break;
    default:
        Console.WriteLine("Exit");
        Environment.Exit(0);
        return;
}

using (controller)
{
    controller.OpenPin(pin, PinMode.Output);
    Console.WriteLine("Press any key to exit.");

    while (!Console.KeyAvailable)
    {
        controller.Write(pin, PinValue.High);
        controller.Write(pin, PinValue.Low);
    }
}
```

### C

```C
#include <wiringPi.h>

int main (void) {
  int pin = 2;
  
  wiringPiSetup();
  pinMode(pin, OUTPUT);

  while(1) {
    digitalWrite (pin, HIGH);
    digitalWrite (pin, LOW);
  }

  return 0;
}
```

krwq, 2021-03-18:
@ZhangGaoxing can you try the same measurements but with `while (!Console.KeyAvailable)` replaced with `while (true)`?

Do we have any information what wiring PI uses underneath?

Perhaps one solution would be to create WiringPiGpioDriver...

krwq, 2021-04-30:
Hey @ZhangGaoxing have you pehaps had a chance to try again with `white (true)`? I'm really curious how this will compare

Callum6677, 2021-05-02:
I do not know precisely which pi and linux flavour were in use for the tests above, but I do have access to plenty of oscilloscopes.
With a raspberry pi 4 running headless (SSH'ed in) and a 64 bit beta raspberry pi OS I get the following using the code provided - including Console.KeyAvailable, this is for comparisons sake.

SysFs: 3.25khz
Raspberrypi3Driver 314khz
LibGpioD driver-crash, repeatedly, something up with my linux install and I do not have time to debug this right now.

Both working setups expectedly had a lot of jitter to them, and also several timing pauses large enough to reliably trigger off of with a level dropout trigger on a PicoScope.

With a while(true) instead of checking console.
Sysfs: 3.30khz
RaspberryPi3Driver: 3.70MHz

pgrawehr, 2021-05-02:
@Callum6677 Thanks for doing these measurements. The last number shows we're not doing _everything_ wrong. The remaining discrepancy to WiringPi is probably the additional verifications we do.

Korporal, 2021-05-06:
@ZhangGaoxing 

I have some questions as I've not even looked at NF for some two years and forgot much of what I did learn about it! I have significant electronics skills as I studied the subject in college but never worked professionally, most of my professional work has been software, anyway I'm very much at home with all this just a bit rusty, so my questions:

- What hardware were these test executed on?
- Can I get a short explanation of what the "driver" does? is it something specific to the kind of board and periphs on the board?
- What was the spec of the scope that was used?

Thx

pgrawehr, 2021-05-06:
@Korporal I can try to answer

* The tests where done on a Raspberry Pi 3 or 4 (I suppose 3, but that should be specified somewhere...)
* The "driver" is the low-level module that communicates with the GPIO hardware. The library contains various drivers for various hardware. On the Raspberry Pi 3 different drivers can be used, some of them are known to be rather slow (SysFs) but on the other had they're supported on various linux versions and also on different hardware.
* Why do you think the scope used matters? I've got little doubt that the measurements are incorrect.

Korporal, 2021-05-06:
> @Korporal I can try to answer
>
> * The tests where done on a Raspberry Pi 3 or 4 (I suppose 3, but that should be specified somewhere...)
> * The "driver" is the low-level module that communicates with [...]

@pgrawehr 

Thanks, I was just curious about the scope, the last trace looks sinusoidal with a 17MHz signal so I wondered if the scope's bandwidth was a factor there. I'd expect to see steep rise/fall time on such GPIO pins (assuming they're not in some analog mode of course).

Korporal, 2021-05-06:
But I think basically the loop (C# loop) is a likely contributor to the duty cycle he's seeing in these images. Also we can't assume that the operation setting a pin High takes the same time as the operation that sets the pin Low. I've seen this kind if thing before were the lower level code isn't necessarily symmetrical for various reasons (ideally it should be of course).

Korporal, 2021-05-06:
This might be more sensible as a test block of code:
```

using (controller)
{
    controller.OpenPin(pin, PinMode.Output);
    Console.WriteLine("Press any key to exit.");

    while (true)
    {
        controller.Write(pin, PinValue.High);

        if (Console.KeyAvailable)
           break;

        controller.Write(pin, PinValue.Low);

        if (Console.KeyAvailable)
           break;
    }
}
```

krwq, 2022-06-07:
I'm removing needs author attention labels, this issue is now to track potential improvements in the implementation and particularly figuring out why there is so much perf diff between wiring pi and our RaspberryPi3Driver implementation

pgrawehr, 2022-11-11:
Redid the test, with removal of the `Console.ReadKey` in the loop. Used a Raspberry Pi 4b, 4GB memory, 64 Bit Raspian, System.Device.Gpio V2.2.0

- SysFS: ~3.144Khz
- LibGpioD: ~193Khz
- RaspberryPi3: between 330Khz and 1.1Mhz

It's still slower than the above WiringPi example, but that's also somewhat expected. But since "flipping a pin as fast as possible" is not a very usefull scenario, I think we can close this ticket as "works as designed". The faster the speed is, the more fluctuation we get.

raffaeler, 2022-11-11:
Totally agree. We are probably paying the PInvoke cost.
Since the code is so short, the periodiciy is also impacted by the GC.

Out of my curiosity, @pgrawehr, did you try the debug or the release compilation?

pgrawehr, 2022-11-12:
I used a debug build, but only of the test code. I used the officially released 2.2 build of the library. A release build won't make much difference there.

raffaeler, 2022-11-12:
Probably not here, but I often use Benchmark.NET and you would be surprised to see how the optimizer can do a huge difference.
I also see apparently small and insignificant changes making a huge difference in benchmarks.

Ellerbach, 2023-11-02:
[Triage] As per @pgrawehr point, there is nothing really we can do to drastically improve the performance. There are always margins for improvements but we will never be as fast as native C/C++ implementation compile just for the specific platform.
PWM is good at outputting waves at a very high speed. SPI also. And if you just need a very fast switching for a short period of time, you can use SPI with outputting an array with 0 and 1 bits in it.

closed 2023-11-02 by Ellerbach