dotnet/iot #1502: Tried to go to discord, fail so comment here

figuerres, 2021-03-28:
i need to work with a sensor related to on in here - what a pain!   this mass of code and build scripts is huge and i just want to pull out one device source and update it.... but the web of dependent stuff is way deep..... would help a lot if the devices code was in a separate repo and they just referenced the iot package so that we could avoid a lot of digging around.

and i did try to use that discord link and it stuck me on a form that was locked and a thing saying my invite was bad. and i could not see how to finish a signup for it....

joperezr, 2021-03-28:
Thanks for the feedback and logging the issue. I’m sorry the discord link didn’t work, @Ellerbach do you mind checking what is wrong with the discord link?

As to for building the repo, we have the list of dependencies and steps here https://github.com/dotnet/iot/blob/main/Documentation/CONTRIBUTING.md#building-iot-repo but in general all you need is to run `build.sh` on non-windows or `build.cmd` on windows from the root of the repo and that should build everything. If you keep having issues building that please let us know. As to why we have the bindings in the same place as the repo, we do that in order to avoid infrastructure code duplication, as well as to be able to consume the underlying library directly instead of having to jump extra hoops to do it. This helps us quickly service any issues and ship packages together, as well as makes it really easy to debug.

figuerres, 2021-03-28:
I have a pi hat with a sensor the BMP388  related to the rest of the BMxx8xx family.
so I am just pulling out the code for  the BMxx8XX sensor to build the BMP388 version.
the device folder for that device has a visual studio .sln file so i tried to use that and it fails with all manner of depends on other things....  i now have made a local sln that has resolved all that so i can create the actual code i need.

from what i can tell all the devices in that source folder are not published in any packages ??  so it seems like they should all be moved out to a package where we can just build and update support for a device and not get into the actual IOT library code; at least that is what i am thinking would be a better approach  ?

Ellerbach, 2021-03-29:
> Thanks for the feedback and logging the issue. I’m sorry the discord link didn’t work, @Ellerbach do you mind checking what is wrong with the discord link?

It seems to perfectly work, just did the test. @figuerres I see you managed to join the Discord server!

We can recommend to use the nuget package. At least, it's easier than the source code. If you want to use the source code, then only extract the code of the device you need, change and adjust the project references if needed.

krwq, 2021-03-29:
@figuerres we've considered creating a separate repos for devices but then if we need to update several devices at the same time it would be a bit of a pain to create 20+ PRs... Most of the devices can be built separately using csprojs in their directory - you will need to do the full build at least once before that though

figuerres, 2021-03-29:
when i got the nuget package i did not even see the device objects.  i was searching for info on the sensor and accidentally found that the github site has the files.  so possibly just publish a nuget of iot.devices  that  could be one package?  then i could get that and use the ones i need as base objects to derive from ?? that would let you keep one rep here but just better expose all that code that has the device specific stuff ?

pgrawehr, 2021-03-30:
That exists already. If you just want to use a binding, you don't need to build anything yourself. Just reference the `Iot.Device.Bindings` nuget and you're good to go. The nuget does not include the examples, though.

To build a project or a sample yourself (using the corresponding SLN, as you have tried before), you need to first run the "build.cmd" file in the root of the repository. That prepares all the dependencies. The API documentation is [here](https://docs.microsoft.com/en-us/dotnet/api/?view=iot-dotnet-1.3). 

@joperezr : The documentation should be updated for 1.4.

Ellerbach, 2021-06-17:
[Triage] @figuerres, if there is anything else on this issue please let us know. Otherwise, we will close it in a few days.

closed 2021-06-25 by msftbot[bot]