dotnet/iot #1519: Support Ack Polling for I2C

philrawlings, 2021-04-13:
When writing to an EEPROM (Microchip 24LC256), the device goes into a "busy" state for a few milliseconds when a page of data has been written. It is possible to "poll" the device to determine when it has finished by essentially performing a zero byte write, essentially meaning that the device address is sent on the bus, then if the device is ready it will acknowledge (ACK) and if it is not ready it will not acknowledge (NACK).

The issue is that there needs to be a way to poll the device to determine whether it is ready for the next command. 

Ideal solution: add the following method to I2cDevice: `bool AckPoll()`, which sends only the device address on the bus and returns a boolean matching the ACK/NACK state. This can then be read in a tight loop to determine when the device is ready.
Could probably also be implemented as an awaitable async method with a timeout to do background polling, but the simple implementation is fine.

Note: I am not using the underlying OS support for I2c via `I2cDevice.Create()`, I have derived from I2cBus and I2cDevice to implement a Serial to I2C converter that works on any OS, so all my testing so far has been done using this running on Windows 10. The trivial implementation below _works_, in that a zero byte write that is not acknowleged by the device throws an exception stating "NACK on address" - it just takes too long to throw/catch the exception longer than the busy time of the device), hence the need for a *proper* AckPoll method that doesnt rely on catching exceptions (I am _assuming_ that the OS support for I2C works in the same way): 

```
bool AckPoll()
{
    try
    {
        i2cDevice.Write(new byte[0]);
        return true;
    }
    catch
    {
        return false;
    }
}
```

joperezr, 2021-04-13:
Thanks for adding your proposal @philrawlings! How common is this patern of writing a zero-byte to receive acknowledgement on I2C devices? To me it sounds like it is probably not a common enough scenario, and given the structure of I2CDevice, we would rather want to only expose API that is very general to all I2cDevices (like Write and Read operations). My suggestion for this case, would be to add the AckPoll API directly to the 24LC256 device binding for now, and if we see that this pattern is common enough then we can think about moving it up to I2CDevice itself. Does that sound reasonable?

pgrawehr, 2021-04-14:
@joperezr I think this is related to the way we're currently performing an I2C bus scan: Attempt to read a byte from a device. If there's no reply the driver throws an exception (which is then translated as "no device on this address"). So having a non-throwing implementation of "IsDeviceAvailable" would make sense IMHO.

philrawlings, 2021-04-14:
@joperezr unless I'm missing something I don't think this can be added to the binding, since it relies on the methods exposed by `I2cDevice`, meaning the only way I could implement an AckPoll function in the binding would be like this:

```C#
    public class Mcp24xx256
    {
        private readonly I2cDevice i2cDevice;

        public Mcp24xx256(I2cDevice i2cDevice)
        {
            this.i2cDevice = i2cDevice;
        }

        private bool AckPoll()
        {
            try
            {
                i2cDevice.Write(new byte[0]);
                return true;
            }
            catch
            {
                return false;
            }
        }
    }
```

While this works, the exception throwing is expensive and takes a lot longer that the "busy" time. Therefore there needs to be a non-throwing zero byte write implementation.
As @pgrawehr mentioned, this is also useful when doing a bus scan. Not sure how this is implemented at the moment, but if you do away with catching exceptions it is likely to be much faster.

krwq, 2021-04-14:
Assuming we want it, would it make sense to add this to I2cBus instead of I2cDevice?

philrawlings, 2021-04-14:
@krwq - it can be added to I2cBus as this is useful for the bus scan scenario, however for device bindings it would need to be available at the I2cDevice level also (otherwise the constructor for the `Mcp24xx256` would need to pass in the bus as well as the device as you cant get to the underlying I2cBus from the I2cDevice)

philrawlings, 2021-04-14:
Another useful feature of adding this such that device bindings can use it: a quick check to see if the device is connected, rather than relying on catching exceptions

Ellerbach, 2021-04-14:
I agree, this is a valid scenario. And if added, agree as well it should be on the I2cBus, not on the I2cDevice.

joperezr, 2021-04-15:
[Triage]: This feature is approved. We would like the API to look like the following:

```diff
    public abstract partial class I2cBus : IDisposable
    {
       //..
+      public virtual bool IsDeviceReady(int deviceAddress);
    }

    public abstract partial class I2cDevice : IDisposable
    {
       //..
+      public virtual bool IsDeviceReady();
    }
```

@philrawlings would you be interested in contributing a PR for this addition?

philrawlings, 2021-04-16:
@joperezr  - sure I can have a go! I'll take a look at this next week

philrawlings, 2021-04-17:
I managed to make the necessary changes and tested on a Raspberry Pi 4. I don't have a Windows IOT system so cant test my changes there but I _think_ it should work.

https://github.com/philrawlings/iot/tree/is-device-ready

I haven't submitted a pull request yet as there are a few quirks that warrant discussion.

I implemented the change as a zero byte write, meaning the data sent on the bus looks like this:

[image]

i.e. sends start condition, then the device address, R/W bit set to write, then sends stop condition without transferring any data bytes.

This seems to work well in my limited testing. I then wondered what would happen if I made it a zero byte read (i.e. R/W bit set to read). The problem here is that I _think_the device is holding the SDA line low after sending an ACK, waiting for additional clock cycles - since it is waiting for the master to *not* acknowledge to signify the end of the transfer (which can only happen after one byte has been sent).

[image]

So, we must use write as opposed to read.

I then wondered what happens with the command (on Linux) `i2cdetect -y 1`. This command performs a bus scan from addresses 0x03 to 0x77 to determine which devices are present. This is interesting as the action is different depending on the address. In my case I have an EEPROM connected on address 0x50. 

- For most addresses, write is sent (as I have implemented above) and the ACK/NAK is used to determine whether the device is present.

[image]

- For addresses 0x30 to 0x3F and addresses 0x50 to 0x5F, it actually performs a one byte read, as can be seen below

[image]

In my case, I have an EEPROM at address 0x50 which responds to the address with an ACK. The master then sends 8 additional clock cycles and does NOT send an ACK, signalling the device to stop sending data and release the SDA line.

For non existent devices (e.g. address 0x51 in my case) the fact that the device does not ACK the address means it doesnt needs to additionally send one byte's worth of clock cycles, since the SDA line is not being held low.

While this approach works, it has the annoying affect of incrementing the address pointer on the EEPROM. While this can be worked around by resetting it as needed - it could be confusing. There could be other unintended consequences on different device types.

Therefore, in my mind there are a few options:
1. Implement as a zero byte write (as I have already done). What I don't know is whether this approach is valid for all devices. Obviously there is some reason why whoever developed `i2cdetect` implemented it as such. 
2. Implement as per `i2cdetect`, with different behaviours for different addresses. Documentation would need to be added to describe the potential effects. 
3. An alternative approach, rather than trying to create a high level `IsDeviceReady()` method would be to simply expose non-throwing methods for `Write()` and `Read()`, which return a status (e.g. OK, NackOnAddress, NackOnData). This then shifts the responsibility onto the developer to use the appropriate methods for their use case. Device bindings would neatly hide any of this complexity from higher-level users.

Personally, I think approach 3 is the most powerful and safest, however I appreciate it isn't as "clean" as an `IsDeviceReady() method.

Thoughts appreciated!

pgrawehr, 2021-04-17:
Thanks for this comprehensive analysis! So you're saying that `i2cdetect` is operating differently on different addresses? That's a bit weird. The [Source Code](https://kernel.googlesource.com/pub/scm/utils/i2c-tools/i2c-tools/+/v3.1.2/tools/i2cdetect.c) would suggest it has different detection types, but only uses one of them at a time. 

So far, we did a Read() and then catch the exception to perform a bus scan, and this seems to work quite Ok, but write could work as well - maybe depends on the devices? I guess that whatever we do, it could under some conditions invalidate the state of a device - a corresponding warning is also always displayed by `i2cdetect` (if the `-y` flag is not given). Typically you don't do a bus scan while the device(s) are in use.

For your approach 3, do we have the low-level possibilities to do that? Can we detect what exactly went wrong?

philrawlings, 2021-04-19:
@pgrawehr looks like on line 92 in the code you linked it does choose the mode depending on the address, so that does match the behaviour I observed.

I take the point about a bus scan potentially leaving devices in an unknown state. However - to clarify - the problem I am trying to resolve is polling the device to determine when it is ready to receive the next command, so it is critical that it does not modify the internal state. More efficient bus scanning (not relying on catching exceptions) is just another bonus rather than the prime use case, in my view.

You make a good point about low-level support for the error type. I am not sure whether all hardware supports this. In my case I am creating my own `I2cBus` and `I2cDevice` implementations to implement a USB->I2C converter so I know I _do_ have access to the underlying error. However, in the case of Linux I2C, I couldnt find a good source (quick search) of valid return codes for `int ioctl(int fd, uint request, IntPtr argp);` - I know by testing that when the device acknowledged it returns 1, then returns -1 when it doesnt. However, I dont know if that code changes if the address is acknowledged and data isn't etc. 
Forcing all implementations to return this might be a step too far - in which case, how about this?

```C#

public virtual bool TryWrite(ReadOnlySpan<byte> buffer)
{
    // Default trivial implementation only, derived I2cDevice should ideally override this
    try
    {
        Write(buffer)
        return true;
    }
    catch
    {
        return false;
    }
}

public virtual bool TryRead(Span<byte> buffer)
{
    // Default trivial implementation only, derived I2cDevice should ideally override this
    try
    {
        Read(buffer)
        return true;
    }
    catch
    {
        return false;
    }
}
```

The default implementations are just there as a fallback - if anyone has already inherited from the base classes then it wont break implementations but it is strongly recommended to override. These can be added to I2cBus too (including deviceAddress parameter) if that is desired.

pgrawehr, 2021-04-19:
That sounds good IMHO. Please do add some unit tests so the behavior is tested. The tests in the `System.Device.Gpio.Tests` assembly are executed on actual hardware (both on Windows and Linux) so as soon as you open the PR, you'll see whether everything works, including the Windows implementation.

krwq, 2021-04-20:
> managed to make the necessary changes and tested on a Raspberry Pi 4. I don't have a Windows IOT system so cant test my changes there but I think it should work.

If you can figure out a reasonable test you can add it to CI, it will run on RPi3 on Windows IoT and Raspbian

philrawlings, 2021-04-21:
Pull request submitted.

A few comments:

I have implemented a simple unit test, using the Bme280 device.  However, this only tests the `TryWrite` method in the case where it is expecting to return true.
I didn't want to create other tests as I need a device address which is **guaranteed** to not be in use (so I can check the false case). Can someone comment on a valid address to use and where to define this in the unit tests?

I haven't added `TryWrite()` and `TryRead()` at the bus level. This could be done, however this is a much bigger departure from the current API - which will likely open up questions as to why `Write` and `Read` aren't also implemented at that level. However, at the concrete bus levels, you'll see that `TryWrite()` and `TryRead()` are there but marked as internal - meaning it would be simple to expose them if required in the future. I'm not against bus-level writes and reads, I just thought that it should be a different issue to this one.

philrawlings, 2021-04-21:
Hmm... looks like there were build errors:
`src\System.Device.Gpio\System\Device\I2c\Devices\Windows10I2cBus.cs(57,21): error CS8602: (NETCORE_ENGINEERING_TELEMETRY=Build) Dereference of a possibly null reference.`

However, this builds for me fine locally. I suspect this is due to the following methods added to Windows10I2cBus.cs, these are currently unused but as described above were added for API compatibility with Linux while allowing the option to expose via `I2cBus` in future.

```C#
        internal bool TryRead(int deviceAddress, Span<byte> buffer)
        {
            if (!_devices.TryGetValue(deviceAddress, out Windows10I2cBusDevice? device))
            {
                // Create ephemeral device and dispose when done
                device = new Windows10I2cBusDevice(this, new I2cConnectionSettings(BusId, deviceAddress));
                try
                {
                    return device.TryRead(buffer);
                }
                finally
                {
                    device.Dispose();
                }
            }
            else
            {
                return device.TryRead(buffer);
            }
        }

        internal bool TryWrite(int deviceAddress, ReadOnlySpan<byte> buffer)
        {
            if (!_devices.TryGetValue(deviceAddress, out Windows10I2cBusDevice? device))
            {
                // Create ephemeral device and dispose when done
                device = new Windows10I2cBusDevice(this, new I2cConnectionSettings(BusId, deviceAddress));
                try
                {
                    return device.TryWrite(buffer);
                }
                finally
                {
                    device.Dispose();
                }
            }
            else
            {
                return device.TryWrite(buffer);
            }
        }
```

Two options here:
- Delete these methods which _might_ fix the issue?
- Work out why this is failing when built as part of the CI pipeline. I have no idea, so would welcome someone telling me what I managed to get wrong in this method.

philrawlings, 2021-12-22:
I haven't had much of a chance to look at this in detail, but I have had a few more thoughts.
In summary, the code in the [pull request](https://github.com/dotnet/iot/pull/1525) was trying to send an address with no data and this didnt work because if the write and read buffer lengths are zero, then nothing is sent on the bus and the function returns -1.

Having a quick read of the Linux I2C documentation [here](https://www.kernel.org/doc/Documentation/i2c/dev-interface) seems (to me) to suggest that it _might_ be possible to do this by using the `I2C_M_NOSTART` flag:

```
I2C_M_NOSTART:
    In a combined transaction, no 'S Addr Wr/Rd [A]' is generated at some
    point. For example, setting I2C_M_NOSTART on the second partial message
    generates something like:
      S Addr Rd [A] [Data] NA Data [A] P
    If you set the I2C_M_NOSTART variable for the first partial message,
    we do not generate Addr, but we do generate the startbit S. This will
    probably confuse all other clients on your bus, so don't try this.
```
Putting aside the "don't try this warning" - my initial thought was: what would happen if we send a single write message, with this flag set, with the a single data byte set to the device address left shifted with the write bit set?
Would this get round the issue of nothing being sent on the bus? Would it actually send <start><addr><w> then check whether the device acknowledges?

While this is something I'd like to try, I want to check first whether it is worth the time investment:
1. It seems that the only I2c flags defined in System.Device.Gpio at present are `I2C_M_WR` and `I2C_M_RD`. It seems that maybe not all hardware would support other flags: [https://docs.huihoo.com/linux/kernel/2.6.26/kernel-api/re1210.html](https://docs.huihoo.com/linux/kernel/2.6.26/kernel-api/re1210.html). Therefore, if there a risk that this wont work for all hardware, is there any point adding this functionality?
2. I have no idea at all whether this is possible in Windows IOT, or how it would be implemented. Even if I come up with a solution that works for Linux - is there any point implementing it if it doesnt work on Windows IOT?

pgrawehr, 2021-12-22:
Regarding your last point: We're kind of having a hard time continuing to support Windows IOT. It officially only works on the Pi3 and has never been updated to the Pi4. Event the upcoming Windows 11 for ARM is (as it currently looks) not supported on the Raspberry Pi (any version). The information you find about it is very poor. Advertisements and "developer resources" everywhere, but no operating system image to download and test. 

I therefore think we currently don't have to worry about windows support, but maybe @joperezr can get some information about whether there will one time be a new Windows Version for a Raspi (or a similar device).