dotnet/iot #1522: LibgpiodDriverEventHandler should support rising and falling edge detection

flitjens, 2021-04-16:
**Is your feature request related to a problem? Please describe.**
Using ALRT pin from the Ads1115, which signals that a conversion is ready to be obtained, requires the use of falling edge detection which is not supported at this moment. The LibgpiodDriverEventHandler only supports both edge detection. This is not sufficient for using the ALRT pin of the Ads1115. 

**Describe the ideal solution**
LibgpiodDriverEventHandler  should support rising, falling edge detection of gpio pins.

pgrawehr, 2021-04-17:
Can you describe your use case a bit more in detail? How do you register for the event? I'm quite sure I had this tested and there exist unit tests for this scenario. Additionally, the PinValueChangedEventArgs contains a field that tells you whether the event happened on a falling or a rising event, so that it should be possible to filter on that, even if both edges fire the event.

flitjens, 2021-04-23:
Hi @pgrawehr,

My configuration is as follows:

Raspberry PI4 8G

Ads1115 connected directly to the PI:
Ads1115 - SDA -> PI GPIO 8
Ads1115 - SCL -> PI GPIO 9
Ads1115 - VCC -> PI Pin 1
Ads1115 - GND ->PI Pin 9 
Ads1115 - ALRT -> GPIO 14

`public Sampler()
{
    I2cConnectionSettings settings = new I2cConnectionSettings(1, (int)I2cAddress.GND);
    I2cDevice device = I2cDevice.Create(settings);

    Sampler.GpioController = new GpioController(PinNumberingScheme.Logical);
    _adc = new Ads1115(device, Sampler.GpioController, 14, true, InputMultiplexer.AIN0, MeasuringRange.FS6144);
    
    _adc.AlertReadyAsserted += OnConversionReady;
    _adc.EnableConversionReady();

    Task.Delay(20000).Wait();

    _adc.Dispose();

    WriteSamplesToFile("sin10Hz0_7Vrms_1_36VOffset_BothEdges.dat").Wait();
}

void OnConversionReady()
{
    // read raw data form the sensor
    short raw = _adc.ReadRaw();
    // raw data convert to voltage
    var voltage = _adc.RawToVoltage(raw).Volts;

    _samples.Add(voltage);
}
`

However when I log the events reported by Interop.libgpiod.gpiod_line_event_read in the Task returned in InitializeEventDetectionTask only returns Rising edges I see only rising edges detected. The OnConversionReady  callback is never called since the ValueRising event handler in LibGpiodDriverEventHandler is null, which is correct, and the ValueFalling event handler is the eventhandler registered. Libgpiod should queue all events, which includes the falling, but I don't see them.

I have a working solution available in which you can subscribe for Rising, Falling and both, which works perfect.

pgrawehr, 2021-04-24:
Your code is very similar to the code of example 3 [here](https://github.com/dotnet/iot/blob/main/src/devices/Ads1115/samples/Program.cs), which I had tested myself and which worked fine **with some extra electronics in place**. Please have a look at the documentation https://github.com/dotnet/iot/blob/main/src/devices/Ads1115/README.md. My observation back then was that the pulses generated by the ADS1115 where just to short to be recognized and I had to add external hardware (basically a flip-flop) to be able to properly recognize them. So this is likely not a software problem, but a hardware limitation.

If you say you have a working solution in place, can you tell me how that looks? Maybe we need some special tricks to be able to recognize the short pulses of the ADS1115.

flitjens, 2021-05-14:
Well you are probably right that the PI is not able to recognize pulses with a length of approximately 8us, as specified in the ADS1115 datasheet. The PI has also some internal hardware to avoid reacting on glitches. However the ADS1115 datasheet also mentions that the conversion is ready on the falling edge of the ALERT/RDY pin, see [https://www.ti.com/lit/ds/symlink/ads1113.pdf?ts=1620894037088&ref_url=https%253A%252F%252Fwww.google.com%252F](url) page 19. The ADS1115 is also configured using falling edge detection, but under the hood it uses the combination of falling and rising edge for which only the rising edges are detected due to the short pulse length. So the solution, in my opinion, is not adding extra hardware but configuring the libgpiod driver to detect only falling edges,. and  this is wat my solution exactly does. It implements gpiod_line_request_falling_edge_events and handle these events in LibGpiodDriverEventHandler where you can enter the PinEventType as PinEventTypes.Falling. For completeness I also added the rising part.

This is according to my opinion the proper solution.

I would like to add this into the code base.

pgrawehr, 2021-05-14:
I'm not sure this will improve the situation, but maybe you are right. If you open a pull request, it will be easier to understand what you intend to change. If it does not break existing code, it's likely that we'll accept the change.

flitjens, 2021-05-28:
I created a pull request. I double checked everything, so I'm looking forward for any comments.

Ellerbach, 2021-06-17:
[Triage], link to #1561