dotnet/iot #1524: GrovePi Relay not reliable, sometimes stays on event when switched off

wojciechczyz, 2021-04-17:
**Describe the bug**

Have not run extensive test, yet seen that on various ocations Relays are not always turned off.
I have two relays and pulsing led connected to the grovepi. led is pulsing on and off every 2 secs, relays are turned on and then off after short delay. I see they are not always turned off.

**Steps to reproduce**

```csharp
// put sample code here
I2cConnectionSettings i2CConnectionSettings = new I2cConnectionSettings(1, Iot.Device.GrovePiDevice.GrovePi.DefaultI2cAddress);
                var i2cDevice = I2cDevice.Create(i2CConnectionSettings);
                UsedIGrovePi = new Iot.Device.GrovePiDevice.GrovePi(i2cDevice);
string GrovePiInformation = "GrovePi with version: " + UsedIGrovePi.GetFirmwareVersion();

                    Output1IRelay   = new Iot.Device.GrovePiDevice.Sensors.Relay(   UsedIGrovePi, GrovePort.DigitalPin2);
                    Output2IRelay   = new Iot.Device.GrovePiDevice.Sensors.Relay(   UsedIGrovePi, GrovePort.DigitalPin3);
                    OutputLed       = new Iot.Device.GrovePiDevice.Sensors.Led(     UsedIGrovePi, GrovePort.DigitalPin8);

//ondemand
                Output2IRelay.On(); 

//ondemand
                Output2IRelay.Off(); 

**Expected behavior**

<!-- Describe what you expected to happen -->

**Actual behavior**

<!-- Describe what actually happened -->

**Versions used**

<!-- Please add the following information:

- `dotnet --info` on the machine being used to build
- `dotnet --info` on the machine where app is being run (not applicable for self-contained apps)
- Version of `System.Device.Gpio` package
- Version of `Iot.Device.Bindings` package (not needed if bug is in `System.Device.Gpio`)
-->

joperezr, 2021-05-06:
cc: @Ellerbach

Ellerbach, 2021-05-08:
Hi @wojciechczyz and thanks for the detail issue. I'll need to investigate with my GrovePi board which I don't have with me currently.
GrovePi is using I2C to communicate with the Raspberry PI and it sometimes fails. There is a retry mechanism in the communication protocol. So if you're using multi threading, you may create this discrepancy.
Note that GrovePi is not thread safe. If you want to make sure a relay is really switch off, please read its status with a DigitalRead on the port.

Ellerbach, 2021-06-28:
Hi @wojciechczyz can you confirm that you are not running in parallel threads? 
I found 2 of those relays, the main board and I can try to find some time in next few weeks to check this. But if you're using multiple threads, then for sure, you'll face concurrency issues as GrovePi is not thread safe.

wojciechczyz, 2021-07-06:
Is it possible to add code in API (maybe activated as additional parameter, or configured before accessing grovepi) that would provide necessary blocks / waiting so that using it across different threads would be safe?

Ellerbach, 2021-07-06:
> Is it possible to add code in API (maybe activated as additional parameter, or configured before accessing grovepi) that would provide necessary blocks / waiting so that using it across different threads would be safe?

That would be possible but will work mainly with lock and other elements like this which will slow down quite a lot the execution. So it's better then on the high level code, so your code, to use this mechanism. In that case, it will be targeted to your usage and not slowing down all the execution.

Ellerbach, 2023-01-21:
Closing this issue as this should be achieved in user code, not in the binding code.

closed 2023-01-21 by Ellerbach