dotnet/iot #1540: VL53L0x: Creation of VL53L0X instance resets the i2c address

Reto411, 2021-05-18:
Hi

I have connected multiple VL53L0x sensors on the I2C Bus of a raspberry pi.
I initialize them this code snippet:
```
// outputs are the shutdown pins of the gpio
// shutdown all
foreach (var output in outputs)
{
    output.SetOutputState(false);
}
Thread.Sleep(10);

// enable all
foreach (var output in outputs)
{
    output.SetOutputState(true);
}

// reset all except output1
foreach (var output in outputs)
{
    if (outputs[0] != output)
    {
        output.SetOutputState(false);   
    }
}
Thread.Sleep(10);

int i = 0;
// Setup Addresses
foreach (var oneConfig in nameAddressPin)
{
    var sensorName = oneConfig.Item1;
    var address = oneConfig.Item2;
    var output = outputs[i];
    output.SetOutputState(true);
    Thread.Sleep(10);
    var connectionSettings = new I2cConnectionSettings(busId, Vl53L0X.DefaultI2cAddress);
    var tmpDevice = I2cDevice.Create(connectionSettings);
    Vl53L0X.ChangeI2cAddress(tmpDevice, (byte)address);
    i++;
}
i = 0;
// // Create the TOF devices
foreach (var oneConfig in nameAddressPin)
{
     var sensorName = oneConfig.Item1;
     var address = oneConfig.Item2;
     var output = outputs[i];
     var connectionSettings = new I2cConnectionSettings(busId, address);
     var tmpDevice = I2cDevice.Create(connectionSettings);
     var sensor = new Vl53L0X(tmpDevice);
     this.nameSensorPin.Add(new Tuple<string, Vl53L0X, DigitalOutput>(sensorName, sensor, output));
     i++;
}
```
At the line: "var sensor = new Vl53L0X(tmpDevice)" the "System.IO.IOException: Error 121 performing I2C data transfer." is thrown.

On checking the i2c bus with i2cdetect, the address of the is set back to 0x29
The whished result would be
```
      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- 2b 2c 2d -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

```

I was able to fix the issue by removing the soft reset of the sensor in the constructors:
https://github.com/dotnet/iot/blob/699f2cb29c5201e55b1de780485c7475cfdabf23/src/devices/Vl53L0X/Vl53L0X.cs#L74

Maybe a additional optional constructor parameter is needed to decide whether the soft reset shall be executed or not.

Best Regards

joperezr, 2021-05-27:
cc: @Ellerbach can you take a look?

Perhaps we should take a look at why Dispose/Reset changes the address.

Ellerbach, 2021-05-27:
@Reto411 Thanks for your submission and the details.
I understand the scenario and the issue. What we can then do is expose the Reset function and not place it in the constructor. So we keep the constructor like it is. You then can decide, depending on the scenario to call soft Reset or not.
If that will work for you, I can submit a PR to adjust this.

Reto411, 2021-05-27:
@Ellerbach Thank you for investigating it. I think your suggestion will work for me. I hope it won't break to much code of other peoples, which are depending on the soft reset in the constructor.

https://github.com/dotnet/iot/blob/e0dbd643875189804da3f090fd26227f9b01a13c/src/devices/Vl53L0X/README.md
Maybe you could add there the soft reset, because with only one sensor most of the times it makes sense to reset it.

Additionally it might be helpful to have there an example with multiple sensors. At least that would have helped me. It's your decision. If wanted you could use snippets of my code.

Best Regards
Reto

Ellerbach, 2021-05-28:
@Reto411 you have a PR to review. That should work. Soft Reset reset everything to the initial state, unfortunately including the I2C address. So correct, it should not be in the constructor. As all the rest is initialized properly in the constructor, the reset is not needed.
The reset function should still be exposed to make sure there is a way to reset the sensor even if that changes back to the default address.

Ellerbach, 2021-06-17:
[Triage] closed by adjusting the API and merging the PR

closed 2021-06-17 by Ellerbach