dotnet/iot #1559: RegisterCallbackForPinValueChangedEvent Exception

bkriegerwinsystems, 2021-05-27:
**Describe the bug**

The RegisterCallbackForPinValueChangedEvent method throws an exception that says  'The data area passed to a system call is too small'

This is run on an Apollo Lake platform running Windows 10 IoT LTSC 1809. The platform's GPIO interrupts were validated using Windows.Devices.Gpio in a UWP app.

**Steps to reproduce**
The following console app throws the exception at RegisterCallbackForPinValueChangedEvent 
```csharp
using System;
using System.Collections.Generic;
using System.Linq;
using System.Text;
using System.Threading.Tasks;
using System.Device.Gpio;

namespace ConsoleApp1
{
    class Program
    {
        static void Main(string[] args)
        {

            GpioController controller = new GpioController();  //GPIO controller
            int inputPin1 = 0;
            PinChangeEventHandler OnInputPinValueChanged = (object sender, PinValueChangedEventArgs arguments) => { Console.WriteLine("interrupt"); };

            controller.OpenPin(inputPin1);                            //power input
            controller.SetPinMode(inputPin1, PinMode.Input);
            controller.RegisterCallbackForPinValueChangedEvent(inputPin1, PinEventTypes.Falling | PinEventTypes.Rising, OnInputPinValueChanged); //Exception thrown
        }
    }
}

```

**Expected behavior**

It should register a callback for whenever a falling or rising edge occur.

**Actual behavior**

I get the following exception on the method:
System.Exception
  HResult=0x8007007A
  Message=The data area passed to a system call is too small.

An unexpected error occurred when communicating with the device.
  Source=Windows.Devices
  StackTrace:
   at Windows.Devices.Gpio.GpioPin.add_ValueChanged(TypedEventHandler`2 handler)
   at System.Runtime.InteropServices.WindowsRuntime.WindowsRuntimeMarshal.NativeOrStaticEventRegistrationImpl.AddEventHandler[T](Func`2 addMethod, Action`1 removeMethod, T handler)
   at System.Runtime.InteropServices.WindowsRuntime.WindowsRuntimeMarshal.AddEventHandler[T](Func`2 addMethod, Action`1 removeMethod, T handler)
   at System.Device.Gpio.Drivers.Windows10DriverPin.AddCallbackForPinValueChangedEvent(PinEventTypes eventTypes, PinChangeEventHandler callback)
   at System.Device.Gpio.Drivers.Windows10Driver.AddCallbackForPinValueChangedEvent(Int32 pinNumber, PinEventTypes eventTypes, PinChangeEventHandler callback)
   at System.Device.Gpio.GpioController.RegisterCallbackForPinValueChangedEvent(Int32 pinNumber, PinEventTypes eventTypes, PinChangeEventHandler callback)
   at ConsoleApp1.Program.Main(String[] args) in ConsoleApp1\Program.cs:line 21

**Versions used**

- `dotnet --info` on the machine being used to build:

.NET SDK (reflecting any global.json):
 Version:   5.0.300
 Commit:    2e0c8c940e

Runtime Environment:
 OS Name:     Windows
 OS Version:  10.0.18363
 OS Platform: Windows
 RID:         win10-x64
 Base Path:   C:\Program Files\dotnet\sdk\5.0.300\

Host (useful for support):
  Version: 5.0.6
  Commit:  478b2f8c0e

.NET SDKs installed:
  3.1.402 [C:\Program Files\dotnet\sdk]
  5.0.300 [C:\Program Files\dotnet\sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.All 2.1.22 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.App 2.1.22 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.1.8 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 5.0.6 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 2.1.22 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.1.8 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 5.0.6 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.WindowsDesktop.App 3.1.8 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 5.0.6 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]

- `dotnet --info` on the machine where app is being run:

Host (useful for support):
  Version: 5.0.6
  Commit:  478b2f8c0e

.NET SDKs installed:
  No SDKs were found.

.NET runtimes installed:
  Microsoft.NETCore.App 5.0.6 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.WindowsDesktop.App 5.0.6 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]

- Version of `System.Device.Gpio` package is 1.4.0

Ellerbach, 2021-06-17:
[Triage] This is related to #1411 even if it seems not to be the exact same issue as in your case, you still manage to get a driver.
Did you try if blinking a led, reading, writing is working properly? Or the driver is really not working at all.
We don't have this specific hardware to check and is necessary to debug and understand.
The provided low level driver may not support some of the higher level API we are calling. In the case the basic read/write are working, you may implement some events by looping and reading all the time the values. Not perfect but that should solve your issue.

bkriegerwinsystems, 2021-06-28:
The driver works with the uwp gpio api (https://docs.microsoft.com/en-us/uwp/api/windows.devices.gpio?view=winrt-20348). I am able to write to it, read it, and use interrupts with it. If I am able to detect edges using those function calls, should I be able to with this library?

Ellerbach, 2021-06-28:
> The driver works with the uwp gpio api (https://docs.microsoft.com/en-us/uwp/api/windows.devices.gpio?view=winrt-20348). I am able to write to it, read it, and use interrupts with it. If I am able to detect edges using those function calls, should I be able to with this library?

Correct, then it *should* just work. That will require some investigation and a similar board to debug. The best I can suggest you is to clone the repo, use a project reference based to the System.Device.Gpio and try to debug to see where and why it breaks.