dotnet/iot #1570: MFRC522 NTAG213, NTAG215 Support?

ismail-ozturk, 2021-06-09:
Hello ,
mfrc522 cannot read ntag213 and ntag215 with ListenToCardIso14443TypeA. is this normal?

Ellerbach, 2021-06-09:
> mfrc522 cannot read ntag213 and ntag215 with ListenToCardIso14443TypeA. is this normal?

When you "cannot read", what does it mean exactly?

Those small tags are not implemented in a high level, so they don't have a high level support like some of the Mifare or CreditCard for example (note: CreditCard won't work with MFRC522 as they are Type B).

You will have to manually send the commands to read/write 4 bytes by 4 bytes. See the detailed documentation:
https://www.nxp.com/docs/en/data-sheet/NTAG213_215_216.pdf#:~:text=NTAG213%2C%20NTAG215%20and%20NTAG216%20%28from%20now%20on%2C%20generally,product%20authentication%2C%20NFC%20shelf%20labels%2C%20mobile%20companion%20tags.

Ellerbach, 2021-06-09:
PS: I think I recently bough few of those ultralight cards. I will try to check if I have the reader with me.

ismail-ozturk, 2021-06-09:
Mifare Classic 1K cards are detected by mfrc522 reader. but It cannot detect tags of type NTAG213 or NTAG215.
the library only detects mifare cards?
`Data106kbpsTypeA card;`
`bool result = mfrc522.ListenToCardIso14443TypeA(out card, TimeSpan.FromSeconds(2));`

The result variable always returns false for NTAG213 and NTAG215

Ellerbach, 2021-06-09:
> The result variable always returns false for NTAG213 and NTAG215

I think I understand why. So if you are willing to help as I don't have the MFRC522 reader with me, it would be good if you could clone this repo and use directly the code and check what's happening in the ListenToCardIso14443TypeA method.
The method is waiting for a UUID which I think you'll get right away looking at the spec of those cards but you may not get the rest.
If you could look at this, that may help to find a fix for those cards.

Ellerbach, 2021-06-20:
So I finally get access to the MFRC522 and some cards. The Ultralight V1 and NTAG213 are perfectly detected. No issue at all.
The rest of the sample is of course not working as it's not doe for ultralight.
If I have a bit of time, I'll add read and write support for the ultralight cards. Please make sure you have a good source for the 3.3V, the read and write operation are consuming quite a bit of energy. That maybe the issue.
So closing this issue.

closed 2021-06-20 by Ellerbach