dotnet/iot #1575: Asus Tinker

mrccarvalho, 2021-06-23:
**Cannot Access/Communicate with Asus Tinker GPIOs / Can Use this Library With Asus Tinker**

Thanks,
Best Regards,
Roger

ZhangGaoxing, 2021-06-24:
You can try this and refer to the datasheet for pin mapping
https://github.com/dotnet/iot/tree/main/src/devices/Gpio/Drivers/Rockchip

mrccarvalho, 2021-06-24:
> 
> 
> You can try this and refer to the datasheet for pin mapping
> https://github.com/dotnet/iot/tree/main/src/devices/Gpio/Drivers/Rockchip

Thanks ZhangGaoxing, 
I will investigate about this and make a test with it.

mrccarvalho, 2021-07-09:
After doing some tests, i find that it is possible to read and write to asus tinker's gpio pins. 

I use the 163 pin number that corresponds to the physical pin number 18  on the tinker board.
Can confirm that the next code wotks fine.

`using System;
using System.Device.Gpio;
using System.Threading;

Console.WriteLine("Blinking LED. Press Ctrl+C to end.");
int pin = 163;   // 163 corresponds to the physical pin number 18  on tinker board
using var controller = new GpioController();
controller.OpenPin(pin, PinMode.Output);
bool ledOn = true;
while (true)
{
    controller.Write(pin, ((ledOn) ? PinValue.High : PinValue.Low));
    Thread.Sleep(1000);
    ledOn = !ledOn;
}`

Thanks!
Best Regards!

Ellerbach, 2021-07-09:
Great news! So closing this issue.

closed 2021-07-09 by Ellerbach