dotnet/iot #1577: hx711

mrccarvalho, 2021-06-27:
**I need to develop a project for Weigh Scales**

I would like to use dotnet core.
I need to use an hx711 device with an asus tinker or raspberry Pi board.

 I see that in master there is no hx711 in device list.

I have read [#770](https://github.com/dotnet/iot/pull/770).

Is there any chances to have this device working properly ?
Is there any plan to add hx711  ?

Thanks

Ellerbach, 2021-06-28:
Hi, the bindings are mainly community driven. so if a community member do a PR with a proper implementation, you'll get it!
In the issue you mentioned you have a link on some code. That may work for you.
PR welcome!

mrccarvalho, 2021-06-28:
Hi!
Thanks @Ellerbach. 

My idea was to try out that code, but in the conversation about the issue, I also got the idea that there was some problem in the code.

I don't really understand why the PR of the issue (#770) was closed.

Anyway, I'll try it and then give feedback. 

PR welcome!

Thanks.

Ellerbach, 2021-06-28:
> I don't really understand why the PR of the issue (#770) was closed.

My understanding is that it's a timing issue. See comment https://github.com/dotnet/iot/pull/770#issuecomment-539039826

I would personally recommend to do this manipulation with SPI rather than GPIO. SPI have a clock, the MOSI pin is synchronized with it, you can then get to the expected timing. Just prepare the buffer you want to output in memory and sent it. That should just work fine.

Ellerbach, 2021-06-28:
It should be very very straight forward to implement. A lot of bindings like this one have the 2 interfaces. Check the LCD ones, it's a similar case.

closed 2022-09-29 by krwq