dotnet/iot #1584: Reproducible Error 121 performing I2C data transfer when using multiple devices on the same bus

nh43de, 2021-07-08:
I am recieving an error 121 with writing to an I2c bus, but only when two devices, an i2c sensor and i2c pwm controller are connected. When only the sensor is connected to the i2c bus, dotnet code can read from it once without failing, and then fails on subsequent writes. When two devices are connected to the same i2c bus, it fails immediately when attempting to read from the sensor. The pwm controller has no issues whatsoever, whether the sensor is connected or not.

**Setup**

- Two I2c devices on bus 1 on Raspberry Pi: one at address x70 and one on x40
- x40 device is a PWM controller PCA9685
- x70 Device is a temp/hum sensor, SHTC3

I can verify that the two devices are present with I2cdetect

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: 70 -- -- -- -- -- -- --
```

I can even get the raw values using i2cget

```
pi@orchidbox:~/TestDevice $ i2cget -y 1 0x70 0x00
0x21
pi@orchidbox:~/TestDevice $ i2cget -y 1 0x70 0x01
0x04
pi@orchidbox:~/TestDevice $ i2cget -y 1 0x70 0x02
0xe2
pi@orchidbox:~/TestDevice $ i2cget -y 1 0x70 0x03
0xe4
```

I can also confirm that i2cwrite succeeds. However, when running the sample code for Shtc3 ([https://github.com/dotnet/iot/tree/main/src/devices/Shtc3](https://github.com/dotnet/iot/tree/main/src/devices/Shtc3)), I get the 121 error immediately with this setup.

```
sudo dotnet Shtc3.Samples.dll
Sensor Id:

Unhandled exception. System.IO.IOException: Error 121 performing I2C data transfer.
   at System.Device.I2c.UnixI2cBus.WriteReadCore(UInt16 deviceAddress, Byte* writeBuffer, Byte* readBuffer, UInt16 writeBufferLength, UInt16 readBufferLength)
   at System.Device.I2c.UnixI2cBus.Write(Int32 deviceAddress, ReadOnlySpan`1 buffer)
   at System.Device.I2c.UnixI2cDevice.Write(ReadOnlySpan`1 buffer)
   at Iot.Device.Shtc3.Shtc3.Write(Register register)
   at Iot.Device.Shtc3.Shtc3.TryReadSensorData(Register cmd, Double& temperature, Double& humidity)
   at Iot.Device.Shtc3.Shtc3.TryGetTemperatureAndHumidity(Temperature& temperature, RelativeHumidity& relativeHumidity, Boolean lowPower, Boolean clockStretching)
   at <Program>$.<Main>$(String[] args) in C:\Users\Nathan Hollis\Desktop\samples\Program.cs:line 18
Aborted
```

When I disconnect the device at address x40 and restart the process, the sensor is able to be read twice (once in normal mode, and once in low power mode), but then fails with the 121 error, the same as above. The sample code can be found here: [https://github.com/dotnet/iot/blob/main/src/devices/Shtc3/samples/Program.cs](https://github.com/dotnet/iot/blob/main/src/devices/Shtc3/samples/Program.cs)

I've tried another i2c temp/hum sensor (AHT10) but still have the same error.

Ideally, this should succeed, and without the ability to interface properly with the sensor in dotnet code makes it problematic.

**Versions used**

```
 dotnet --info

.NET SDK (reflecting any global.json):
 Version:   5.0.301
 Commit:    ef17233f86

Runtime Environment:
 OS Name:     raspbian
 OS Version:  10
 OS Platform: Linux
 RID:         linux-arm
 Base Path:   /opt/dotnet/sdk/5.0.301/

Host (useful for support):
  Version: 5.0.7
  Commit:  556582d964

.NET SDKs installed:
  5.0.301 [/opt/dotnet/sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.App 5.0.7 [/opt/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 5.0.7 [/opt/dotnet/shared/Microsoft.NETCore.App]

To install additional .NET runtimes or SDKs:
  https://aka.ms/dotnet-download
```

Some packages I'm using:

```
    <PackageReference Include="Iot.Device.Bindings" Version="1.4.0" />
    <PackageReference Include="Microsoft.Extensions.Logging.Abstractions" Version="5.0.0" />
    <PackageReference Include="Microsoft.Extensions.Options" Version="5.0.0" />
    <PackageReference Include="System.Collections" Version="4.3.0" />
    <PackageReference Include="System.Data.SqlClient" Version="4.8.2" />
    <PackageReference Include="System.Device.Gpio" Version="1.4.0" />
    <PackageReference Include="UnitsNet" Version="4.89.1" />
```

Raspbian Environment info

```
 cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
```

krwq, 2021-07-08:
1. Can you share the code you're using for testing
2. Have you tried using both devices at the same time from the command prompt or other language without issues? Some devices are implemented badly and may interfere with how the bus works

Ellerbach, 2021-07-08:
[Triage] On top of the previous questions:

* Do you have any level shifter in your chain? They typically create those issues
* This 121 error is really about the device not giving an acknowledge which can be cause by a wrong bus pull up. So you may want to check if you don't have any additional pull up.  And please check that it is in the correct values. While only addressing 1 device, the behavior is different than when trying to address both.
* What are the exact model of the devices you are using? So we may have a look at the design and understand if that can come from there. Example: the RPI already has an internal pull up so if you have one of your device with an extra pull up, you can go thru those issues.

Bottom line, and that's why in the previous questions, we ask you to test with the OS commands or with another language if you have similar issues.

nh43de, 2021-07-15:
@Ellerbach @kwkraus 

I was able to resolve the issue by switching the PCA9685 to address 0x41 by soldering a jumper, and then using an Si7021 sensor on address 0x40. 

If you want to try to reproduce, original hardware listed above:

Two I2c devices on bus 1 on Raspberry Pi 3 model B+:
- x40 device is a PWM controller PCA9685
- x70 Device is a temp/hum sensor, Shtc3

Using these devices you should be able to reproduce the error. When looking at a logic analyzer, everything looked ok. Also, OS commands to read and write to the device were successful.

Ellerbach, 2021-07-17:
Thanks a lot @nh43de for the details. Challenge now will be to get the devices and go in the root of the issue. So not sure we will be able to debug this one shortly.
Good news is that you found a solution. And thanks again for the details. I4ll place this bug as blocked as we need more hardware on our side to reproduce it.

pgrawehr, 2022-10-02:
This appears to be caused by hardware that somehow fails to properly follow the I2C protocol, and therefore causes bus timeouts. Not something that can be addressed in software. 

I'll close this, please reopen if there appears something that needs fixing in software.

closed 2022-10-02 by pgrawehr