dotnet/iot #1585: VideoDevice.CaptureContinuous low FPS

CodedBeard, 2021-07-08:
**Describe the bug**

I was attempting to use the continuous capture on a raspberry pi 4b to stream from the attached camera. When running the app I was only getting 2 FPS from the camera, where as, if I use the inbuilt raspivid tool I get 30 FPS from the same camera. After confirming that I wasn't doing anything wrong, I went digging in the `VideoDevice` source code and found the cause of the slowness. Both the `Capture` and `CaptureContinuous` methods are calling the same `ProcessCaptureData` method, which opens and closes the data stream for every call. In my measurements, the opening and closing of the stream takes around 400ms, which explains why the `CaptureContinuous` is not able to reach a usable frame rate, as most of the time is spent repeatedly opening and closing the stream.

To confirm my suspicion, [I forked the repo](https://github.com/CodedBeard/iot/commit/cf0e4439fe8f9e686c139f76538c3395fc3ea308) and added a new method `CaptureContinuousBytes` that keeps the stream open for the length of the continuous capture. I also added another method `GetFrameDataPooled` for processing the frame data, which uses `ArrayPool<byte>` to reduce the allocations and thus GC pressure. I also avoided using `Stream` as in my use case it was causing uneccessery allocations converting the `byte[]` to `Stream` and then immediately calling ToArray on the stream.

With these changes, while its still not quite as fast as raspivid's framerate, it is at least hitting 20 FPS. I'd be happy to create a PR to have these changes incorrporated into the next version, but thought it best to open an issue for discussion first, as I suspect my current implementation isn't the best.

**Steps to reproduce**

This can be reproduced using the [MJPEG example project](https://github.com/dotnet/iot/tree/main/src/devices/Media/VideoDevice/samples/MjpegStream) in the repo. 

**Expected behavior**

The framerate matches that of the inbuilt raspivid tool

**Actual behavior**

Iot can only manage 2 FPS

**Versions used**

```
.NET SDK (reflecting any global.json):
 Version:   6.0.100-preview.5.21302.13
 Commit:    d6380bcae7

Runtime Environment:
 OS Name:     debian
 OS Version:  10
 OS Platform: Linux
 RID:         debian.10-arm64
 Base Path:   /home/pi/dotnet/sdk/6.0.100-preview.5.21302.13/

Host (useful for support):
  Version: 6.0.0-preview.5.21301.5
  Commit:  ec3e0b276b

.NET SDKs installed:
  6.0.100-preview.2.21155.3 [/home/pi/dotnet/sdk]
  6.0.100-preview.5.21302.13 [/home/pi/dotnet/sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.App 6.0.0-preview.2.21154.6 [/home/pi/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 6.0.0-preview.5.21301.17 [/home/pi/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 6.0.0-preview.2.21154.6 [/home/pi/dotnet/shared/Microsoft.NETCore.App]
  Microsoft.NETCore.App 6.0.0-preview.5.21301.5 [/home/pi/dotnet/shared/Microsoft.NETCore.App]

To install additional .NET runtimes or SDKs:
  https://aka.ms/dotnet-download
```
System.Device.Gpio version 1.4
Iot.Device.Bindings version 1.4

Ellerbach, 2021-07-08:
[Triage] Thanks @CodedBeard for the issue submission and for the analyze and to propose a PR.
Please pass the PR, we can then look at the implementation. To be honest the continuous video implementation was done quite quickly without performance in mind at al. So any improvement on this side is more than welcome.

CodedBeard, 2021-07-09:
@Ellerbach I've already found an issue with my thrown together implementation, so will make sure to resolve that first, so it at least works properly in my use case, and then will send a PR over.

PR #1587 (2021-08-28): Improve performance of VideoDevice continuous capture
Fixes #1585 

The intent of this PR is to provide a more performant implementation of the continuous capture methods. This has been tested on a raspberry pi 4b, and was able to provide a 25fps MJPEG stream to multiple clients.

I've not modified any of the existing methods, as I wasn't sure on the backward compatability policy.

###### Microsoft Reviewers: [Open in CodeFlow](https://portal.fabricbot.ms/api/codeflow?pullrequest=https://github.com/dotnet/iot/pull/1587)
closed 2021-08-28 by pgrawehr