dotnet/iot #1616: Create a plan for joint maintainership of the iot repo

terrajobst, 2021-07-26:


terrajobst, 2021-07-26:
> @terrajobst ? I don't see the list of repos in the themesof repo, so I'm guessing it's local to your app.

Sorry, didn't see the comment. See [this document](https://themesof.net/about). themesof.net only looks in certain repos for themes/epcis/user stories, but that's the starting set. Any issue that is linked from a theme/epic/user story can be found. In practice this means only the repos with themes need to be known to themesof.net. Anything else is discovered from there.

krwq, 2022-04-28:
[Triage] @richlander is there anything we need to do here to improve the current state? Can we close this now?

richlander, 2022-05-03:
We can close it.

We did the following:
- @raffaeler and @pgrawehr have triager or commiter roles.
- We changed the signing cert to .NET Foundation (not Microsoft).

closed 2022-05-03 by richlander