dotnet/iot #1618: BME280 Error performing I2C data fransfer

awrobel196, 2021-07-28:
Hello

I want to use Bme280 sensor with Raspberry 4. For try get data from sensor i use [example Bme280 code](https://github.com/dotnet/iot/blob/main/src/devices/Bmxx80/samples/Bme280.sample.cs) and little edit it [for use in my project](https://pastebin.com/y6wS6n4P)

**Expected behavior**
no error

**Actual behavior**
I run program on Raspberry Pi 4 and got:
```
Hello Bme280!
Unhandled exception. System.IO.IOException: Error 121 performing I2C data transfer.
at System.Device.I2c.UnixI2cBus.WriteReadCore(UInt16 deviceAddress, Byte* writeBuffer, Byte* readBuffer, UInt16 writeBufferLength, UInt16 readBufferLength)
at System.Device.I2c.UnixI2cBus.Write(Int32 deviceAddress, ReadOnlySpan`1 buffer)
at System.Device.I2c.UnixI2cDevice.Write(ReadOnlySpan`1 buffer)
at System.Device.I2c.UnixI2cDevice.WriteByte(Byte value)
at Iot.Device.Bmxx80.Bmxx80Base..ctor(Byte deviceId, I2cDevice i2cDevice)
at SmartHome.IoT.Program.Main(String[] args) in C:\Users\adam\source\repos\SmartHome\SmartHome.IoT\Program.cs:line 33
Aborted
```

**Versions used**

**Machine being used to build**
Zestaw .NET SDK (odzwierciedlenie dowolnego pliku global.json):
 Version:   5.0.301
 Commit:    ef17233f86

Środowisko uruchomieniowe:
 OS Name:     Windows
 OS Version:  10.0.22000
 OS Platform: Windows
 RID:         win10-x64
 Base Path:   C:\Program Files\dotnet\sdk\5.0.301\

Host (useful for support):
  Version: 5.0.7
  Commit:  556582d964

.NET SDKs installed:
  5.0.101 [C:\Program Files\dotnet\sdk]
  5.0.201 [C:\Program Files\dotnet\sdk]
  5.0.202 [C:\Program Files\dotnet\sdk]
  5.0.301 [C:\Program Files\dotnet\sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.All 2.1.28 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.2.8 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.App 2.1.28 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.2.8 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.1.16 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 5.0.7 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 2.1.28 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.2.8 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.1.16 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 5.0.7 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.WindowsDesktop.App 3.1.16 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 5.0.7 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]

**Machine being used to build**
NET SDK (reflecting any global.json):
Version: 5.0.302
Commit: c005824e35

Runtime Environment:
OS Name: raspbian
OS Version: 10
OS Platform: Linux
RID: linux-arm
Base Path: /opt/dotnet/sdk/5.0.302/

Host (useful for support):
Version: 5.0.8
Commit: 35964c9215

.NET SDKs installed:
5.0.302 [/opt/dotnet/sdk]

.NET runtimes installed:
Microsoft.AspNetCore.App 5.0.8 [/opt/dotnet/shared/Microsoft.AspNetCore.App]
Microsoft.NETCore.App 5.0.8 [/opt/dotnet/shared/Microsoft.NETCore.App]

**Version of `System.Device.Gpio`**
System.Device.Gpio 1.5.0

**Version of `Iot.Device.Bindings`**
Iot.Device.Bindings 1.5.0

awrobel196, 2021-07-28:
I try use solution for get data from BMP280 using[ python script](https://www.raspberrypi-spy.co.uk/2016/07/using-bme280-i2c-temperature-pressure-sensor-in-python/) and i got correct output:

```
Chip ID : 96
Version : 0
Temperature : 27.51 C
Pressure : 969.635889079 hPa
Humidity : 56.5809839339 %
```

Ellerbach, 2021-07-28:
> I want to use Bme280 sensor with Raspberry 4. For try get data from sensor i use [example Bme280 code](https://github.com/dotnet/iot/blob/main/src/devices/Bmxx80/samples/Bme280.sample.cs) and little edit it [for use in my project](https://pastebin.com/y6wS6n4P)

Have you try both the Default and Secondary I2C address? Depending on the system, they are inverted.

pgrawehr, 2021-07-29:
Does the error happen every time or only occasionally?

krwq, 2021-07-29:
@awrobel196 can you try after creating bme280 call this:

```csharp
            bme280.SetPowerMode(Bmx280PowerMode.Normal);
            Thread.Sleep(100);
```

and see if that makes any difference? I think I recall seeing some issues unless I added a timeout but didn't have time to dig

ikivanc, 2021-09-06:
@awrobel196 I got same error using BMP280 while I was using stable build for `System.device.Gpio` and `Iot.Device.Bindings`
Read

Then used `--prereleases` and now my code is working as expected

```bash
dotnet add package System.Device.Gpio --source https://pkgs.dev.azure.com/dnceng/public/_packaging/dotnet5/nuget/v3/index.json --prerelease
dotnet add package Iot.Device.Bindings --source https://pkgs.dev.azure.com/dnceng/public/_packaging/dotnet5/nuget/v3/index.json --prerelease
```

[image]

Can you try this one?

Ellerbach, 2021-09-16:
[Triage] @awrobel196 do you still have the issue? Also are you using BME280 or BMP280? They are not the same sensors. So you have to select the right one.
On your original issue, you mentioned BME and on the test you've been doing with python, you mention BMP. And that makes a big difference because those are not the same sensors.

alvipeo, 2022-01-03:
I'm having the same problem with BME280 on Raspberry Pi 3B+. The app crashes with the following:
```
Unhandled exception. System.IO.IOException: Error 121 performing I2C data transfer.
   at System.Device.I2c.UnixI2cBus.WriteReadCore(UInt16 deviceAddress, Byte* writeBuffer, Byte* readBuffer, UInt16 writeBufferLength, UInt16 readBufferLength)
   at System.Device.I2c.UnixI2cBus.Write(Int32 deviceAddress, ReadOnlySpan`1 buffer)
   at System.Device.I2c.UnixI2cDevice.Write(ReadOnlySpan`1 buffer)
   at System.Device.I2c.UnixI2cDevice.WriteByte(Byte value)
   at Iot.Device.Bmxx80.Bmxx80Base..ctor(Byte deviceId, I2cDevice i2cDevice)
   at Iot.Device.Bmxx80.Bmx280Base..ctor(Byte deviceId, I2cDevice i2cDevice)
   at Iot.Device.Bmxx80.Bme280..ctor(I2cDevice i2cDevice)
   at RaspPiApp.TemperatureHumidityService.ReadSensorAndUploadValuesToAzure(Double calibrateDegreeC, CancellationToken cancellationToken) in <app_path>\TemperatureHumidityService.cs:line 48
   at RaspPiApp.MonitorLoop.StartMonitorLoop() in <app_path>\MonitorLoop.cs:line 109
   at RaspPiApp.Program.Main(String[] args) in <app_path>\Program.cs:line 38
   at RaspPiApp.Program.<Main>(String[] args)
```

It used to work with prev versions of packages. I tried nightly but the `Iot.Device.Bindings` can't be added to the project because:
>error: NU1101: Unable to find package System.Drawing.Common. No packages exist with this id in source(s): https://pkgs.dev.azure.com/dotnet/IoT/_packaging/nightly_iot_builds/nuget/v3/index.json
error: Package 'Iot.Device.Bindings' is incompatible with 'all' frameworks in project 'RaspPiApp.csproj'.

I really need this to work with the latest version (or prerelease).

alvipeo, 2022-01-03:
Tried `netcoreapp3.1` version - same thing. Something is definitely wrong with it.

pgrawehr, 2022-01-04:
If it crashes in the constructor of Bme280, the typical cause is a bad connection or a wrong address. What is the output of `i2cdetect -y 1`?

alvipeo, 2022-01-04:
Re-connected BME280. `i2cdetect -y 1` shows:

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- 77

Now I got this:
>Unhandled exception. System.NullReferenceException: Object reference not set to an instance of an object.
   at System.Device.I2c.UnixI2cDevice.Write(ReadOnlySpan`1 buffer)
   at System.Device.I2c.UnixI2cDevice.WriteByte(Byte value)
   at Iot.Device.Bmxx80.Bmxx80Base..ctor(Byte deviceId, I2cDevice i2cDevice)
   at Iot.Device.Bmxx80.Bmx280Base..ctor(Byte deviceId, I2cDevice i2cDevice)
   at Iot.Device.Bmxx80.Bme280..ctor(I2cDevice i2cDevice)
   at RaspPiApp.TemperatureHumidityService.ReadSensorAndUploadValuesToAzure(Double calibrateDegreeC, CancellationToken cancellationToken) in <app_path>\TemperatureHumidityService.cs:line 54
   at RaspPiApp.MonitorLoop.StartMonitorLoop() in <app_path>\MonitorLoop.cs:line 111
   at RaspPiApp.Program.Main(String[] args) in <app_path>\Program.cs:line 40
   at RaspPiApp.Program.<Main>(String[] args)
Aborted

Here's the code:
```
            if (_i2CDevice == null)
            {
                _logger.LogWarning("i2cDevice is null. Creating again...");
                CreateI2cDevice();
                if (_i2CDevice == null) return null;
            }

            using Bme280 sensorBme280 = new(_i2CDevice)
                {TemperatureSampling = Sampling.UltraHighResolution, PressureSampling = Sampling.UltraHighResolution, HumiditySampling = Sampling.Standard};
            {
                sensorBme280.SetPowerMode(Bmx280PowerMode.Normal); // Forced - set mode forced so device sleeps after read

                // wait for measurement to be performed
                int measurementTime = sensorBme280.GetMeasurementDuration();

                Thread.Sleep(measurementTime + 30);
            }

            SensorValues? values = ReadSensorValues(sensorBme280);
...

    private void CreateI2CDevice()
    {
        _i2CDevice = I2cDevice.Create(_i2CConnectionSettings);
        if (_i2CDevice == null)
            _logger.LogError("Could not create Bme sensor device");
        else
            _logger.LogInformation("Bme sensor device is created successfully.");
    }

    private I2cDevice _i2CDevice;
    private readonly I2cConnectionSettings _i2CConnectionSettings = new(BUS_ID, 0x77); // BUS_ID = 1

```

.csproj:
```
    <PackageReference Include="Azure.Storage.Blobs" Version="12.10.0" />
    <PackageReference Include="Iot.Device.Bindings" Version="1.5.0" />
    <PackageReference Include="Microsoft.ApplicationInsights.WorkerService" Version="2.20.0" />
    <PackageReference Include="Microsoft.Azure.Cosmos.Table" Version="1.0.8" />
    <PackageReference Include="Microsoft.Azure.ServiceBus" Version="5.2.0" />
    <PackageReference Include="Microsoft.Extensions.Hosting" Version="6.0.0" />
    <PackageReference Include="Microsoft.Extensions.Logging" Version="6.0.0" />
    <PackageReference Include="System.Text.Json" Version="6.0.1" />
```

pgrawehr, 2022-01-05:
I can't see what could be null there, but I will need to give it a test. If possible, you could try the prerelease feed from https://pkgs.dev.azure.com/dotnet/IoT/_packaging/nightly_iot_builds/nuget/v3/index.json

alvipeo, 2022-01-05:
Not possible:
>error: NU1101: Unable to find package System.Drawing.Common. No packages exist with this id in source(s): https://pkgs.dev.azure.com/dotnet/IoT/_packaging/nightly_iot_builds/nuget/v3/index.json
error: Package 'Iot.Device.Bindings' is incompatible with 'all' frameworks in project

pgrawehr, 2022-01-05:
That's a weird error message, because obviously System.Drawing.Common is not in that source and it should use the default for that package. Which target framework are you using?

alvipeo, 2022-01-05:
ok, how should I set the target in VS2022? I do make an executable with the following script:
```
dotnet publish -r linux-arm --self-contained True -c Release -p:PublishSingleFile=true -p:GenerateRuntimeConfigurationFiles=true -p:PublishTrimmed=True -p:TrimMode=Link
```
but this has nothing to do with the target. How can I do this?

alvipeo, 2022-01-05:
my csproj:
```
<Project Sdk="Microsoft.NET.Sdk">

  <PropertyGroup>
    <OutputType>Exe</OutputType>
    <TargetFramework>net6.0</TargetFramework>
    <Platforms>AnyCPU;ARM64</Platforms>
  </PropertyGroup>

  <PropertyGroup Condition="'$(Configuration)|$(Platform)'=='Debug|AnyCPU'">
    <DefineConstants>$(DefineConstants)TRACE;SENSOR</DefineConstants>
  </PropertyGroup>

  <PropertyGroup Condition="'$(Configuration)|$(Platform)'=='Debug|ARM64'">
    <DefineConstants>$(DefineConstants)TRACE;SENSOR</DefineConstants>
  </PropertyGroup>

  <PropertyGroup Condition="'$(Configuration)|$(Platform)'=='Release|AnyCPU'">
    <DefineConstants>$(DefineConstants)TRACE;SENSOR</DefineConstants>
  </PropertyGroup>

  <PropertyGroup Condition="'$(Configuration)|$(Platform)'=='Release|ARM64'">
    <DefineConstants>$(DefineConstants)TRACE;SENSOR</DefineConstants>
  </PropertyGroup>

  <ItemGroup>
    <None Remove="appsettings.json" />
  </ItemGroup>

  <ItemGroup>
    <Content Include="appsettings.json">
      <CopyToOutputDirectory>PreserveNewest</CopyToOutputDirectory>
    </Content>
  </ItemGroup>

  <ItemGroup>
    <PackageReference Include="Azure.Storage.Blobs" Version="12.10.0" />
    <PackageReference Include="Iot.Device.Bindings" Version="1.5.0" />
    <PackageReference Include="Microsoft.ApplicationInsights.WorkerService" Version="2.20.0" />
    <PackageReference Include="Microsoft.Azure.Cosmos.Table" Version="1.0.8" />
    <PackageReference Include="Microsoft.Azure.ServiceBus" Version="5.2.0" />
    <PackageReference Include="Microsoft.Extensions.Hosting" Version="6.0.0" />
    <PackageReference Include="Microsoft.Extensions.Logging" Version="6.0.0" />
    <PackageReference Include="System.Text.Json" Version="6.0.1" />
  </ItemGroup>

</Project>
```

pgrawehr, 2022-01-05:
That csproj looks good. In VS2022 you can edit the nuget sources by clicking the cogweel in the top right hand corner of the nuget package manager window. Make sure it lists nuget.org as well as the above source in the package source list.

alvipeo, 2022-01-05:
I do have all the nuget sources. As you can see it finds the package but for some reason it errors out with this:
> error: Package 'Iot.Device.Bindings' is incompatible with 'all' frameworks in project

And I don't know how to fix this.

alvipeo, 2022-01-06:
Just in case, I added this nightly feed to Nuget sources - didn't help a bit.

pgrawehr, 2022-01-06:
I'm afraid then you have to wait for our dependency expert @joperezr, but he's still enjoying vacations.

joperezr, 2022-01-06:
@alvipeo sorry for missing this. Seems like in your case you have some misconfigured package sources causing it to not use nuget.org as a source as well. To solve this, create a nuget.config (either by manually creating the file, or by running `dotnet new nugetconfig`) on the same folder that you have your .csproj file, and set the contents as:

```xml
<?xml version="1.0" encoding="utf-8"?>
<configuration>
  <packageSources>
    <!--To inherit the global NuGet package sources remove the <clear/> line below -->
    <clear />
    <add key="nuget" value="https://api.nuget.org/v3/index.json" />
    <add key="nightly_iot_builds@Local" value="https://pkgs.dev.azure.com/dotnet/IoT/_packaging/nightly_iot_builds%40Local/nuget/v3/index.json" />
  </packageSources>
</configuration>
```

After saving that file, then running `dotnet restore` on that folder should work as expected and the packages shoulld be restored.

alvipeo, 2022-01-07:
@joperezr added file but it didn't help. still getting:
>  Determining projects to restore...
  Writing ...\AppData\Local\Temp\tmpEF48.tmp
info : Adding PackageReference for package 'Iot.Device.Bindings' into project '<app_path>\RaspPiApp.csproj'.
info :   CACHE https://pkgs.dev.azure.com/dotnet/9a6dded6-c408-4b7c-8faa-7c38e0b4051f/_packaging/7ab79c7f-daa9-4e9c-88c8-61478097d0ff/nuget/v3/registrations2-semver2/iot.device.bindings/index.json
info : Restoring packages for <app_path>\RaspPiApp.csproj...
info :   GET https://pkgs.dev.azure.com/dotnet/9a6dded6-c408-4b7c-8faa-7c38e0b4051f/_packaging/7ab79c7f-daa9-4e9c-88c8-61478097d0ff/nuget/v3/flat2/system.drawing.common/index.json
info :   NotFound https://pkgs.dev.azure.com/dotnet/9a6dded6-c408-4b7c-8faa-7c38e0b4051f/_packaging/7ab79c7f-daa9-4e9c-88c8-61478097d0ff/nuget/v3/flat2/system.drawing.common/index.json 1192ms
error: NU1101: Unable to find package System.Drawing.Common. No packages exist with this id in source(s): https://pkgs.dev.azure.com/dotnet/IoT/_packaging/nightly_iot_builds/nuget/v3/index.json
error: Package 'Iot.Device.Bindings' is incompatible with 'all' frameworks in project '<app_path>\RaspPiApp.csproj'.

Something's utterly wrong with VS2022... I already reinstalled it - it didn't help. What if I share my project with you somehow?

alvipeo, 2022-01-07:
I tried on another machine - same problem.

alvipeo, 2022-02-10:
alright, I updated VS to 17.0.6 and was able to install required packages. The problem is still there, UNLESS I stop creating/destroying Bme280 device. So I keep the Bme280 object from start to finish and this time it doesn't throw, obviously. Because I don't run the Bme280 constructor anymore.

pgrawehr, 2022-02-11:
That's (for performance reasons) a good idea anyway. The Bme devices are known to cause this behavior, I've seen this several times as well. The only known workaround I have is to retry in case of a failure.

closed 2022-09-30 by msftbot[bot]