dotnet/iot #1623: Consider re-using location interfaces from full framework

krwq, 2021-08-06:
In full framework System.Device.dll contains some classes used for geo location. We might consider bringing those into .NET IoT

https://docs.microsoft.com/en-us/dotnet/api/system.device.location?view=netframework-4.8

https://github.com/dotnet/runtime/issues/23250
https://github.com/dotnet/runtime/issues/30433
https://github.com/dotnet/runtime/issues/31202
https://github.com/dotnet/runtime/issues/56751

cc: @richlander

Redth, 2021-08-06:
In Xamarin.Essentials (and now .NET MAUI) we also have a type for geo location: https://github.com/dotnet/maui/blob/main/src/Essentials/src/Types/Location.shared.cs#L22

This looks very similar to the full framework's location type, but we have a couple of additional data points:

- Timestamp (DateTimeOffset)
- AltitudeReferenceSystem (Unspecified, Terrain, Ellipsoid, Geoid, Surface)

Timestamp is useful for hopefully obvious reasons, though one could make an argument to build another type to contain this result alongside the Location instance instead of coupling them together.

Altitude Reference system is a bit more important as different platforms have native API's which use different reference systems.  For instance, iOS/tvOS/ipadOS/macOS use Geoid, Android uses Ellipsoid, and UWP can return any of the types I just listed.  

I can't recall if there's a meaningful and not too lossy conversion between these reference systems, but perhaps if there is, it's possible to just pick one and normalize it and make it the implicit unit for the type and call it a day.

It would be interesting to see this added to .NET 7 as a type to use everywhere and not just bring it into IoT :)

pgrawehr, 2021-08-07:
@Redth There are conversion tables you can download that make the conversion from elipsoid to WGS geoid. For the current location, the GPS receiver typically delivers the delta. At least, it's part of the NMEA specification, not sure about Android/IOS APIs (I have only ever worked with GPS receivers for professional equipment, not in embedded contexts). 

@krwq I absolutely agree we should have these types. I'm actually already using them :-)

pgrawehr, 2021-08-10:
The TimeStamp doesn't actually belong to a generic location primitive. It is correct within the result of a GNSS receiver, of course, but not as part of the generic primitive. Since a position can also indicate a waypoint or the position of an entity (i.e a house). You typically don't use timestamps with those.