dotnet/iot #1626: VirtualOutputSegment Write offset

Jimmys20, 2021-08-07:
Hello, in `Iot.Device.Multiplexing` the class`VirtualOutputSegment` has a method `void Write(byte value)` which states in the xml doc that the **_Least significant bit will written to index 0._**

For example:
```
VirtualOutputSegment segment = new(8);
segment.Write(0b_1001_0110);

// Result:
// segment index: | 0 | 1 | 2 | 3 | 4 | 5 | 6 | 7
//         value: | 0 | 1 | 1 | 0 | 1 | 0 | 0 | 1
```

What I find odd is that when using a segment with bigger length that the written value the result is this:
```
VirtualOutputSegment segment = new(12);
segment.Write(0b_1001_0110);

// Result (offset to the left side of the byte value)
// segment index: | 0 | 1 | 2 | 3 | 4 | 5 | 6 | 7 | 8 | 9 | 10 | 11
//         value: | 0 | 0 | 0 | 0 | 0 | 1 | 1 | 0 | 1 | 0 |  0 |  1

// Expected result (offset to the right side of the byte value)
// segment index: | 0 | 1 | 2 | 3 | 4 | 5 | 6 | 7 | 8 | 9 | 10 | 11
//         value: | 0 | 1 | 1 | 0 | 1 | 0 | 0 | 1 | 0 | 0 |  0 |  0
```

To my understanding the least significant bit should always end up to index 0 and any possible offset should end up to the right. To explain with unit tests, I believe that the following two tests should be both be passing.

```
[Fact]
public void SegmentValuesWriteByte()
{
    VirtualOutputSegment segment = new(8);
    segment.Write(0b_1001_0110);

    Assert.True(
        segment[0] == 0 &&
        segment[1] == 1 &&
        segment[2] == 1 &&
        segment[3] == 0 &&
        segment[4] == 1 &&
        segment[5] == 0 &&
        segment[6] == 0 &&
        segment[7] == 1);
}

[Fact]
public void SegmentValuesWriteByteOffset()
{
    // segment length bigger than the written value
    VirtualOutputSegment segment = new(12);
    segment.Write(0b_1001_0110);

    Assert.True(
        segment[0] == 0 &&
        segment[1] == 1 &&
        segment[2] == 1 &&
        segment[3] == 0 &&
        segment[4] == 1 &&
        segment[5] == 0 &&
        segment[6] == 0 &&
        segment[7] == 1);
}
```

So is this possibly a bug or I didn't understand correct the way it should work?

pgrawehr, 2021-08-07:
@richlander Probably your domain

TheZlodziej, 2021-09-16:
It looks like the "offset" variable in Iot.Device.Multiplexing.Utility.VirtualOutputSegment.WriteByteAsValues(byte value, int offset) is unnecessary.

```
// calculate index by subtracting i from 7.
int index = 7 - i;
_values[index] = data;
```

joperezr, 2021-10-08:
This is a great question. I believe your assumption above is right, and we do want those two unit tests you wrote to be passing. Sorry that I closed the PR #1662 which now I understand was trying to fix this other issue. Would you mind either re-opening that one, or a new one, and add the fix for this plus both of the unit tests above?

PR #1684 (2021-10-20): Offset in VirtualOutputSegment.Write(byte value) shouldn't be calculated (is always 0).
Fixes #1626

The #1627 didn't fix the #1626 issue. Since the PR was merged and it had conflict with mine, I'm redoing my changes.

The offset in VirtualOutputSegment.Write(byte value) is calculated with following formula:
```c#
int offset = _length - 8;
```
when in reality it should always be 0.

I tested my code with the same tests as last time:
```c#
[Fact]
public void VirtualOutputSegmentWriteByteOffset()
{
    // segment length > written value
    MyOptSeg segment = new(12);
    segment.Write(0b_1001_01 [... cut]
closed 2021-10-20 by joperezr