dotnet/iot #1632: Is there an example of a remote control controlling its own program? Or net core Bluetooth.

bbhxwl, 2021-08-11:
Is there an example of a remote control controlling its own program? Or net core Bluetooth.

I want the remote control to interact with my program.

Ellerbach, 2021-08-11:
Hi @bbhxwl 
you can have a WebServer for example with a set of API.
Check https://github.com/nanoframework/Samples/tree/main/samples/Webserver

raffaeler, 2021-08-12:
@bbhxwl Could you please explain the scenario?
If you want to control devices on the Raspberry PI (or similar board), you can just write an ASP.NET application accepting REST calls and controlling the connected devices over WiFi or ethernet.
With regards to Bluetooth, I am not aware of any library (Xamarin apart) that can let you use it on .NET Core on Linux.

bbhxwl, 2021-08-12:
> @bbhxwl Could you please explain the scenario?
> If you want to control devices on the Raspberry PI (or similar board), you can just write an ASP.NET application accepting REST calls and controlling [...]

I made a software to play video, and then I want Xiaomi to press up and down the Bluetooth remote control to control the up and down of my software video.

raffaeler, 2021-08-12:
> I made a software to play video, and then I want Xiaomi to press up and down the Bluetooth remote control to control the up and down of my software video.

Hi @bbhxwl,
I agree that supporting Bluetooth would be a great feature to add. It probably would deserve a seprarate library as the Bluetooth specifications are very large and may involve a lot of work.

As I said I am currently not aware of any library enabling support for BT on Linux with .NET Core. Anyway, you may rely on a simple serial communication over Bluetooth by using a small and cheap JDY card like [this one](https://www.aliexpress.com/item/32911430913.html) or [this other one](https://www.aliexpress.com/item/32869488063.html) that is more powerful.

Please note that BT already provide an official service specification for remote controls: https://www.bluetooth.com/specifications/specs/a-v-remote-control-profile-1-6-2/
In theory, you should implement this spec on the server side, and rely on Android APIs to implement the client side.

raffaeler, 2021-08-12:
@bbhxwl I found some time to investigate, and it looks like the most recent Linux kernels (>= 2.6) already provide built-in Bluetooth support.
The official library is http://www.bluez.org/
The API documentation (C Language) is here: https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc

Said that, I found a library that may be interesting for you:
- Repo: https://github.com/phylomeno/dotnet-ble-server
- Nuget: https://www.nuget.org/packages/DotnetBleServer/0.1.0-snapshot
But I never used it, so I have no clue if and how it works.

Ellerbach, 2021-08-13:
> Hi @bbhxwl
> you can have a WebServer for example with a set of API.
> Check https://github.com/nanoframework/Samples/tree/main/samples/Webserver

Quote replying myself as I think I get wrongly the question! Sorry for that. I understood you wanted to control a small MCU, not your Raspberry Pi! Still for the Raspberry Pi, the API is a good answer, you can use ASP.NET on it! 
Otherwise, the previous answers are all correct as well.

raffaeler, 2021-08-22:
@bbhxwl given there is currently no support for BT, this issue is a placeholder for a possible enhancement if the existing libraries are not sufficient.
Did you have the time to take a look at them?
And beyond that, is there any other outstanding issue in your request?

Ellerbach, 2021-09-16:
[Triage] @bbhxwl Did you have time to look at all this? Is this still a valid issue?

bbhxwl, 2021-09-17:
> [Triage] @bbhxwl Did you have time to look at all this? Is this still a valid issue?

Net should not have a good solution in this regard. There are no other problems now. thank you.

closed 2021-09-17 by bbhxwl