dotnet/iot #1637: After unregister all callbacks, the new registered callbacks events are not fired in Raspberry Pi 4

nmunozgarcia, 2021-08-17:
**Describe the bug**

In GpioController, when you register some callbacks with RegisterCallbackForPinValueChangedEvent() function, the events are fired as expected.
Then, If you unregister all the callbacks registered before with UnregisterCallbackForPinValueChangedEvent() function, your events are not fired now. (This is the expected behaviour).
Now, if you registered new callbacks in GpioController with RegisterCallbackForPinValueChangedEvent() function, the new events are not fired. No matter how many times you unregister and register callbacks, the events will never be fired again.

**Steps to reproduce**

- Create new GpioController
- Open one pin as input / InputPullUp / InputPullDown
- Register some callbacks with RegisterCallbackForPinValueChangedEvent() function.
- Unregister all of them with UnregisterCallbackForPinValueChangedEvent() function.
- Register again some callbacks.

```csharp
class Program
{
    static void Main(string[] args)
    {
        int pin = 12;
        PinMode pinMode = PinMode.InputPullUp;
        using GpioController controller = new();

        for(int i = 0; i < 3; i++)
        {
            Console.WriteLine($"Iteration: {i+1}");
            OpenAndRegisterPin(controller, pin, pinMode);
            Console.WriteLine($"Try pin:{pin} events in 30 seconds.");
            Task.Delay(30000).Wait();
            CloseAndUnregisterPin(controller, pin);
        }
    }

    protected static void OpenAndRegisterPin(GpioController controller, int pin, PinMode pinMode)
    {
        Console.WriteLine($"Opening Pin: {pin}");
        controller.OpenPin(pin, pinMode);

        Console.WriteLine($"Registering pin events.");
        controller.RegisterCallbackForPinValueChangedEvent(pin, PinEventTypes.Falling, OnFalling);
        controller.RegisterCallbackForPinValueChangedEvent(pin, PinEventTypes.Rising, OnRising);
    }

    protected static void CloseAndUnregisterPin(GpioController controller, int pin)
    {
        Console.WriteLine($"Unregistering pin events.");
        controller.UnregisterCallbackForPinValueChangedEvent(pin, OnFalling);
        controller.UnregisterCallbackForPinValueChangedEvent(pin, OnRising);

        Console.WriteLine($"Closing Pin: {pin}");
        controller.ClosePin(pin);
    }

    protected static void OnFalling(object sender, PinValueChangedEventArgs ev) 
    {
        Console.WriteLine($"OnFalling Pin:{ev.PinNumber}");
    }
    protected static void OnRising(object sender, PinValueChangedEventArgs ev) 
    {
        Console.WriteLine($"OnRising Pin:{ev.PinNumber}");
    }
}
```

**Expected behavior**

In all iterations, the console must show OnFallling and/or OnRising messages with the Pin:12.
For example:
```
Iteration: 1
Opening Pin: 12
Registering pin events.
Try pin:12 events in 30 seconds.
OnFalling Pin:12
OnRising Pin:12
Unregistering pin events.
Closing Pin: 12
Iteration: 2
Opening Pin: 12
Registering pin events.
Try pin:12 events in 30 seconds.
OnFalling Pin:12
OnRising Pin:12
Unregistering pin events.
Closing Pin: 12
Iteration: 3
Opening Pin: 12
Registering pin events.
Try pin:12 events in 30 seconds.
OnFalling Pin:12
OnRising Pin:12
Unregistering pin events.
Closing Pin: 12
```

**Actual behavior**

Only in the iteration 1 (i == 0) the event messages are showed.
In the next iterations, this doesn't happened.
For example:
```
Iteration: 1
Opening Pin: 12
Registering pin events.
Try pin:12 events in 30 seconds.
OnFalling Pin:12
OnRising Pin:12
Unregistering pin events.
Closing Pin: 12
Iteration: 2
Opening Pin: 12
Registering pin events.
Try pin:12 events in 30 seconds.
Unregistering pin events.
Closing Pin: 12
Iteration: 3
Opening Pin: 12
Registering pin events.
Try pin:12 events in 30 seconds.
Unregistering pin events.
Closing Pin: 12
```

**Versions used**

Checked In Raspberry Pi 4 Model B with Raspberry Pi OS

- `dotnet --info` on the machine being used to build
```
SDK de .NET (que refleje cualquier global.json):
 Version:   5.0.400
 Commit:    d61950f9bf

```

- `dotnet --info` on the machine where app is being run (not applicable for self-contained apps)
```
SDK de .NET (que refleje cualquier global.json):
 Version:   5.0.400
 Commit:    d61950f9bf
```
Also checked with:
```
SDK de .NET Core (reflejando cualquier global.json):
 Version:   3.1.411
 Commit:    d100cdf718
``` 

- Version of `System.Device.Gpio` package
```
1.5.0
Also checked with:
1.3.0
```

raffaeler, 2021-08-17:
@nmunozgarcia Could you please give a bit of context about the operating system you are using?
Is it Raspbian? Which version?

nmunozgarcia, 2021-08-17:
This Bug is caused because of private readonly _eventThreadCancellationTokenSource variable in System.Device.Gpio.Drivers.SysFsDriver class.
```
private readonly CancellationTokenSource _eventThreadCancellationTokenSource;
```
When all callbacks are removed, the CancellationTokenSource _eventThreadCancellationTokenSource is cancelled with the Cancel() function in RemovePinFromPoll() function.
After cancelled, the CancellationTokenSource is never recreated (because is readonly) and, with the token cancelled forever, the events are never fired.
One solution could be to change _eventThreadCancellationTokenSource (removing the readonly property).
```
private CancellationTokenSource _eventThreadCancellationTokenSource;
```
And create a new CancellationTokenSource before execute _eventDetectionThread.Start(); in the InitializeEventDetectionThread() function.
```
private void InitializeEventDetectionThread()
        {
            if (_eventDetectionThread == null)
            {
                _eventDetectionThread = new Thread(DetectEvents)
                {
                    IsBackground = true
                };
                _eventThreadCancellationTokenSource = new CancellationTokenSource();
                _eventDetectionThread.Start();
            }
        }
```

raffaeler, 2021-08-17:
It sounds reasonable for me.
Are you willing to make a PR, eventually adding a specific test?

nmunozgarcia, 2021-08-17:
Hi @raffaeler.
It's a Raspbian OS (now I think it's called Raspberry Pi OS) in the latest version:
```
pi@raspberrypi:~/ConsoleTestDotNet5/publish $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

nmunozgarcia, 2021-08-17:
I've never made a pull request, but I'm going to try it now. (Let's see what happens...)
I don't know how to make a Test with events fired externally out of my code.

raffaeler, 2021-08-17:
Great!
If you look at the other tests, you will see there are mock classes to simulate those kind of events.
The very basic steps to make a PR are:
- fork the repo
- make the changes + test
- make sure to accept the required license as a collaborator
- submit a PR by referencing this thread. At this point all the tests will be run automatically. There may be fails not depending on you. Someone from the team will take care of that, if needed.
Finally, the triage team will discuss with you any eventual issue or doubt.

Good work and thank you :)

pgrawehr, 2021-08-17:
@nmunozgarcia While this is really a bug, we encourage the use of libgpiod instead of sysfs for event registration. It is much faster and more reliable. All you have to do is run

```
sudo apt install -y libgpiod-dev
```

I'm pretty sure this will make the error go away.

nmunozgarcia, 2021-08-19:
Hi @raffaeler!
Thank you so much!
Sorry, I don't have so much time this days but I tried to make the contributing enviroment in Windows 10 (following this guide https://github.com/dotnet/iot/blob/main/Documentation/CONTRIBUTING.md) without success, because I don't have the Linux enviroment to check the Linux tests (all SysFsDriverTests fail with the error message: "System.PlatformNotSupportedException : SysFsDriver is only supported on Linux/Unix") and the Solution Configuration doesn't have the options showed in the contributing documentation.
[https://github.com/dotnet/iot/blob/main/Documentation/images/configurations.png](https://github.com/dotnet/iot/blob/main/Documentation/images/configurations.png)

@pgrawehr Thank you. I will try libgpiod-dev after uploading the changes.

raffaeler, 2021-08-19:
@nmunozgarcia You are welcome :)
Be aware that, if you have just a Raspberry PI available, you can use Visual Studio Code with the extensions called "Visual Studio Code Remote - SSH" and "Remote Development" to write code on your preferred OS and run on the RPi.

If you feel "lucky" :) you can also create the PR and rely on the tests that are automatically run on GitHub. I am suggesting also this opportunity as the code change looks straightforward.
If the tests fail, you just modify the code and commit and the CI will run the tests again.

In any case, it is also worth give a try to the suggestion by @pgrawehr. I agree it should work.

HTH

joperezr, 2021-08-19:
[Triage]: @nmunozgarcia thanks for finding the issue, would you be interested in contributing a PR to fix the SysFsDriver? What we need to do is to make sure that the CTS life-span equates the one for the eventDetection thread, so have it be created every time that the thread is created, and disposed every time that the thread is disposed.

nmunozgarcia, 2021-09-02:
Hi @raffaeler!
I tried to execute the GpioControllerTestBase.cs tests, but in Windows X86 I got the "System.PlatformNotSupportedException : SysFsDriver is only supported on Linux/Unix" message, obviously as I said in my previous message.

In Raspberry, I tried to connect remotely from Windows to Raspberry (with Visual Studio Code Remote - SSH as you suggested to me) and also installing Visual Studio Code in the propper Raspberry. In both cases, I got an error message: "The C# extension for Visual Studio Code (powered by OmniSharp) is incompatible on linux armv7l".

In the OmniSharp-vscode proect (https://github.com/OmniSharp/omnisharp-vscode) I take the armv7l_aarch64 branch and compile it to get the C# extension working in Raspberry. But .NET Core Test Explorer plugin in Visual Studio Code didn't recognize the tests.

My last try was installing a virtual machine with Ubuntu 20.04 (X86_64) to get the Linux architecture and the c# pluging working. However, Visual Studio Code (again) didn't recognize the tests (and I don't know why!).

Right now, I only achieve to get the tests recognized in my Windows machine in Visual Studio 2022 Preview and in Visual Studio Code (with C# and .NET Core Test Explorer plugins working).

What do you use to test the Linux architecture tests?

Hi joperezr!
Yes, I want to contribute to the fix by making the Pull Request. But it's taken a lot of time to get the tests working (to create my test and check the test fails without the patch and passes with the patch).

raffaeler, 2021-09-02:
Hi @nmunozgarcia ,
installing vscode and the extensions should be straightforward. You don't need to download any package from Github, just follow the instructions:
- vscode on RPi: https://code.visualstudio.com/docs/setup/raspberry-pi
- remote SSH: https://www.raspberrypi.org/blog/coding-on-raspberry-pi-remotely-with-visual-studio-code/

> "The C# extension for Visual Studio Code (powered by OmniSharp) is incompatible on linux armv7l"
this is probably happening because you manually installed the extension instead of letting vscode install the right one from the vscode marketplace.

You should also be able to run the tests via `dotnet` cli if you install the dotnet sdk on the RPi. But I would rather prefer doing everything from vscode.

Let us know if you are successful!

nmunozgarcia, 2021-09-02:
> "The C# extension for Visual Studio Code (powered by OmniSharp) is incompatible on linux armv7l"
this is probably happening because you manually installed the extension instead of letting vscode install the right one from the vscode marketplace.

No. I got the incompatible message with the extension from the vscode marketplace. That's why I searched for a solution and found in the propper project a branch to work with ARM and ARM64 (and recompile it, with that branch, and tried it without success).

raffaeler, 2021-09-02:
My fault, @nmunozgarcia I am reading that the extension can't run on ARM. Someone did it by recompiling the code.

Anyway you should be able to run the tests without the extension. See this doc:
https://github.com/dotnet/iot/blob/main/src/System.Device.Gpio.Tests/README.md

nmunozgarcia, 2021-09-03:
Don't worry, @raffaeler.
My apologies, I didn't see that Readme before.

Thank you so much! I finally can run the tests.

Now I'm trying to create the test and check it out.

Should I put it in SysFsDriverTests.cs file for only the SysFsDriver driver or in GpioControllerTestBase.cs file for all the drivers?

raffaeler, 2021-09-03:
Well done, great work.
I am personally ok with that. Just have a bit of patience so that others may find some time to review the PR.

PR #1653 (2021-09-10): Fix for SysFsDriver when callbacks registered events are not fired, after unregister all registered callbacks.
Fixes #1637

When you registered callbacks in GpioController (when using SysFsDriver), their events are fired as expected.
Then, if you unregistered **all** callbacks, the events are not fired (this is the expected behavior).
Now, If you registered callbacks again (the same or others), their events are not fired. 
This is the bug.

The problem is in SysFsDriver.cs file. The _eventThreadCancellationTokenSource variable is created the first time when SysFsDriver class is also created.
T [... cut]
closed 2021-09-10 by msftbot[bot]