dotnet/iot #1640: RPI4 PWM can't be used with Windows11

ChesterHuang999, 2021-08-19:
**Describe the bug**
When I execute the Sample code, I get an error message like the following. 

Unhandled exception. System.ArgumentException: No PWM device exists for PWM chip at index 0. (Parameter 'chip')
   at System.Device.Pwm.Channels.Windows10PwmChannel..ctor(Int32 chip, Int32 channel, Int32 frequency, Double dutyCycle)
   at System.Device.Pwm.PwmChannel.CreateWindows10PwmChannel(Int32 chip, Int32 channel, Int32 frequency, Double dutyCyclePercentage)
   at System.Device.Pwm.PwmChannel.Create(Int32 chip, Int32 channel, Int32 frequency, Double dutyCyclePercentage)
   at GPIOTry.Program.Main(String[] args) in D:\Chester\Project\RPI4\Windows\GPIOTry\GPIOTry\Program.cs:line 16

I changed the boot/config according to "https://github.com/dotnet/iot/blob/main/Documentation/raspi-pwm.md", 
But the same problem still occurs.
```csharp
// put sample code here
using System;
using System.Device.Gpio;
using System.Device.I2c;
using System.Device.Spi;
using System.Device.Pwm;
using System.Threading;
namespace GPIOTry
{
    class Program
    {
        static void Main(string[] args)
        {
            Console.WriteLine("Hello World!");
            var gpiocontroller = new GpioController();
            
            PwmChannel pwmcontroller = PwmChannel.Create(0,0,400,0);
            pwmcontroller.Start();
            gpiocontroller.OpenPin(2, PinMode.Output);
            byte bringth = 0;
            try
            {
                while (true)
                {
                    bringth++;
                    if (bringth == 11)
                        bringth = 0;
                    pwmcontroller.DutyCycle=bringth*0.1;
                    gpiocontroller.Write(2, PinValue.High);
                    Console.WriteLine("Duty is " + bringth.ToString());
                    Console.WriteLine("Set GPIO2 High");
                    Thread.Sleep(1000);
                    gpiocontroller.Write(2, PinValue.Low);
                    Console.WriteLine("Set GPIO2 Low");
                    Thread.Sleep(1000);
                }
            }
            finally
            {
                pwmcontroller.Stop();
                gpiocontroller.ClosePin(2);
            }
        }
    }
}

Platform: RPI4B
OS: Windows11
System.Device.Gpio: 1.5.0

krwq, 2021-08-19:
@ChesterHuang999 Windows 11 is not supported at the moment. I'm not sure how could you have change /boot/config.txt which is linux file on Windows 11... I believe the only scenarios which will work on Window at the moment are:
- Windows IoT Core - @joperezr do you perhaps remember if you have to do anything special for PWM?
- using GPIO/SPI/I2C with devices such as FT4222
- using GPIO/SPI/I2C Firmata

then once you get I2C you can get [Pca9685](https://github.com/dotnet/iot/tree/main/src/devices/Pca9685) for PWM.

I don't think anyone of us have tried to make it work with Win11 but supposedly we could make Linux `/dev/mem` implementation work on Windows with some tiny modifications... cc: @joperezr  @Ellerbach @pgrawehr @raffaeler in case any of you have tried Win 11 on PI already

raffaeler, 2021-08-19:
Sorry, I did not try Win11 on the PI.

FWIW in my projects I always use a small external PWM card as I found to be more reliable (because of the timings) than the internal RPi PWM.

pgrawehr, 2021-08-19:
@ChesterHuang999 Please describe exactly on what kind of hardware and operating system you were trying to run the code. There's no PWM hardware on typical Windows desktop PCs, and there's no Windows 11 for Raspberry Pi.

joperezr, 2021-08-19:
[Triage]: If you could share steps on how to install windows 11 on Pi4 then we can try to get a repro environment and investigate.

ChesterHuang999, 2021-08-20:
Thank you all for your replies.

The hardware platform is RPI4 B which ram is 4GB.
Hardware details: https://www.raspberrypi.org/products/raspberry-pi-4-model-b/

First, in addition to PWM, I experimented using "System.Device.GPIO" to control GPIO. The GPIO part can work normally.

For how to install Windows 11 on RPI4, please refer to:
https://www.tomshardware.com/how-to/install-windows-11-raspberry-pi

@krwq   "I'm not sure how could you have changed /boot/config.txt which is linux file on Windows 11"
When you have an SD Card made according to "https://www.tomshardware.com/how-to/install-windows-11-raspberry-pi". Use the "WoR-Boot-Mounter" program, you can Modify "Boot\Config.txt".
It is possible to modify Config.txt because I have to modify it to adjust the default output screen resolution.

raffaeler, 2021-08-20:
@ChesterHuang999 I rapidly read the article and, at a certain point, it says "The Raspberry Pi’s onboard Wi-Fi, Bluetooth and GPIO do not work so you’ll need to use Ethernet or a USB Wi-Fi dongle to get online".
If there are those limitations, they will also affect PWM as well. I can't exclude there may be a way to enable them (as raspiconfig does under Linux) but I have no clue how to do it under Windows.

BTW, you could give a try with Windows 10 and see if this is a Win11 specific issue or, as I believe, you need the "Windows IoT" specific SKU to make the peripheral work.

pgrawehr, 2021-08-21:
I think there's nothing we can do here, until the low-level drivers are in place.