dotnet/iot #1656: Make RaspberryBoardInfo public

Loreno10, 2021-09-07:
**Is your feature request related to a problem? Please describe.**

I'd like to be able to get an information about the device that my code is running on. I see that the `RaspberryBoardInfo` class is internal (https://github.com/dotnet/iot/blob/54469318f33124e3455bf974e6a75167dfb831e6/src/System.Device.Gpio/System/Device/Gpio/RaspberryBoardInfo.cs).

**Describe the ideal solution**

Either `RaspberryBoardInfo` should be public or any other way of getting information about the device model should be retrievable.

raffaeler, 2021-09-07:
May I ask you the reasons why you need to distinguish the RPi model?
Please consider that the IoT libraries works on other devices as well, therefore I would find weird to have a class that is specific to the RPi.
I would rather prefer abstracting the info you need so that it may be used on other devices as well.

Loreno10, 2021-09-07:
@raffaeler Yes, you are right, it should be possible to get that information on any device that this library supports. I mentioned RPi, because that's what I'm using, and I found the `RaspberryBoardInfo` class in the repo.

I need to get the model of the device because I'm creating a device twin in the cloud when the device starts. One of the properties that I'd like to know about the device is its model because potentially I could run my app on many different devices, and I want to know what RPi version each of them is, just by looking at the data in the cloud.

raffaeler, 2021-09-07:
I understand the need, but I am not sure the `/proc/cpuinfo` output is homogeneous on all the devices and all the Linux distributions. 
Also, the `RaspberryBoardInfo` contains Linux specific code, while the IoT libraries can run on Windows as well.

In your specific file, since you are trying an exact match between the two, it can be simpler to just parse the few info you need from `/proc/cpuinfo` and compare them.

Loreno10, 2021-09-07:
Yes, I will do that for now. I just thought it would be nice if the libraries had an API to get the info with one line of code like `var device info = await System.Device.Model.GetDeviceInfo()` (just an example). I guess I'm not the only guy who wants to be able to get info on the hardware, it makes sense to have it implemented once for all.

raffaeler, 2021-09-07:
I have no other device similar to RPi but here we go with a comparison.
These are the final lines of `/proc/cpuinfo` on the RPi:
```
Hardware        : BCM2835
Revision        : a02082
Serial          : 00000000aa3252a42
Model           : Raspberry Pi 3 Model B Rev 1.2
```
I then tried on Centos (VM), Ubuntu (VM) and Ubuntu (physical) but they do not return any of those fields.
It looks like they only report the lowercase fields.

My guess is that those fields only apply to the Raspberri PI OS (aka Raspbian) distribution. 
I did not try the Ubuntu for RPi, but I don't expect to find those fields.

If anyone reading this issue has different info, please let us know.

pgrawehr, 2021-09-07:
The reason why we currently don't want to make that class public is because it's not clear how a stable API for it should look like. New models come out all the time, and when running different OSes on them, they might even identify differently. 

We would need a class structure that allows retrieving arbitrary system information for all kinds of OS/Hardware combinations. But then, there's probably nothing you could use that information for, except for printing it somewhere (because it will mostly be some kind of string). If you want to log the type of Raspberry Pi used (because your application runs only on Raspis) just log the contents of /proc/cpuinfo. You don't need a library function for that. For most other hardware related queries (e.g. Little endian/big endian, 32bit/64bit)  the framework already has appropriate functions.

joperezr, 2021-09-09:
@marcinjahn The big problem about making this type public, is that it lives in System.Device.Gpio library, which is a library that has [full support and backwards compat guarantees](https://github.com/dotnet/iot/tree/main/Documentation#repo-layout) which makes us uncomfortable to ship an API that a) only works on a RaspberryPi and b) the RaspberryPi must be running Raspbian for it to work. It is also an API that haven't gone through the full design review process (because it is just internal) so it is not ready to be shipped in System.Device.Gpio. That said, I can propose two alternatives here:
1. We could ship this API publicly in Iot.Device.Bindings library. This package, contrary to System.Device.Gpio, doesn't have the same level of support or guarantees so it allows us to change the API after shipping it publicly there. This will allow consumers to use the type and contribute to its logic to help fine-tuning it. We would still ship an internal version in System.Device.Gpio which we would use there, but consumers would now be able to get access to it via Iot.Device.Bindings.
2. We could suggest you using reflection to call the internal type in your code. Of course Reflection has it's down sides, like performance or a very fragile link that could be easily broken (given you would be doing private reflection) but that is an alternative that could work as well.

marcinjahn, 2021-09-09:
@joperezr It's a shame that this information is not standardized in Linux. I guess providing such a class that is usable only under the condition that you are using RPi in combination with Raspberry OS could be a bit limiting, or even confusing.
Although, on the other hand, it's generally better to have some option than not to have it at all. Tough call :|

Myself, I went with a simple `/proc/cpuinfo` reader class for now.

joperezr, 2021-09-09:
I see, yeah if you want to manually read cpuinfo yourself then that's fine and I guess we don't need to take any of the two approaches I suggested above and keep the type internal only. That probably also means that we can close this issue for now. If you still think that exposing this type as-is via Iot.Device.Bindings package would be valuable, let us know and we can re-open this and consider that option. Change should be trivial.

closed 2021-09-09 by joperezr