dotnet/iot #1659: INA219 - strange readings

marcinjahn, 2021-09-13:
**Describe the bug**

I have a small circuit involving INA219. At this point, during experimentation it looks the same as an example at https://github.com/dotnet/iot/blob/ca63c979bd8fd9c47acff379f09ff01ab8458abe/src/devices/Ina219/README.md.

I'm using the following code:

```csharp
public class INA219Driver : IElectricalParametersMonitor, IDisposable
{
    private readonly ILogger _logger;
    private readonly Ina219 _device;
    
    private const byte i2cBus = 0x1;
    private const byte i2cAddress = 0x40;
    
    public INA219Driver(ILogger<INA219Driver> logger)
    {
        _logger = logger ?? throw new ArgumentNullException(nameof(logger));
        _device = new Ina219(new I2cConnectionSettings(i2cBus, i2cAddress));
        _device.Reset();
        _device.BusVoltageRange = Ina219BusVoltageRange.Range16v;
        _device.PgaSensitivity = Ina219PgaSensitivity.PlusOrMinus40mv;
        _device.SetCalibration(33574, (float)12.2e-6);
        
        _logger.LogInformation("INA219 chip is initialized");
    }
    
    public double GetCurrent()
    {
        return _device.ReadCurrent().Milliamperes;
    }

    public void Dispose()
    {
        _device?.Dispose();
    }
}
```

I'm reading the current measurements in a loop every second, like this:

```csharp
while (true)
{
    _logger.LogInformation($"Current: {_electricalMonitor.GetCurrent()}");
    await Task.Delay(1000);
}
```

It's very similar to what you've shown in the README.md that I linked to at the top.

I observed that my current readings are very unstable. The results change every second significantly - it varies between 7mA and 15mA, which is not true.

To make sure that my INA219 is not damaged in some way I tried to run the Python sample from https://learn.adafruit.com/adafruit-ina219-current-sensor-breakout/python-circuitpython (the "Full Example Code" from the bottom of the page).
The results are much better. The current readings are stable, representing reality quite well.

Another weird thing that I observer is that the C# code returns current values even when there is no load at all! The results are then either 0mA or 799mA - it just jumps between these 2 values somewhat randomly. The Python code reports correctly 0 all the time.

Is there something wrong with I2C communication in the library?

**Steps to reproduce**

Provided

**Expected behavior**

Measurements of current should be stable.

**Actual behavior**

Results are not stable at all, and misrepresent reality.

**Versions used**

Device: RaspberryPi 4B 8GB (used both for build and running)
Dotnet:

```
pi@rpi:~/code/python-ina219 $ dotnet --info
.NET SDK (reflecting any global.json):
 Version:   5.0.400
 Commit:    d61950f9bf

Runtime Environment:
 OS Name:     raspbian
 OS Version:  10
 OS Platform: Linux
 RID:         linux-arm
 Base Path:   /home/pi/.dotnet/sdk/5.0.400/

Host (useful for support):
  Version: 5.0.9
  Commit:  208e377a53

.NET SDKs installed:
  5.0.400 [/home/pi/.dotnet/sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.App 5.0.9 [/home/pi/.dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 5.0.9 [/home/pi/.dotnet/shared/Microsoft.NETCore.App]

To install additional .NET runtimes or SDKs:
  https://aka.ms/dotnet-download

```

Libraries (Iot.Device.Bindings and System.Device.Gpio) are in version 1.5.0.

krwq, 2021-09-13:
cc: @Frankenslag

krwq, 2022-09-22:
[Triage] @marcinjahn are you still having this issue? we don't have the device so we weren't able to investigate this

marcinjahn, 2022-09-22:
Hi @krwq, if I recall correctly, eventually I ended up writing a small “driver” for INA219 in Python and invoking that thing from .NET. The Python implementation of INA219 was much more stable.

krwq, 2022-09-23:
@marcinjahn the implementation is pretty simple so there is probably some bug in there - if you're interested in taking a look - I can help with that on discord if needed but without device it's hard to make this better so we'll need some help

marcinjahn, 2022-09-23:
@krwq Unfortunately, I do not have the device anymore, it was just a short-term projekt at work :/

DocBrown101, 2023-05-17:
Unfortunately, I can confirm that. I wanted to use it for a small solar monitoring and at night it came again and again to milli ampere jumps of 800 mA.

[image]

Ellerbach, 2023-05-17:
@DocBrown101 can you filter those unique spikes? just remove anything that is not possible. Any hardware make measurement error, it's very usual in the embedded world. In your case, it's quite easy to handle. sometimes it's more complicated. In ADC, one of the pattern is to do long averaging when the reading are not super precise for example.

DocBrown101, 2023-05-17:
However, this behavior can only observed at idle and does not occur when used on an esp32.
Such extreme spikes cannot be filtered out so easily.

[image]

DocBrown101, 2023-05-19:
Yes, of course something like that always works, but makes the code unnecessarily complicated!

closed 2026-10-08 by raffaeler