dotnet/iot #1660: Consider adding timestamp to PinValueChangedEventArgs

stevehayles, 2021-09-13:
**It would be useful in some scenarios to expose the time of a pin change event**

The libgpiod driver specifically expose timestamps for any pin event and it would be useful to pass that timespan through to any subscribed event handler.  In situations where the driver does not expose the time from an underlying native event then the timespan could be added from managed code or left null

joperezr, 2021-09-14:
Thanks for the suggestion @stevehayles. Can you share the scenarios you had in mind where this could be useful? I'm not disagreeing of this being a good idea, just want to make sure that whatever API we end up proposing will work on the scenarios you had in mind.

stevehayles, 2021-09-14:
Thanks for looking at this @joperezr,  the specific scenario I am thinking of is in using interrupts on a GPIO pin to sample the output from a hall-effect water flow sensor.  At very high frequencies a pulse counter over a fixed time period is fine but I need to count down to almost zero flow rate and in that situation timing the pulses is a much better technique.

Adding timestamps within my application code in the event handler is OK but when the underlying native drivers expose a lower latency more consistent timestamp it makes sense to use that.  I think are other situations where the timestamp would be useful and it's often seen in other libraries like pigpio and wiringpi.

Ellerbach, 2021-09-16:
[Triage] The scenario is very valid and it seems to be something to add. Now, it is going to change the API definition and that should be done carefully even if it's just a change in the structure. 
So we should continue the discussion here to find the right name for the property and the value to be returned (Ticks or DateTime).

josesimoes, 2021-09-18:
Interestingly, we had similar requests a couple of times in nanoFramework. For devs needing to access the exact time stamp that the event occurred it seems a pretty reasonable request.

I suggest a property named TimeStamp. As for the type I would vote for ticks. 
Reasoning 🤓 
1. Anyone needing is probably looking for the best possible resolution, ticks is perfect for that.
2. DateTime merely "hides" ticks, which would make a developer life harder by having to add code to reach for the millisecs or ticks in order to get that level of resolution.
3. Comparing DateTime's it's more "expensive" (execution wise) than comparing ticks.
4. For those wanting a DateTime (there could be those too) there's a constructor from a ticks value.
5. DateTime type brings the time zone in tow which can be a potential source of confusion. If we go that path, DateTimeOffset would be a better type for this.

stevehayles, 2021-09-18:
Interesting observsations @josesimoes and I would agree with both the naming and use of ticks in this scenario.  Any application code can easily construct 'higher level' DateTime objects from ticks and in very hot paths the overhead of DateTime or DateTimeOffset is hard to justify.  I would be particularly concerned about DateTimeOffset which has a reasonable performance penalty over DateTime in my testing.  In many scenarios the timestamp is used in to calculate a duration or is subject to some form of digital filtering and the use of ticks works well for both these cases

Ticks would loose some resolution from a library like Lipgpiod which uses a Timespec with nanosecond resolution but given the entire dotnet framework is constrained to the same minimum time period and given the non-realtime nature of pretty much all the OS's this will run on it seems the accuracy is very unlikely to approach anything like a single tick

raffaeler, 2021-09-19:
I agree with the ticks, but I would rather focus on the resolution/precision. In this discussion we are giving for granted the 100ns interval provided by datetime/timespan.
But I would not limit to that, therefore a unit of measure should be exposed as well to avoid incorrect conversions.
There is also another important parameter that is the error percentage which may vary depending on the originating hardware. For example, it is useful to understand whether a communication would be affected from any anticipated/delayed signal.

Ellerbach, 2021-09-19:
Ticks seems indeed a great choice to avoid overhead. On your comment:

> There is also another important parameter that is the error percentage which may vary depending on the originating hardware. For example, it is useful to understand whether a communication would be affected from any anticipated/delayed signal.

You always need to know your hardware when making things like this. Even if ticks can be different from platform to platform, you can always know how many ticks represent 1 nano second, so you can always make some good precise calculation if you want.

And I do fully agree on this:

> given the non-realtime nature of pretty much all the OS's this will run on it seems the accuracy is very unlikely to approach anything like a single tick

You can't get such a precision that you'll be at a nano second and on a real OS it doesn't even make sense. It's already hard on a MCU running bare metal without even an RTOS so on a real OS, just forget about it :-)

raffaeler, 2021-09-19:
Hi @Ellerbach 
> You always need to know your hardware when making things like this [cut]

While the OS has an overhead, you may still rely on the reading with a higher precision of your capacity to react to the event.
If the event was generated by a hw timer or by the OS in ring0, you can get better and reliable prevision. This timing can be written in the event and used, later on, in ring 3 by your application.
This can be very useful to compute the distance among events, but not to react to those events (like sustaining a RX/TX protocol).

For the OS I fully agree that it doesn't make any sense unless you are running in ring 0, but for a HW timer it does. Let's look at this parameter as a "capability" of the generator.

josesimoes, 2021-09-19:
Please correct me if I'm wrong, but on all .NET platforms - IoT Core included - a tick is **a** tick just like defined/explained in the docs:

"A single tick represents one hundred nanoseconds or one ten-millionth of a second. There are 10,000 ticks in a millisecond (see TicksPerMillisecond) and 10 million ticks in a second."

Therefore, any tick value that will be exposed through such property will the best possible and more accurate that the implementation can produce.

If I understood correctly your concern, you're suggesting that there should be extra data on" "how long" the actual event takes from the instant it occurs up to the instant the C# property is filled in. That would have to take into account the time it takes for the hardware block to process the signal, how long the interrupt handler took to get CPU time and serve the interrupt, process it, the time the code that bubbles up that up to the execution engine.

To mee that seems waaaay overengineered and way out of scope for the purpose of this platform. And that's hardware dependent, as Laurent pointed out.

raffaeler, 2021-09-19:
I am just suggesting to do the same of StopWatch did: https://docs.microsoft.com/en-us/dotnet/api/system.diagnostics.stopwatch.frequency?view=net-5.0
It can be either low or high resolution, but still internally it can be converted in seconds.

I am therefore asking not to fix the tick, but to provide tick and conversion unit into seconds so that a higher resolution can be obtained, when available. The reason is exactly because the hw can change and can offer different resolution.
The resolution field does not necessarily need to be part of the event. For Pins, it can be part of the controller as it does not change dynamically.

crozone, 2022-09-01:
I have a few usecases that would also benefit from this, exposing the libgpio pin change timestamp would be hugely beneficial.

A simple example is the [Wiegand protocol](https://en.wikipedia.org/wiki/Wiegand_interface). It simply works with two data lines, DATA0 is pulled low to send a 0 bit, DATA1 is pulled high to send a 1 bit. There is no fixed clock speed, it could in theory even be variable, all that matters is the ordering of the falling edge events on each pin.

When the pin changed events fire on the separate pins, there is no guarantee that they arrive in order, which is problematic. In fact during first run we see much higher startup latency. This is possibly from the JIT, it happens with R2R as well, but marshaling in .NET 6 still requires JIT to work. This often causes the pin changed events to arrive out of order, leading to bad data. We can easily verify this behaviour using an external logic analyzer.

Once the code is warm, things improve greatly, but there's still absolutely no guarantee that in a pre-emptively multitasking system (and a multicore one at that) that the events will arrive with any sort of predicable order. AFAIK the events are scheduled and run on the threadpool as well, adding an extra layer of unpredictability.

Exposing the libgpiod pin event timestamps would solve this. It would allowing us to buffer the pin changed events and then order them correctly by the timestamp. For this usecase it doesn't even matter what units the timestamps are in, they could even be a simple unit-less `ulong` since the only thing that matters is that they are sortable. However I agree with the other comments that exposing a tickrate, just like `Stopwatch.Frequency` would definitely be desirable for applications that are more timing critical.

crozone, 2022-09-01:
I have a few usecases that would also benefit from this, exposing the libgpio pin change timestamp would be hugely beneficial.

A simple example is the [Wiegand protocol](https://en.wikipedia.org/wiki/Wiegand_interface). It simply works with two data lines, DATA0 is pulled low to send a 0 bit, DATA1 is pulled high to send a 1 bit. There is no fixed clock speed, it could in theory even be variable, all that matters is the ordering of the falling edge events on each pin.

When the pin changed events fire on the separate pins, there is no guarantee that they arrive in order, which is problematic. In fact during first run we see much higher startup latency. This is possibly from the JIT, it happens with R2R as well, but marshaling in .NET 6 still requires JIT to work. This often causes the pin changed events to arrive out of order, leading to bad data. We can easily verify this behaviour using an external logic analyzer.

Once the code is warm, things improve greatly, but there's still absolutely no guarantee that in a pre-emptively multitasking system (and a multicore one at that) that the events will arrive with any sort of predicable order. AFAIK the events are scheduled and run on the threadpool as well, adding an extra layer of unpredictability.

Exposing the libgpiod pin event timestamps would solve this. It would allowing us to buffer the pin changed events and then order them correctly by the timestamp.

`libgbpiod` uses `std::chrono::nanoseconds` so as discussed, `TimeSpan` won't cut it. It really should be a 64 bit signed `long` timestamp value. Other GPIO drivers may use a completely different tickrate so it could be exposed as a `static readonly long TimestampFrequency` on the GPIO driver class. There could also be a helper method `ToTimeSpan(long timestamp)` that performs the conversion to `TimeSpan` when needed.

A complication: Is the timestamp source and frequency ever likely to change depending on the pin number itself? As in, is there ever likely to exist a GPIO driver that manages multiple GPIO hardware interfaces that use different timestamp sources?  And in such a case, should the driver just normalize it to a standard tickrate anyway?

krwq, 2022-09-01:
I haven't read the thread above and perhaps it was already mentioned but if we go for ticks I think we should also expose static readonly property to provide ticks per seconds in case we need to make it more accurate in the future. I'd also make sure the type is 64-bit and name property to have `Ticks` somewhere. The docs should clearly mention that `TicksPerSeconds` should be used to resolve that to time (or we document that we use i.e. TimeSpan.TicksPerSecond).

Also if .NET doesn't have any builtin type which could hold this timestamp I'd recommend to create a suggestion in dotnet/runtime to add such type and try to use it here. Especially this API will have to go in System.Device where we're not likely to make breaking changes. Optionally we can make it internal API for now and expose it through extension method in Iot.Device.Bindings - this way we will be able to make breaking change once this ships in framework (assuming it's actually needed)

crozone, 2022-09-01:
> Also if .NET doesn't have any builtin type which could hold this timestamp I'd recommend to create a suggestion in dotnet/runtime to add such type and try to use it here.

@krwq What's wrong with `long`?

krwq, 2022-09-01:
type has more advantages like i.e. you can make it easier to convert to other time related types. The question is if dotnet/runtime is planning to ever add type which represents ticks. If it's not then long/ulong might be ok, otherwise I'd prefer we waited for that type.

josesimoes, 2022-09-01:
Following my reasoning above, I would say that `Ticks` should be enough to hold this and it's more _meaningful_ than `long`.

Even if libgbpiod can offer a better resolution than that for the event timestamp, I don't think that's useful at all considering all that has been already discussed.

pgrawehr, 2022-09-01:
What resolution do we need? For reference: The resolution of Datetime and Timespan is 100ns, the resolution of TickCount64 only 1ms.

crozone, 2022-09-01:
> What resolution do we need? For reference: The resolution of Datetime and Timespan is 100ns, the resolution of TickCount64 only 1ms.

The libgpiod timestamp resolution is 1ns. It's possible that in reality, most hardware doesn't actually provide that level of timestamp resolution, so `TimeSpan` could potentially suffice... but for a low level interface, is it really worth throwing away the precision provided by the raw value simply to meet an abstraction? It feels wasteful to not expose the actual hardware provided value *at all*.

An option is that `PinValueChangedEventArgs` could simply expose both directly. It could have a  `public long HardwareTimestamp` member, as well as a `TimeSpan Timestamp` getter that performs the conversion based on the underlying hardware timestamp tick rate.

Providing a conversion method somewhere like `GetTimeSpanFromHardwareTimestamp(long timestamp)` is still a must if the hardware timestamp is exposed. This would allow arithmetic to be performed on the raw timestamps in full precision and then converted to `TimeSpan` afterwards.

As a useful reference, the [Windows.Devices.Gpio](https://docs.microsoft.com/en-us/uwp/api/windows.devices.gpio?view=winrt-22621) GPIO implementation on Windows has a different, buffer based mechanism for detecting GPIO changes. However the [GpioChangeRecord](https://docs.microsoft.com/en-us/uwp/api/windows.devices.gpio.gpiochangerecord?view=winrt-22621) struct found in the namespace is basically equivalent to `PinValueChangedEventArgs` in this context, and it contains a `public System.TimeSpan RelativeTime;` field. So the designers of that implementation probably decided that `TimeSpan` was adequate on Windows at least.

josesimoes, 2022-09-01:
[TimeSpan](https://docs.microsoft.com/en-us/dotnet/api/system.timespan?view=net-6.0) represents a time interval. 
What would be the time "reference" that each time stamp will be referring to?  

The comparison that you use with `GpioChangeRecord` demonstrates exactly that: it was meant to _compare_ time between two "capture" events. I don't think this is what's being discussed here.

At most this could be a [`DateTimeOffset`](https://docs.microsoft.com/en-us/dotnet/api/system.datetimeoffset?view=net-6.0) but that has two points against it:
1. That doesn't exist in .NET nanoFramework.
2. We go back to the extra overhead of "encoding" and "decoding" a plain simple long value.

Regarding `HardwareTimestamp`: I believe that's adding too much weight on the name... why _hardware_? 
I would agree with `Timestamp` instead of `Ticks`. This name is already used in several other .NET classes. 
But, then again, coming up with names can be hard... 😅

crozone, 2022-09-02:
> What would be the time "reference" that each time stamp will be referring to?

As I understand it, the actual timestamp value is completely arbitrary. It might be based on some time since boot, or when the hardware was initialized, but what is actually important is that it's monotonic and can be used to derive accurate time deltas. Algorithms that use the timestamp value are almost always going to be concerned with time deltas between pin events so they can implement some wire protocol on very short timescales. This task requires relative timestamps at high precision, which is exactly what the libgpiod timestamp aims to be.

How the timestamp relates to actual UTC time is almost completely irrelevant, and even attempting to derive a `DateTime` value for a pin event opens its own can of worms. The tick value provided by libgpiod is a monotonic timestamp, whereas `DateTime` derived from an RTC is subject to skewing and updates from `ntp` or other sources make `DateTime` not great for accurately measuring durations of time.

> Regarding HardwareTimestamp: I believe that's adding too much weight on the name... why hardware?

Good point, hardware is probably too specific. Using  a `long Timestamp` is simpler and has precedent in the `Stopwatch` class:

* `Stopwatch.GetTimestamp()` is a "A long integer representing the tick counter value of the underlying timer mechanism.". 
* `Stopwatch.Frequency` "Gets the frequency of the timer as the number of ticks per second.".

I think it makes sense that the pin event timestamp follows the same convention as `Stopwatch`, since both are solving the same problem. They both need to deal with high resolution monotonic timestamps of platform-specific resolution (hence giving the timestamp as `long`, and a `Frequency` property exposing the tick rate).

So a `public readonly long PinValueChangedEventArgs.Timestamp` would be consistent with what is already in the framework. A trickier question is where to store the `.Frequency`. I feel like `long GpioDriver.Frequency` is probably correct, but not 100% on that.

stevehayles, 2022-09-30:
With https://github.com/dotnet/runtime/issues/23799 in the upcoming .NET 7 release  it looks like TimeStamp, DateTime, DateTimeOffset, and TimeOnly will natively handle microseconds and nanoseconds.

crozone, 2022-10-01:
> With [dotnet/runtime#23799](https://github.com/dotnet/runtime/issues/23799) in the upcoming .NET 7 release it looks like TimeStamp, DateTime, DateTimeOffset, and TimeOnly will natively handle microseconds and nanoseconds.

This appears to expose nanoseconds as a property directly but doesn't actually increase the tick resolution of TimeSpan. So `NanosecondsPerTick = 100`, meaning that it still doesn't provide 1ns precision. Whether or not this matters is still debatable.

crozone, 2023-07-24:
Any movement on this? Should I work on a pull request to add a `long Timestamp` to the event args?

Ellerbach, 2023-07-25:
> Any movement on this? Should I work on a pull request to add a `long Timestamp` to the event args?

Yes, please go ahead. Review may take a bit of time to make sure we aligne properly.