dotnet/iot #1663: Mcp23017 initialization resets pin values

dme-development, 2021-09-17:
**Describe the bug**

Every time a new instance of Mcp23017() is being created the pins are being reset to low. This is an issue when the written software reboots, eg when updating the software version.

**Steps to reproduce**

```csharp
            var pinNumber = 0;

            var i2cConnectionSettings = new I2cConnectionSettings(1, 0x20);

            var i2cDevice = I2cDevice.Create(i2cConnectionSettings);

            var device = new Mcp23017(i2cDevice);

            var controller = new GpioController(PinNumberingScheme.Logical, Mcp23017Device);

            controller.Write(pinNumber, PinValue.High);

            var device2 = new Mcp23017(i2cDevice);

            var pinValue = controller.Read(pinNumber);
```

**Expected behavior**

Pins need to keep their value after re-initializing the Mcp23017() adapter.

**Actual behavior**

Pins are being reset every time a new instance of Mcp23017() is being declared. Looks like it is embedded in the init code here:
https://github.com/dotnet/iot/blob/main/src/devices/Mcp23xxx/Mcp23xxx.cs - starting at line 84.

dme-development, 2021-09-23:
> Do I see it correctly that your use case is that you drive some external hardware (relays or similar) using the Mcp23017 and you want to be able to update/restart the software without these changing?

@pgrawehr That is exactly our use case: when updating a docker image version, we want the relays to keep their values.

We have created our own PCB's, pull-up/pull-down is currently not an option, but we created a workaround by removing the reset functionality from the library. Of course, we would like to see it being resolved in the library.

I think there have been similar issues for other device bindings, eg: #1540

pgrawehr, 2021-09-23:
Yes, this is an use case that has been forgotten in the initial design of several APIs (see also my links above). You are welcome to submit a PR, but try not to break existing APIs (that means that you probably have to add an extra bool argument to the ctor).

krwq, 2022-04-28:
[Triage] we need to fix initialization similarly to what we did here: https://github.com/dotnet/iot/issues/1479

managedmetal, 2026-10-10:
I'd like to pick this up and contribute a tested fix.

I've reproduced it on a Raspberry Pi 5 with a MIKROE Pi Click shield and an Expand 2 Click module (MCP23017).  I've built a program that sets outputs and restarts, after which the pins are switched back to inputs by the constructor.

Two ways to fix this (both built and proven with tests):

1. Add an overload to every chip class that lets developers skip the initialization, keeping today's behavior as the default, e.g. `new Mcp23017(device, initialize: false)`.
2. Change the default, in the spirit of #1520: constructors stop modifying the pin directions/states, and an explicit `Reset()` method resets all pins to inputs (defaults) for existing code that needs that.

Do you have a preference between these two?  I'd recommend option 2 to correct the constructor's behavior; but option 1 avoids changing behavior for users that rely on today's constructor-time pin reset.

Either solution should also fix the `Toggle` bug: the driver currently assumes that a freshly opened pin is always low, so the first `Toggle` on a pin which is high doesn't actually toggle.

pgrawehr, 2026-10-15:
@managedmetal I think solution 1 would be better, as resetting the state is probably ok for many existing use cases. In #1520 we also added new overloads for "not changing" while opening IIRC.