dotnet/iot #1682: WS2812B on SPI1 behaves different than on SPI0

OJapes, 2021-10-08:
**Describe the bug**

When trying to run the sample [from the repo](https://github.com/dotnet/iot/tree/main/src/devices/Ws28xx#raspberry-pi-setup-bootconfigtxt) with using SPI1 (GPIO 20) instead of SPI0 (GPIO 10), the strip lights up as almost completely white. A hint of color can be seen (red becomes a very light pink) can be noticed, as well as changes when using the TheatreChase. But when using e.g. Black in ColorSwipe, the strip becomes white.

Because we need to use SPI0 for another device, we need to switch to SPI1. Alternatively, an implementation with PWM would possibly work.

**Steps to reproduce**
Enable spi1 by adding 
```
dtoverlay=spi1-1cs
```

to /boot/config.txt and reboot.

Then change
```csharp
SpiConnectionSettings settings = new(0, 0)
```

to 
```csharp
SpiConnectionSettings settings = new(1, 0)
```
and run the code

**Expected behavior**

The strip color should be the same when using either SPI0 or SPI1.

**Actual behavior**

The color is almost completely white when using SPI1.

**Versions used**

- `dotnet --info` on the machine being used to build: 5.0.401
- `dotnet --info` on the machine where app is being run: 5.0.400
- Version of `System.Device.Gpio` package: 1.5.0
- Version of `Iot.Device.Bindings` package (not needed if bug is in `System.Device.Gpio`): 1.5.0

raffaeler, 2021-10-08:
I currently have no devices to make a test, sorry. Let's see who has the chance to test it.

rg0815, 2022-07-10:
I have the same problem. 
Is there a solution in the meantime?

Ellerbach, 2022-07-10:
Note: I can recommend you to check how to active properly the different SPI here: https://github.com/dotnet/iot/blob/main/Documentation/raspi-spi.md

krwq, 2022-07-11:
Note that WS2812B requires `core_freq` to be set (https://github.com/dotnet/iot/tree/main/src/devices/Ws28xx#raspberry-pi-setup-bootconfigtxt) - I do not know if that applies to both SPI0 and SPI1 but assuming you've set it and you're having issue I'd guess no. You might want to read notes in that file to figure out if `core_freq` is the right setting or you need to use something else for SPI1

closed 2022-09-30 by msftbot[bot]