dotnet/iot #1688: VideoDevice issues with color accuracy.

CodedBeard, 2021-10-13:
**Describe the bug**

While testing various different cameras (microsoft usb lifecam, raspberry pi camera v2, raspberry pi HD cam, raspberry pi nightvision cam) on a selection of different Raspberry Pi (all 4b but with varying ram and armv7/arm64 os) I've noticed that the color accuracy does not match that of the inbuilt raspistill/raspivid tools for capturing from the same cameras. In all instances the captured image is retuned with to much green.

During futher testing, I noticed that applying a `ColorEffect` other than `None` would sometimes return an accurate color for the given effect (`BW`/`Sepia` both appear accurate) but others also appear to have too much green in the image.

**Steps to reproduce**

This can be reproduced using the MJPEG sample project in the repo.

**Expected behavior**

The captured image matches the color accuracy of the built in raspistill/raspivid tools like the below example:

[image]

**Actual behavior**

Iot using the same settings as raspistill/raspivid produces an image with a green hue like below:

[image]

**Versions used**

```
dotnet --info
.NET SDK (reflecting any global.json):
 Version:   6.0.100-rc.1.21458.32
 Commit:    d7c22323c4

Runtime Environment:
 OS Name:     raspbian
 OS Version:  10
 OS Platform: Linux
 RID:         linux-arm
 Base Path:   /home/pi/dotnet/sdk/6.0.100-rc.1.21458.32/

Host (useful for support):
  Version: 6.0.0-rc.1.21451.13
  Commit:  d7619cd4b1

.NET SDKs installed:
  6.0.100-rc.1.21458.32 [/home/pi/dotnet/sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.App 6.0.0-rc.1.21452.15 [/home/pi/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 6.0.0-rc.1.21451.13 [/home/pi/dotnet/shared/Microsoft.NETCore.App]
```
- Version of `System.Device.Gpio`: built from main
- Version of `Iot.Device.Bindings` built from main

joperezr, 2021-10-21:
[Triage] Thanks a lot for reporting this and for the testing.  We definitely would want to match raspistill/raspivid so we should investigate what is going on in here. Can you please share a repro project that captured that image above? Also if you can share exactly which camera was used that would be great as well, so that we can investigate and figure out what is going on.

CodedBeard, 2021-10-21:
Hi @joperezr, I can be reproduced this with the [MJPEG Sample](https://github.com/dotnet/iot/tree/main/src/devices/Media/VideoDevice/samples/MjpegStream/) in the repo. The current sample is functionally identical to my project, as thats what I based the updates on with #1587. In confirming I could replicate the issue with the sample in the repo, I've noticed it doesn't currently run, which I'll submit a PR to resolve.

The camera used for these specific shots was a [Raspberry Pi Camera Module V2](https://www.raspberrypi.com/products/camera-module-v2/), however as mentioned, I've had the same results across mutiple camera devices.

CodedBeard, 2021-11-16:
@jesseryoung thanks for the heads up, but as far as I recall, the capture methods don't use the converters internally, and the mjpeg sample is capturing as jpeg and just returning the bytes.

I thought I'd double check this, and if I update the sample to use YUYV capture format, and then run the resulting bytes through the converter, it produces the same pixel errors you describe in #1726. While not very obvious taking a picture of the same wall, you can see the incorrect pixels in both bottom corners.

[image]

joperezr, 2022-06-13:
Opening as per comment https://github.com/dotnet/iot/issues/1799#issuecomment-1154248881

krwq, 2022-08-25:
[Triage] We should either support libcamera or throw a useful exception

raffaeler, 2023-11-10:
My new https://github.com/dotnet/iot/pull/2150 and will soon be packaged in version 3.1 as soon as the milestone is reached: https://github.com/dotnet/iot/milestone/7

I invite all the people in this thread to try it and tell whether we can obsolete the VideoDevice binding or not.
I am not sure if it is possible to create a "fake" VideoDevice class that redirects all the calls to the new binding. At a first glance it doesn't fully fit.

Please @CodedBeard tell us if you can migrate the code to the new binding and close this issue.

CodedBeard, 2023-11-13:
@raffaeler I'll give it a try later in the week when I've got some time.

raffaeler, 2023-11-13:
Great, thank you @CodedBeard

CodedBeard, 2023-11-14:
@raffaeler Started having a play with the new code, and immediately ran into an issue. Using the included sample app, the first thing I tried was using the `list` command which failed with an exception:
```bash
~/src/iot/src/devices/Camera/samples/Camera.Samples $ dotnet run list
Using the following command line:
libcamera-still --output - --list-cameras

Output being parsed:

Unhandled exception. System.Exception: This text does not comply to the expected list of available cameras
   at Iot.Device.Camera.CameraInfo.From(String listOutputString) in /home/pi/src/iot/src/devices/Camera/Camera/CameraInfo.cs:line 40
   at Camera.Samples.Capture.List() in /home/pi/src/iot/src/devices/Camera/samples/Camera.Samples/Capture.cs:line 136
   at Camera.Samples.Program.Main(String[] args) in /home/pi/src/iot/src/devices/Camera/samples/Camera.Samples/Program.cs:line 48
   at Camera.Samples.Program.<Main>(String[] args)
```

If I run the same command it was trying to run, it does return the expected result:

```bash
~/src/iot/src/devices/Camera/samples/Camera.Samples $ libcamera-still --output - --list-cameras
Available cameras
-----------------
0 : imx708_wide [4608x2592] (/base/soc/i2c0mux/i2c@1/imx708@1a)
    Modes: 'SRGGB10_CSI2P' : 1536x864 [120.13 fps - (768, 432)/3072x1728 crop]
                             2304x1296 [56.03 fps - (0, 0)/4608x2592 crop]
                             4608x2592 [14.35 fps - (0, 0)/4608x2592 crop]
```

I then tried `still-libcamera` and `video-libcamera`, both of which worked fine.

```bash
~/src/iot/src/devices/Camera/samples/Camera.Samples $ dotnet run still-libcamera
Using the following command line:
libcamera-still --output - --timeout 1 --vflip --hflip --quality 90 --encoding jpg --width 640 --height 480

Captured the picture: 20231114_1758270300.jpg

~/src/iot/src/devices/Camera/samples/Camera.Samples $ dotnet run video-libcamera
Using the following command line:
libcamera-vid --output - --timeout 0 --flush --vflip --hflip --width 640 --height 480

Captured the video: 20231114_1800320187.h264
```

Can't see any obvious reason why it thinks the output to parse is empty, but thought id let you know.

raffaeler, 2023-11-14:
Thank you for testing @CodedBeard 
I made a quick test by testing the output you provided into the Camera Info test and it worked:
[image]

I assumed the output was:
```
Available cameras
-----------------
0 : imx708_wide [4608x2592] (/base/soc/i2c0mux/i2c@1/imx708@1a)
    Modes: 'SRGGB10_CSI2P' : 1536x864 [120.13 fps - (768, 432)/3072x1728 crop]
                             2304x1296 [56.03 fps - (0, 0)/4608x2592 crop]
                             4608x2592 [14.35 fps - (0, 0)/4608x2592 crop]
```

The exception you see comes from this code:
```csharp
if (listOutputString == null || listOutputString.Length < Line1.Length + Line2.Length)
{
    throw new Exception($"This text does not comply to the expected list of available cameras");
}
```
Where `Line1` and `Line2` are defined as:
```csharp
private const string Line1 = "Available cameras";
private const string Line2 = "-----------------";
```
and `listOutputString` is the output coming from the `--list-cameras` command line.

I will investigate more but the output coming from your installation doesn't look the culprit.
If you can try to debug the issue on your side, it could be very precious because I am afraid not to be in the same conditions to repro the issue.

Apart from this, if you believe that the "color accuracy" problem is resolved with this binding, I believe it would be better to close this issue and open a new one you found. Does it work for you?

CodedBeard, 2023-11-14:
@raffaeler np, thought id just mention it in case there was anything obvious that could be causing it that I was missing.

I'm just trying to replicate the original setup to confirm that the color accuracy issue is resolved, and yep happy to open a separate issue for that exception if i don't find the cause in the meantime

CodedBeard, 2023-11-15:
@raffaeler good news, all the cameras I've tested so far no longer have the color accuracy issue. I've come across a few other issues while testing, but I'll log those separately when I've had some more time to investigate whats going wrong.

closed 2023-11-15 by CodedBeard

raffaeler, 2023-11-15:
Great to hear that, thank you @CodedBeard .

When you find issues on the new Camera binding, I kindly ask you to post:
- the generated command line
- the result you obtain by running the command line manually on the terminal

PR #1726 (2021-11-24): Clamp YUV values to be within byte ranges
This PR adds a clamp in `VideoDevice.YuvToRgb` function. 

In some cases the formula for converting from YUV to RGB used here can result in negative R, G and B values as well as values over 255. This causes some color inaccuracies. 

On the left I converted the YUYV data coming from a video capture device to a jpg using `VideoDevice.YuyvToRgb`, on the right is the same frame converted using the clamp method in this PR.
[image]

I believe this may also fix #1688. 

###### Microsoft Reviewers: [Open in CodeFlow](https://portal.fabricbot.ms/api/codeflow?p [... cut]