dotnet/iot #1715: Button binding invokes the Pressed event at app startup time

raffaeler, 2021-11-09:
**Describe the bug**
Hardware: Raspberry PI3B+ with a physical button attached.

The [Button](https://github.com/dotnet/iot/tree/main/src/devices/Button) binding can be configured in three different configurations: simple input, input pull-up and input pull-down. By default the constructor configures it with a pull-up (internal) resistor.
The issue happens only when using the pull-up configuration.

When the application is run, the `Pressed` event-handler is invoked (during the configuration time) even if the button was stale. This behavior only happens the very first time the application is run. Re-running the application does not cause this behavior anymore.
After rebooting the RPi, the first time the application is run, the problem appears again.

**Steps to reproduce**

```csharp
private static readonly int _pin1 = 22;

static void Main(string[] args)
{
    GpioButton button = new(_pin1, null, true, PinMode.InputPullUp);
    button.Press += (sender, args) =>
    {
        Console.WriteLine("Button Pressed");
    };

    Console.ReadKey();
}

```

**Expected behavior**
Since the button has not been pressed, nothing should be printed on the console.

**Actual behavior**
```
pi@piraf1:~/button$ ./ButtonTest
Button Pressed
```

**Versions used**

**dotnet --info**
.NET SDK (reflecting any global.json):
 Version:   6.0.100-rc.2.21505.57
 Commit:    ab39070116

Runtime Environment:
 OS Name:     Windows
 OS Version:  10.0.19043
 OS Platform: Windows
 RID:         win10-x64
 Base Path:   C:\Program Files\dotnet\sdk\6.0.100-rc.2.21505.57\

Host (useful for support):
  Version: 6.0.0-rc.2.21480.5
  Commit:  6b11d64e7e

.NET SDKs installed:
  2.1.526 [C:\Program Files\dotnet\sdk]
  2.1.818 [C:\Program Files\dotnet\sdk]
  3.0.100 [C:\Program Files\dotnet\sdk]
  3.1.414 [C:\Program Files\dotnet\sdk]
  5.0.100-rc.1.20452.10 [C:\Program Files\dotnet\sdk]
  5.0.104 [C:\Program Files\dotnet\sdk]
  5.0.402 [C:\Program Files\dotnet\sdk]
  6.0.100-rc.2.21505.57 [C:\Program Files\dotnet\sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.All 2.1.1 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.2 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.4 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.12 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.13 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.15 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.16 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.20 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.21 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.30 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.App 2.1.1 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.2 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.4 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.12 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.13 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.15 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.16 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.20 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.21 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.30 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.1.2 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.1.20 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 5.0.0-rc.1.20451.17 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 5.0.4 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 5.0.10 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 5.0.11 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 6.0.0-rc.2.21480.10 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 2.1.12 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.13 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.15 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.16 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.20 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.21 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.30 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.1.2 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.1.20 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 5.0.0-rc.1.20451.14 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 5.0.4 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 5.0.10 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 5.0.11 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 6.0.0-rc.2.21480.5 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.WindowsDesktop.App 3.1.2 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 3.1.17 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 3.1.18 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 3.1.19 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 3.1.20 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 5.0.0-rc.1.20452.2 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 5.0.4 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 5.0.10 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 5.0.11 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 6.0.0-rc.2.21501.6 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]

- Version of `System.Device.Gpio` package: current main (from sources)
- Version of `Iot.Device.Bindings` package: current main (from sources)
- App deployed as self-contained

raffaeler, 2021-11-09:
I instrumented the Button code to see the sequence of the events in relationship with the Button code:
```
pi@piraf1:~/button$ ./ButtonTest
OpenPin
RegisterCallback
ctor end
Falling
Rising
Button Pressed
```

The initial events are inside the Button constructor, while the `Falling` and `Raising` log happen inside the PinStateChanged.

raffaeler, 2021-11-09:
I suspect that the delay time between setting the pull-up and the effective change depends on the electrical "R-C" of the pin where R is the resistance value of the pull-up and "C" the capacitance applied of the GPIO pin.

If I add the following line after `OpenPin` but before `RegisterCallback...`, it obviously works correctly:
```
var waitResult = _gpioController.WaitForEvent(_buttonPin, PinEventTypes.Rising, TimeSpan.FromMilliseconds(10));
```
But the `waitResult.TimedOut` is always true which is not a great way to handle this situation (which may vary depending on the hardware).

If instead I open the pin setting `PinValue.High` as initial value, nothing changes (it doesn't fix the isssue).
Similarly, it does not help adding some code at the very beginning of the Button constructor that opens the pin as output, set the high value and closes the pin.

At this point I wonder whether the SetPinMode should wait for the pin value to go either high or low depending on the configured pull-up/down. This is probably the most correct behavior.

raffaeler, 2021-11-11:
After a long discussion in the Triage, we agreed that the native subscription to the pin changed event should be moved out from the constructor.
Also, we agreed on the best place where to move it, which is the Button 'add' event handlers, while leaving the removal in the Dispose method (no changes).
This solution leaves the native subcription under user's control. It also offers the opportunity to add a delay between the button constructor and the event subscription which may depend on the underlying hardware.

krwq, 2022-08-25:
[Triage] @raffaeler please validate if this is still an issue

pedroveiga99, 2024-01-08:
Is there any updates on this? I still get the same problem (expected, since this is still open). I believe that @raffaeler is right, it must be something to do with the timings. I have a workaround (that I don't like very much...)

```csharp
GpioButton button = new(_pin1, null, true, PinMode.InputPullUp);
await Task.Delay(500);
button.Press += (sender, args) =>
{
    Console.WriteLine("Button Pressed");
};
```

Is there any expectation of this being resolved? Also I believe that this should have a higher priority since it has been +3 years and it is still not solved.

raffaeler, 2024-01-08:
@pedroveiga99 It's on me, sorry. I am the assignee and planned to do during the holidays, but I have to resolve some issues on my fork first.
The fix should be straightforward.
Stay tuned.