dotnet/iot #1719: The Button implementation does not provide debouncing

raffaeler, 2021-11-10:
## Feature request
The current https://github.com/dotnet/iot/tree/main/src/devices/Button implementation does not provide any debouncing mechanism.

There is another issue (https://github.com/dotnet/iot/issues/674) requesting debouncing at a lower level. Apparently the Linux driver does not provide any debouncing mechanism.

It would probably better to implement debouncing at least at button level, given that it is very unpractical to use a button without debouncing.

pgrawehr, 2021-11-10:
I agree, on the button binding this makes sense. I wouldn't put it on the general drivers, because it would add domain-specific knowledge where it doesn't belong and/or suggest hardware support for something that will, in many cases, be simulated in software.

raffaeler, 2021-11-10:
I first tried to see if it was available in the native driver but I could not find anything like that. Maybe @joperezr can either confirm or not about my research.
In any case, I already submitted a PR for the button changes so that it is immediately available.

closed 2021-11-12 by raffaeler