dotnet/iot #1724: VideoDevice does not work on raspberry os arm64 rtm

CodedBeard, 2021-11-11:
**Describe the bug**

The RTM release of [raspberry pi arm64 os](https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2021-11-08/), based on bullseye, was released earlier this week. When attempting to run the sample MJPEG project on a raspberry pi 4 using this OS, the code runs without error, however, invalid images are produced.

I suspect this is due to the fact that the [os now relies on the unicam driver](https://www.raspberrypi.com/documentation/accessories/camera.html#driver-differences-when-using-libcamera-or-the-legacy-stack), instead of the original broadcom implementation.

**Steps to reproduce**

run the sample MJPEG project on a device using the new raspberry pi os

**Expected behavior**

an image is produced with the following starting bytes, captured in vscode:

[0] [byte]:255
[1] [byte]:216
[2] [byte]:255
[3] [byte]:225

**Actual behavior**

no errors occur, but the frames returned are not valid.

again debuging with vscode, the following bytes are returned for the start of the image:

Example 1:

[0] [byte]:16
[1] [byte]:17
[2] [byte]:16
[3] [byte]:16

Example 2:

[0] [byte]:17
[1] [byte]:17
[2] [byte]:16
[3] [byte]:16

Example 3:

[0] [byte]:16
[1] [byte]:16
[2] [byte]:16
[3] [byte]:16

my tests consistenly result in some combination of 16/17 as the starting bytes in the frame.

**Versions used**

```
.NET SDK (reflecting any global.json):
 Version:   6.0.100
 Commit:    9e8b04bbff

Runtime Environment:
 OS Name:     debian
 OS Version:  11
 OS Platform: Linux
 RID:         debian.11-arm64
 Base Path:   /home/pi/dotnet/sdk/6.0.100/

Host (useful for support):
  Version: 6.0.0
  Commit:  4822e3c3aa

.NET SDKs installed:
  6.0.100 [/home/pi/dotnet/sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.App 6.0.0 [/home/pi/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 6.0.0 [/home/pi/dotnet/shared/Microsoft.NETCore.App]

To install additional .NET runtimes or SDKs:
  https://aka.ms/dotnet-download
```
- Version of `System.Device.Gpio` package: built from main
- Version of `Iot.Device.Bindings` package: built from main

CodedBeard, 2021-11-16:
I've been doing some more testing with the new driver, and discovered the interface it presents for the same camera is quite different to the original broadcom one.

The following results are produced by `v4l2-ctl --all` using the same hardware (pi4b 1GB using pi v2 camera), but swapping between a sd card using the [last arm release](https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/) and the new [arm64 release](https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2021-11-08/)

`ARM`
```
Driver Info:
        Driver name      : bm2835 mmal
        Card type        : mmal service 16.1
        Bus info         : platform:bcm2835-v4l2-0
        Driver version   : 5.10.17
        Capabilities     : 0x85200005
                Video Capture
                Video Overlay
                Read/Write
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x05200005
                Video Capture
                Video Overlay
                Read/Write
                Streaming
                Extended Pix Format
Priority: 2
Video input : 0 (Camera 0: ok)
Format Video Capture:
        Width/Height      : 1920/1080
        Pixel Format      : 'YUYV' (YUYV 4:2:2)
        Field             : None
        Bytes per Line    : 3840
        Size Image        : 4177920
        Colorspace        : SMPTE 170M
        Transfer Function : Default (maps to Rec. 709)
        YCbCr/HSV Encoding: Default (maps to ITU-R 601)
        Quantization      : Default (maps to Limited Range)
        Flags             :
Format Video Overlay:
        Left/Top    : 150/50
        Width/Height: 1024/768
        Field       : None
        Chroma Key  : 0x00000000
        Global Alpha: 0xff
        Clip Count  : 0
        Clip Bitmap : No
Framebuffer Format:
        Capability    : Extern Overlay
                        Global Alpha
        Flags         : Overlay Matches Capture/Output Size
        Width         : 1024
        Height        : 768
        Pixel Format  : 'YU12'
Streaming Parameters Video Capture:
        Capabilities     : timeperframe
        Frames per second: 30.000 (30000/1000)
        Read buffers     : 1

User Controls

                     brightness 0x00980900 (int)    : min=0 max=100 step=1 default=50 value=50 flags=slider
                       contrast 0x00980901 (int)    : min=-100 max=100 step=1 default=0 value=0 flags=slider
                     saturation 0x00980902 (int)    : min=-100 max=100 step=1 default=0 value=0 flags=slider
                    red_balance 0x0098090e (int)    : min=1 max=7999 step=1 default=1000 value=1000 flags=slider
                   blue_balance 0x0098090f (int)    : min=1 max=7999 step=1 default=1000 value=1000 flags=slider
                horizontal_flip 0x00980914 (bool)   : default=0 value=0
                  vertical_flip 0x00980915 (bool)   : default=0 value=0
           power_line_frequency 0x00980918 (menu)   : min=0 max=3 default=1 value=0
                      sharpness 0x0098091b (int)    : min=-100 max=100 step=1 default=0 value=0 flags=slider
                  color_effects 0x0098091f (menu)   : min=0 max=15 default=0 value=0
                         rotate 0x00980922 (int)    : min=0 max=360 step=90 default=0 value=0 flags=modify-layout
[... 117 lines cut ...]
				1: Color Bars
				2: Solid Color
				3: Grey Color Bars
				4: PN9
                   digital_gain 0x009f0905 (int)    : min=256 max=4095 step=1 default=256 value=256
```

After finding this, I updated the sample project to print out more details about what colors are supported, which returned the following:

```
format = SRGGB10P
Supported resolutions:
[3280x2464]->[3280x2464], Step [0,0] [1920x1080]->[1920x1080], Step [0,0] [1640x1232]->[1640x1232], Step [0,0] [640x480]->[640x480], Step [0,0]
 
format = SRGGB10
Supported resolutions:
[3280x2464]->[3280x2464], Step [0,0] [1920x1080]->[1920x1080], Step [0,0] [1640x1232]->[1640x1232], Step [0,0] [640x480]->[640x480], Step [0,0] 

format = SRGGB8
Supported resolutions:
[3280x2464]->[3280x2464], Step [0,0] [1920x1080]->[1920x1080], Step [0,0] [1640x1232]->[1640x1232], Step [0,0] [640x480]->[640x480], Step [0,0] 

```

I then tried updating the video settings to caputre in one of these colors, however the returned image still wasn't valid. I'll try and do some more investigating later in the week.

jesseryoung, 2021-11-23:
Tested on my pi 3 and pi 4 (running the 64bit Raspbian) I'm seeing the unicam driver for both. I can't find any documentation of it - but it looks like the unicam driver dropped support for the MPJEG pixel format. The only other place that I can find that mentions this is a few complaints about other mjpeg applications breaking after upgrading to Bullseye: https://forums.raspberrypi.com/viewtopic.php?t=323390&sid=a716cd5bd111e1f48f53a80e82fa4fa1 . 

Based on my understanding of `VideoDevice` I don't believe the Mjpeg sample will work if the device itself does not support that pixel format.

CodedBeard, 2021-11-23:
That discussion appears to be around the old apps `raspivid/raspistill` not working, [which is expected](https://www.raspberrypi.com/documentation/accessories/camera.html#libcamera-and-the-legacy-raspicam-camera-stack) as they have been replaced with new apps `libcamera-still/libcamera-vid` designed for the new driver.

The new apps do have an [encoder for mjpeg](https://github.com/raspberrypi/libcamera-apps/blob/main/encoder/mjpeg_encoder.cpp) so i don't think the pixel format is unsupported, just that the current `VideoDevice` implementation no longer maps correctly.

raffaeler, 2022-06-13:
I made some tests that I documented [here](https://github.com/dotnet/iot/issues/1799#issuecomment-1154057026).
I confirm the problems on both 32 and 64 arm OSes.

krwq, 2022-06-30:
[Triage] Closing this in favor of https://github.com/dotnet/iot/issues/1799 since it has the same root cause

raffaeler, 2022-07-01:
I updated [my comment here](https://github.com/dotnet/iot/issues/1799#issuecomment-1154057026) to clarify the short/medium/long term strategies to successfully access a video device.