dotnet/iot #1727: Update Iot.Device.Media to use Span<T> where possible.

jesseryoung, 2021-11-16:
While I was investigating color issues with YUYV data for #1726 I noticed that implementations in [VideoDevice.Converter.cs](https://github.com/dotnet/iot/blob/main/src/devices/Media/VideoDevice/VideoDevice.Converter.cs) could benefit from utilizing `Span<T>` to speed up processing. 

My initial tests just after updating `VideoDevice.YuyvToRgb`  and `VideoDevice.RgbToBitmap` showed 200% improved processing time on a Raspberry PI 3B+ when capturing a YUYV image and converting it to a bitmap.

I believe (but haven't tested) that changing some of the internal functions in [UnixVideoDevice.cs](https://github.com/dotnet/iot/blob/main/src/devices/Media/VideoDevice/Devices/UnixVideoDevice.cs) to use `Span<byte>` instead of `byte[]` may net even more speed improvements, however it would be great if the `Capture` method returned a `Span<byte>` (as well as the `ImageBuffer` property on `NewImageBufferReadyEventArgs`).

I can certainly open a PR with the VideoDevice.Converter.cs improvements, but would it be acceptable to revisit the public API for `VideoDevice` to take full advantage of `Span<T>`?

Updated VideoDevice.Converter.cs code from my testing:
```csharp
public static Color[] YuyvToRgb(Stream stream)
{
    Span<YUYV> colorData = new YUYV[stream.Length / Unsafe.SizeOf<YUYV>()];
    var colorDataBuffer = MemoryMarshal.Cast<YUYV, byte>(colorData);
    stream.Read(colorDataBuffer);

    Color[] colors = new Color[colorData.Length * 2];
    int i = 0;
    foreach (var value in colorData)
    {
        colors[i++] = YuvToRgb(value.Y0, value.U, value.V);
        colors[i++] = YuvToRgb(value.Y1, value.U, value.V);
    }

    return colors;
}

public static Bitmap RgbToBitmap((uint Width, uint Height) size, Color[] colors, System.Drawing.Imaging.PixelFormat format = System.Drawing.Imaging.PixelFormat.Format24bppRgb)
{
    int width = (int)size.Width, height = (int)size.Height;
    Bitmap pic = new Bitmap(width, height, System.Drawing.Imaging.PixelFormat.Format24bppRgb);
    Span<BGR> bgrData = new Span<BGR>(colors
        .Select(c => new BGR()
        {
            R = c.R,
            G = c.G,
            B = c.B,
        }).ToArray());
    var imageData = pic.LockBits(new Rectangle(0, 0, width, height), ImageLockMode.WriteOnly, System.Drawing.Imaging.PixelFormat.Format24bppRgb);
    unsafe
    {
        var imageDataSpan = new Span<BGR>(imageData.Scan0.ToPointer(), imageData.Width * imageData.Height * 3);
        bgrData.CopyTo(imageDataSpan);
    }
    pic.UnlockBits(imageData);
    
    return pic;
}

private struct YUYV
{
    public byte Y0;
    public byte U;
    public byte Y1;
    public byte V;
}
private struct BGR
{
    public byte B;
    public byte G;
    public byte R;
}
```

CodedBeard, 2021-11-16:
I did consider adding `Span<byte>` when making the modifications for #1587, but decided against it after realising that netcore 2.1 was still included in the TFM's. I don't recall exactly what the issue was, but it was something to do with extensions that didn't exist in 2.1, so resulted in `#ifdef` in a few places. I'd certainly support adding `Span<byte>` if 2.1 is being dropped now its out of support.

It might also be worth considering the fact that `System.Drawing` is no longer supported outside of windows in `net6.0` and beyond. Attempting to use it will result in a `System.PlatformNotSupportedException: System.Drawing.Common is not supported on non-Windows platforms` exception.

Add to that, the issue I've mentioned in #1724 where the latest raspberry os no longer uses l4v2, it might be worth taking a deeper look at `VideoDevice` as a whole.

jesseryoung, 2021-11-17:
Ah, wasn't aware of the System.Drawing deprecation, looks like this was discussed in #1403 as well. I wonder if it makes sense to consolidate the issues with `VideoDevice` into a single issue and address them from there?

I'd certainly be willing to test and possibly help develop any changes here.

CodedBeard, 2021-11-18:
I've mainly been trying to figure out #1724, as I was hoping it might resolve the other two issues in the process, but would be happy to get involved in a larger redesign incorporating all of these issues. I have at least one of each new Pi including the zero 2 w, and array of different cameras to test any changes with, but I'm certainly no expert in driver interop or imagery in general.