dotnet/iot #1729: Raspberry Pi Zero 2 W Support

anwarminarso, 2021-11-17:
Hi guys, 

I've used this library with .NET 5 on Raspberry Pi 3 for controlling GPIO, I2C and it works! 
Is it possible for this library to run on Raspberry Pi Zero 2 W?
Raspberry Pi Zero 2 W has the same chip (BCM2710A1) as the Raspberry Pi 3.

Thank you...

raffaeler, 2021-11-17:
Hello. We discussed this in the Triage sometime ago and some of us (not me) got it and successfully tested the device.
If you have any issue, please file a new issue specifying the most detailed info about the issue you meet.

*Note: If this answer is what you need, please close this issue.*

anwarminarso, 2021-11-17:
hi, thank you for your response..

The main issue in this library is not supported for ``Raspberry Pi Zero 2 W``
I have changed the [code](https://github.com/anwarminarso/iot/commit/a2473e87e08416db17f36ac11eae8727b6c12d71) by adding Model dan board revision number on ``RaspberryBoardInfo`` class.

Currently, I have tested the ``GPIO`` and so far it is working fine. I have not tested ``I2C`` and ``SPI``

Thanks

joperezr, 2021-11-17:
Do you want to send out a PR with that change :smiley:? I did test the library and worked out well in Raspberry Pi W 2 but I did install the libgpiod library and was testing that one in particular.

raffaeler, 2021-11-17:
@joperezr commented faster than me :-D (glad you did it)
BTW @Ellerbach tested the SPI successfully

Ellerbach, 2021-11-17:
> BTW @Ellerbach tested the SPI successfully

Yes, SPI, I2C and GPIO works fine :-)

joperezr, 2021-11-18:
Closing as the question seems to be answered now. @anwarminarso please feel free to reopen if that is not the case or if you have any follow up questions.

closed 2021-11-18 by joperezr