dotnet/iot #1730: GPIO SysFsDriver Input not working with Xinlinx-Zynq processors / custom drivers not possible due to "internal" protection

jholzer, 2021-11-18:
Hello!

In SysFsDriver the GpioContoller is hard coded to "pinctrl".
But on some boards containing the Xylinx-Zynq-controller the GPIO "label" file (e.g. "/sys/class/gpio/gpiochip906/label") contains the content "zynq_gpio".
Because of that the reading of the "ReadOffset"-Method fails.

Unfortunately writing an own driver based on e.g. "SysFsDriver", "UnixDriver" or "GpioDriver" is not possible, since these classes cannot be derived in 3rd-party components, since e.g. GpioDriver contains "protected internal abstract" methods, that cannot be overridden from outside the "System.Device.Gpio"-DLL.

Suggestion:
Best option for future and 3rd-party extensions might be, to remove the protection from the base-classes.
By that custom drivers based on GpioDriver will be possible and users are not limited to built-in driver.

Thanks & Regards
Jürgen

raffaeler, 2021-11-18:
> But on some boards containing the Xylinx-Zynq-controller the GPIO "label" file (e.g. "/sys/class/gpio/gpiochip906/label") contains the content "zynq_gpio".

Out of my curiosity, can you tell what brand/model are those boards?

Thank you

jholzer, 2021-11-18:
Sure!

It´s a board from german manufacturer "Trenz electronic" using a Xilinx SOC named "SoC Module with Xilinx Zynq XC7Z015-2CLG485I, 1 GByte DDR3L, 4 x 5 cm"

Here´s the link to their product page:

[https://shop.trenz-electronic.de/en/TE0715-04-52I33-A-SoC-Module-with-Xilinx-Zynq-XC7Z015-2CLG485I-1-GByte-DDR3L-4-x-5-cm](https://shop.trenz-electronic.de/en/TE0715-04-52I33-A-SoC-Module-with-Xilinx-Zynq-XC7Z015-2CLG485I-1-GByte-DDR3L-4-x-5-cm)

pgrawehr, 2021-11-19:
> Unfortunately writing an own driver based on e.g. "SysFsDriver", "UnixDriver" or "GpioDriver" is not possible, since these classes cannot be derived in 3rd-party components, since e.g. GpioDriver [...]

I think you're being misstaken here. `protected internal` means `protected` **or** `internal`. It is possible to derive from the class and override the members (as protected) from outside. We already have a few such extended drivers in `src\devices\Gpio\Drivers`. What you cannot do from outside is directly call one of the `protected internal` members from another class, but you won't normally need that, because you would still feed it as argument to `GpioController`.

joperezr, 2021-11-19:
correct, the only reason why we made them `protected internal` as opposed to just `internal` was so that they could be extended outside of System.Device.Gpio. The way it works, is that when you try to derive from that an override methods, your members will only become protected, and when you pass your driver as a parameter to GpioController it will still be able to call the overwritten methods since it can call the GpioDriver internal ones which would be virtual now and use yours instead.

jholzer, 2021-11-22:
Well, as mentioned the original issue was, that the content of the "label" was different on our board, than expected from the SysFsDriver.
Since it´s a constant there the original idea was, to make an own implementation of SysFsDriver  (let´s name it "ZynqSysFsDriver" for now), based on UnixDriver. But that´s not possible due to the "internals" used by the SysFsDriver (e.g. "UnixDriverDevicePin" is internal sealed).
So implementing a custom driver is not as simple as it could be due to that limitations.
If that limitation was not there, the SysFsDriver could be quite simply be adapted to boards that have a different behavior than the ones taken into account in the original implementation...
I´m pretty open to other ways to solve this issues, but implementing a completely new driver just because of one contant (which in fact solves the issue for us) is not really satisfying. 
Most of all considering future improvements of the System.Device.Gpio project or the SysFsDriver...

krwq, 2022-10-13:
[Triage] You can actually override internal protected. Closing this as by design. `internal protected` means `internal` or `protected`. `internal` and `protected` members are marked with `private protected`

closed 2022-10-13 by krwq